mavlink/mavros #638: Using Mavros, Drone always see east

winszenYoung, 2016-12-22:
### Issue details

Hi, I have a few problem of using mavros.

I control iris using Mavros in simulator gazebo.
Controlling drone works very well.
but it has problem.

I using this code : http://dev.px4.io/ros-mavros-offboard.html.
Drone heading always see  east.

In gazebo, Iris change yaw. 
using that code iris automatically see east.

Test set:
yaw1:https://youtu.be/dPGjgZBk82s
yaw2 : https://youtu.be/efkYQjqigRY
yaw3: https://youtu.be/fmzJt1ABc2g
yaw4: https://youtu.be/MFGAMfSHzug

And, I trying to  real drone.
The problem occurs same.

Test set:
https://youtu.be/_arn7If4UM8

### MAVROS version and platform

Mavros: 0.18
ROS: Kinetic
Ubuntu: 16.04

### Autopilot type and version

[  ] ArduPilot
[ V] PX4

Version: ?3.7.1?

### Node logs

### Diagnostics

### Check ID

vooon, 2016-12-22:
Please add code, where you fill PoseStamped orientation quaternion.

winszenYoung, 2016-12-22:
thanks . it works.
but I don't want to  spin .
so i try read fcu imu data
using /mavros/local_position/local
but dosen't work
how can I read imu data?

vooon, 2016-12-22:
You can use setpoint_raw to send only position vector, without yaw.
Position local depends on FCU configuration. Did you setup companion?

vooon, 2016-12-22:
And you may send orientation q for angle [0 0 pi]:

```python
import tf
from math import pi

In [12]: tf.transformations.quaternion_from_euler(0,0,pi)
Out[12]: 
array([  0.00000000e+00,   0.00000000e+00,   1.00000000e+00,
         6.12323400e-17])
```

winszenYoung, 2016-12-27:
thanks,  I solve  this problem.
I get  heading using /mavros/globoal_position/compass_hdg
and rotate quaternion transform it works.

closed 2016-12-27 by winszenYoung