mavlink/mavros #640: mavros imu topic details

xiaopenghuang, 2016-12-27:
### Issue details
Hi all,
I am now read imu message from mavros. I find there exist two imu topic, I donot know there any difference, except the orientation.(one is zero, another is compute by FCU)

- /mavros/imu/data_raw
```yaml
header: 
  seq: 4069
  stamp: 
    secs: 1482805673
    nsecs: 433939136
  frame_id: fcu
orientation: 
  x: 0.0
  y: 0.0
  z: 0.0
  w: 0.0
orientation_covariance: [-1.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
angular_velocity: 
  x: 0.00447485502809
  y: -0.00115696352441
  z: -0.00598003994673
angular_velocity_covariance: [1.2184696791468346e-07, 0.0, 0.0, 0.0, 1.2184696791468346e-07, 0.0, 0.0, 0.0, 1.2184696791468346e-07]
linear_acceleration: 
  x: -0.0385106429458
  y: -0.0702816992998
  z: 9.6683549881
linear_acceleration_covariance: [8.999999999999999e-08, 0.0, 0.0, 0.0, 8.999999999999999e-08, 0.0, 0.0, 0.0, 8.999999999999999e-08]

```
- /mavros/imu/data
```yaml
header: 
  seq: 13198
  stamp: 
    secs: 1482805665
    nsecs: 745116288
  frame_id: fcu
orientation: 
  x: -0.0039281528076
  y: -0.000290539941681
  z: 0.548376567251
  w: 0.836222234631
orientation_covariance: [1.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 1.0]
angular_velocity: 
  x: 0.00107183912769
  y: -0.000517296080943
  z: -0.000133892521262
angular_velocity_covariance: [1.2184696791468346e-07, 0.0, 0.0, 0.0, 1.2184696791468346e-07, 0.0, 0.0, 0.0, 1.2184696791468346e-07]
linear_acceleration: 
  x: -0.036279976368
  y: -0.0749365612864
  z: 9.6851272583
linear_acceleration_covariance: [8.999999999999999e-08, 0.0, 0.0, 0.0, 8.999999999999999e-08, 0.0, 0.0, 0.0, 8.999999999999999e-08]
```

And I find one topic is 168hz ,the another topic is 46hz. Why their frequency is different so big. And the imu message mavros publish is through pixhawk algorithmn(EKF ..) to get it. or just read from imu sensors directly.

Is there exist any solution that I can read imu sensor directly , do not need any algorithmn process it , just raw data about sensor.

Thanks!

mhkabir, 2016-12-27:
Raw data (no sensor fusion) : `/mavros/imu/data_raw`
Fused data (bias corrected, orientation computed) : `/mavros/imu/data`

closed 2016-12-27 by mhkabir

xiaopenghuang, 2016-12-27:
@mhkabir ,how to compute the imu  bias and covariance , has any material or formula about it .And why the raw data frequency is so low , the sensor should has a much higher  frequency output.