mavlink/mavros #642: serial0: receive: End of file

meuchel, 2017-01-01:
I have trouble with baudrates and companien computer on TELEM1.

[ INFO] [1483270621.203265724]: serial0: device: /dev/ttyUSB0 @ 57600 bps
[ INFO] [1483270621.204183295]: GCS URL: udp-b://@
[ERROR] [1483270621.204251893]: serial0: receive: End of file
terminate called after throwing an instance of 'std::system_error'

My setup is mavros (latest master) with ROS Kinetic on Intel NUC with Ubuntu 16.04 LTS and APM Rover v3.1.0 on Pixhawk
I'm using a USB to Serial interface connected to TELEM1 
The interface is similar as shown here
http://ardupilot.org/dev/docs/odroid-via-mavlink.html
It is connected to TELEM1

I can configure SERIAL1_BAUD with any of below listed baudrates and when connecting via apmplanner2 it is always working even with 1500000
57600
115200
921600
1500000

but with mavros I get always
[ERROR] [1483270621.204251893]: serial0: receive: End of file

What could be wrong?
This is the output when launching mavros:

$ roslaunch mavros.launch
... logging to /home/ros/.ros/log/7edf52c0-d00d-11e6-81c7-c03fd56730d7/roslaunch-uav-ros1-7055.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://uav-ros1:39528/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: ROLL_180
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: ROLL_180
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: ROLL_180
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: ROLL_180
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /mavros/fcu_url: /dev/ttyUSB0:57600
 * /mavros/gcs_url: udp-b://@
 * /mavros/global_position/frame_id: fcu
 * /mavros/global_position/rot_covariance: 99999.0
 * /mavros/global_position/tf/child_frame_id: fcu_utm
 * /mavros/global_position/tf/frame_id: local_origin
 * /mavros/global_position/tf/send: True
 * /mavros/image/frame_id: px4flow
 * /mavros/imu/angular_velocity_stdev: 0.000349065850399
 * /mavros/imu/frame_id: fcu
 * /mavros/imu/linear_acceleration_stdev: 0.0003
 * /mavros/imu/magnetic_stdev: 0.0
 * /mavros/imu/orientation_stdev: 1.0
 * /mavros/local_position/frame_id: fcu
 * /mavros/local_position/tf/child_frame_id: fcu
 * /mavros/local_position/tf/frame_id: local_origin
 * /mavros/local_position/tf/send: False
 * /mavros/local_position/tf/send_fcu: False
 * /mavros/mission/pull_after_gcs: True
 * /mavros/mocap/use_pose: True
 * /mavros/mocap/use_tf: False
 * /mavros/plugin_blacklist: ['safety_area', '...
 * /mavros/plugin_whitelist: []
 * /mavros/px4flow/frame_id: px4flow
 * /mavros/px4flow/ranger_fov: 0.0
 * /mavros/px4flow/ranger_max_range: 5.0
 * /mavros/px4flow/ranger_min_range: 0.3
 * /mavros/safety_area/p1/x: 1.0
 * /mavros/safety_area/p1/y: 1.0
 * /mavros/safety_area/p1/z: 1.0
 * /mavros/safety_area/p2/x: -1.0
 * /mavros/safety_area/p2/y: -1.0
 * /mavros/safety_area/p2/z: -1.0
 * /mavros/setpoint_accel/send_force: False
 * /mavros/setpoint_attitude/reverse_throttle: False
 * /mavros/setpoint_attitude/tf/child_frame_id: attitude
 * /mavros/setpoint_attitude/tf/frame_id: local_origin
 * /mavros/setpoint_attitude/tf/listen: False
 * /mavros/setpoint_attitude/tf/rate_limit: 10.0
 * /mavros/setpoint_position/tf/child_frame_id: setpoint
 * /mavros/setpoint_position/tf/frame_id: local_origin
 * /mavros/setpoint_position/tf/listen: False
 * /mavros/setpoint_position/tf/rate_limit: 50.0
 * /mavros/startup_px4_usb_quirk: True
 * /mavros/sys/disable_diag: False
 * /mavros/sys/min_voltage: 10.0
 * /mavros/target_component_id: 1
 * /mavros/target_system_id: 1
 * /mavros/tdr_radio/low_rssi: 40
 * /mavros/time/time_ref_source: fcu
 * /mavros/time/timesync_avg_alpha: 0.6
 * /mavros/vibration/frame_id: vibration
 * /mavros/vision_pose/tf/child_frame_id: vision
 * /mavros/vision_pose/tf/frame_id: local_origin
 * /mavros/vision_pose/tf/listen: False
 * /mavros/vision_pose/tf/rate_limit: 10.0
 * /mavros/vision_speed/listen_twist: False
 * /rosdistro: kinetic
 * /rosversion: 1.12.6

NODES
  /
    mavros (mavros/mavros_node)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[mavros-1]: started with pid [7074]
[ INFO] [1483270621.199813205]: FCU URL: /dev/ttyUSB0:57600
[ INFO] [1483270621.203265724]: serial0: device: /dev/ttyUSB0 @ 57600 bps
[ INFO] [1483270621.204183295]: GCS URL: udp-b://@
[ERROR] [1483270621.204251893]: serial0: receive: End of file
terminate called after throwing an instance of 'std::system_error'
  what():  Resource deadlock avoided
================================================================================REQUIRED process [mavros-1] has died!
process has died [pid 7074, exit code -6, cmd /home/ros/catkin_mavros/devel/lib/mavros/mavros_node __name:=mavros __log:=/home/ros/.ros/log/7edf52c0-d00d-11e6-81c7-c03fd56730d7/mavros-1.log].
log file: /home/ros/.ros/log/7edf52c0-d00d-11e6-81c7-c03fd56730d7/mavros-1*.log
Initiating shutdown!
================================================================================
[mavros-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

vooon, 2017-01-01:
Does miniterm.py works with that port?

meuchel, 2017-01-01:
Yes.
But meanwhile its working again anyway.
At the end a restart of the companien computer solved it.

But it is still not clear to me when or why the computer comes into this state that mavros can no more connect to pixhawk (or better receive: End of file happens) whereas mavproxy or apmplanner2 have no problem to connect and are still working.

After changing properly TELEM12 to 921600 and rebooting the computer+fcu i could work with mavros all the day with no problems (start/stop many times).
But after you told me to try miniterm the issue came back suddenly (mavproxy or apmplanner2 can still connect but mavros no more)

I stopped my working mavros node, then I tested with miniterm a little bit around as you told me and I could see the stream from fcu.
Then I killed miniterm from another terminal.
At this moment the issue came back and connecting to fcu with mavros worked no more and I needed to restart. (miniterm worked always, before the issue came up and also after the issue came back again)

Anyway here the output from minicom:
$ miniterm.py - 921600

--- Available ports:
---  1: /dev/ttyUSB0         CP2102 USB to UART Bridge Controller
--- Enter port index or full name: 1
--- Miniterm on /dev/ttyUSB0  921600,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
<streamoutput>

vooon, 2017-01-02:
Why not to use `CTRL + ]` combination? Looks like port stay blocked by pyserial, so ASIO can't open it.
Perhaps easier workaround is reboot.

meuchel, 2017-01-02:
Even when I stop miniterm with `CTRL + ]` afterwards mavros can no more connect (mavproy/apmplanner2 can still do) ) and I have to reboot.
Yes, there is a workaround, and this issue can be closed if you want, but nevertheless it's an ugly bug.

I have another question (I don't want to open another new issue because it is also serial port related)
Why I cant use baudrate 1500000 with movros (here also mavproxy/apmplanner2/px4v2 works with this baudrate)?

<arg name="fcu_url" default="/dev/ttyUSB0:1500000" />
...
[ INFO] [1483349478.760910881]: serial0: device: /dev/ttyUSB0 @ 1500000 bps
[FATAL] [1483349478.762333895]: FCU: DeviceError:serial:set_option: Invalid argument

(independed if using 1500 instead of 921 gives really a benefit, I would use 1500 if it would work)

vooon, 2017-01-02:
As you can see , ASIO do not support that baud rate.

`/usr/include/boost/asio/impl/serial_port_base.ipp`
```cpp
boost::system::error_code serial_port_base::baud_rate::store(
    BOOST_ASIO_OPTION_STORAGE& storage, boost::system::error_code& ec) const
{
```

closed 2017-01-02 by vooon

mhkabir, 2017-01-02:
@vooon Is there a possibility to remove the ASIO dependency? Not supporting some high baud rates is especially annoying. e.g For the Intel Aero, default data rate is 1500000. I would also like to move to higher bauds to enable things like log streaming.

vooon, 2017-01-02:
@mhkabir my previous attempt fails. And now i don't see any good reason to do it.

vooon, 2017-01-02:
Also did you really use whole 921600 bandwidth (~90 KiB/s)?

meuchel, 2017-01-02:
its more if something is send it can be received faster and therefore processed earlier.
The difference 912 vs 1500 isn't just marginal

vooon, 2017-01-02:
`SET_POSITION_TARGET_LOCAL_NED`

52 + 10 = 62 bytes => transmission time 673 usec at sub 1M and 415 usec as 1.5M.
That all much smaller, than usual OS tick. So i'm not so optimistic at "processed faster".

And anyway 200 usec is no matter, because of frame inertia.

meuchel, 2017-01-02:
Every bit counts :), thanks for your efforts figuring this out!

amilcarlucas, 2020-11-20:
There [seams to be a solution](https://www.programmersought.com/article/95921196958/)

It adds support for 1500Kbps in ASIO :)

dssdyx, 2023-04-04:
> 

It's a good solution. But I think if just for higher trans speed, we can choose other baud like 2000000 or 3000000 which is also supported by FCU.