mavlink/mavros #643: APM Rover only: Control APM Rover via teleop twist message?

meuchel, 2017-01-06:
I want to take over control of APM Rover with current firmware on PX4v2 somehow from mavros with teleop twist message.

The rover setup is like this:
RC receiver -> PX4v2 (no GPS) - > motor controller board in mixing mode (throttle+steering)
So quite a normal setup.

My first idea was to use /mavros/setpoint_velocity/cmd_vel and set the fcu into OFFBOARD mode.
But this seems not to be possible
a) Rover seems not to support setpoint_velocity/cmd_vel 
b) OFFBOARD is no APM Rover mode (http://ardupilot.org/rover/docs/rover-control-modes.html)

Then I searched thru the mavros issues if I can find something related to this topic.
There are many issue comments but not always related to APM Rover or they are confusing or there are needs to change FCU firmware which is no acceptable.
I like programming but why always reinvent everything if there are already other possibilities to realize it.

So I want just to ask the clear question:

What needs to be done in a sensefull way with the current state of mavros /  APM Rover firmware to take over control somehow from mavros with teleop twist message?

khancyr, 2017-01-11:
Infortunatly, Apmrover doesn't have velocity setpoint yet, and it needs gps. To use the position setpoint you need to be in Guided mode.

khancyr, 2017-03-06:
Support for setpoint_velocity should come with next rover release : PR in review https://github.com/ArduPilot/ardupilot/pull/5735

TSC21, 2017-04-04:
Closing as it is not MAVROS related

closed 2017-04-04 by TSC21

jtroeth, 2017-04-19:
Why was this closed and how is it not mavros related? 
There doesn't seem to be much info on the internet for and offboard/guided mode mavros control for rover. 
I am trying to use guided mode for obstacle avoidance with mavros, using the offboard example as a base and changing the it from off board to guided moded. This works no problems however the rover will only go in a forwards direction no matter the x,y point that is put it. 

Can anyone help with this?

Thanks

TSC21, 2017-04-19:
> Why was this closed and how is it not mavros related?

The answer:

> Support for setpoint_velocity should come with next rover release : PR in review ArduPilot/ardupilot#5735

TSC21, 2017-04-19:
> Can anyone help with this?

Also if you are not specific about your issue, share code and logs, it will be difficult to someone to help you out.