mavlink/mavros #645: Mavros dies and kills jetson TX1 also

skysingh, 2017-01-12:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details
I have been running mavros on ubuntu laptop 14.04 indigo fine but I am moving to jetson TX1 running 16.04 kinetic. I am connected over UART1 (ttyS0)
_Please describe the problem, or desired feature_
I can see traffic on ttyS0 using cat /dev/ttyS0 so I know the hardware is working. when I launch px4.launch I get the output shown below but within a few seconds the system dies and the TX1 is completety unresponsive. Need to disconnect the pixhawk and then restart TX1 !

### MAVROS version and platform

Mavros: ?0.18.4?
ROS: Kinetic - installed using procdedure by Jetson hacks. ROS works ok
Ubuntu: ?16.04?

kinetic 16.06  tegra

### Autopilot type and version

[  ] ArduPilot
[ X ] PX4

Version: ?3.7.1?

### Node logs

```
copy output of mavros_node. Usually console where you run roslaunch
```NODES
  /
    mavros (mavros/mavros_node)
    mavros_node (rosservice/rosservice)

auto-starting new master
process[master]: started with pid [3146]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to fb84298a-d905-11e6-8ee2-00044b65c8de
process[rosout-1]: started with pid [3160]
started core service [/rosout]
process[mavros-2]: started with pid [3178]
process[mavros_node-3]: started with pid [3179]
[ INFO] [1484253041.516252133]: FCU URL: /dev/ttyS0:921600
[ WARN] [1484253041.517881599]: init: message from ASLUAV, MSG-ID 201 ignored! Table has different entry.
[ WARN] [1484253041.518318092]: init: message from autoquad, MSG-ID 150 ignored! Table has different entry.
[ WARN] [1484253041.518368923]: init: message from autoquad, MSG-ID 152 ignored! Table has different entry.
[ WARN] [1484253041.518770886]: init: message from matrixpilot, MSG-ID 150 ignored! Table has different entry.
[ WARN] [1484253041.518815260]: init: message from matrixpilot, MSG-ID 151 ignored! Table has different entry.
[ WARN] [1484253041.518853852]: init: message from matrixpilot, MSG-ID 152 ignored! Table has different entry.
[ WARN] [1484253041.518900933]: init: message from matrixpilot, MSG-ID 153 ignored! Table has different entry.
[ WARN] [1484253041.518935776]: init: message from matrixpilot, MSG-ID 155 ignored! Table has different entry.
[ WARN] [1484253041.518968899]: init: message from matrixpilot, MSG-ID 156 ignored! Table has different entry.
[ WARN] [1484253041.519009471]: init: message from matrixpilot, MSG-ID 157 ignored! Table has different entry.
[ WARN] [1484253041.519085301]: init: message from matrixpilot, MSG-ID 158 ignored! Table has different entry.
[ WARN] [1484253041.519127695]: init: message from matrixpilot, MSG-ID 170 ignored! Table has different entry.
[ WARN] [1484253041.519189203]: init: message from matrixpilot, MSG-ID 171 ignored! Table has different entry.
[ WARN] [1484253041.519222639]: init: message from matrixpilot, MSG-ID 172 ignored! Table has different entry.
[ WARN] [1484253041.519252638]: init: message from matrixpilot, MSG-ID 173 ignored! Table has different entry.
[ WARN] [1484253041.519282897]: init: message from matrixpilot, MSG-ID 174 ignored! Table has different entry.
[ WARN] [1484253041.519313104]: init: message from matrixpilot, MSG-ID 175 ignored! Table has different entry.
[ WARN] [1484253041.519342843]: init: message from matrixpilot, MSG-ID 176 ignored! Table has different entry.
[ WARN] [1484253041.519373675]: init: message from matrixpilot, MSG-ID 177 ignored! Table has different entry.
[ WARN] [1484253041.519402997]: init: message from matrixpilot, MSG-ID 178 ignored! Table has different entry.
[ WARN] [1484253041.519467734]: init: message from matrixpilot, MSG-ID 179 ignored! Table has different entry.
[ WARN] [1484253041.519505024]: init: message from matrixpilot, MSG-ID 180 ignored! Table has different entry.
[ WARN] [1484253041.519534866]: init: message from matrixpilot, MSG-ID 181 ignored! Table has different entry.
[ WARN] [1484253041.519564605]: init: message from matrixpilot, MSG-ID 182 ignored! Table has different entry.
[ WARN] [1484253041.519592833]: init: message from matrixpilot, MSG-ID 183 ignored! Table has different entry.
[ WARN] [1484253041.519621425]: init: message from matrixpilot, MSG-ID 184 ignored! Table has different entry.
[... 80 lines cut ...]
process has died [pid 3178, exit code -9, cmd /opt/ros/kinetic/lib/mavros/mavros_node __name:=mavros __log:=/home/ubuntu/.ros/log/fb84298a-d905-11e6-8ee2-00044b65c8de/mavros-2.log].
log file: /home/ubuntu/.ros/log/fb84298a-d905-11e6-8ee2-00044b65c8de/mavros-2*.log
Initiating shutdown!
================================================================================
[mavros-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

### Diagnostics

```
place here result of:
rostopic echo -n1 /diagnostics
```
cannot run this is uart connected to pixhawk !

### Check ID

```
rosrun mavros checkid
```
gives no response. need to ctrl-C

skysingh, 2017-01-12:
Following up on my own question. I changed the TX1 to use UART1 instead of UART0. This shows up in 16.04 L4T as /dev/ttyTHS2. I can see traffic from the pixhawk using cat. Now if run px4.launch with ttyTHS2 as the _fcu_url mavros starts up and communicates with the pixhawk. So some progress. Maybe the TX1 only wants console on UART0. HOWEVER, I cannot keep mavros running - gives all kinds of timeouts:
[mavros_node-3] process has finished cleanly
log file: /home/ubuntu/.ros/log/ef0866ae-d90e-11e6-9afb-00044b65c8de/mavros_node-3*.log
[ INFO] [1484256982.599069436]: CON: Got HEARTBEAT, connected. FCU: ArduPilotMega / ArduCopter
[ INFO] [1484256982.605230523]: FCU: Initialising APM
[ INFO] [1484256982.607989299]: FCU: Calibrating barometer
[ WARN] [1484256984.641218281]: VER: broadcast request timeout, retries left 4
[ WARN] [1484256985.621655956]: VER: broadcast request timeout, retries left 3
[ WARN] [1484256991.628238323]: VER: unicast request timeout, retries left 2
[ WARN] [1484256992.714774777]: CON: Lost connection, HEARTBEAT timed out.
[ WARN] [1484256993.605895284]: PR: request list timeout, retries left 2
[ WARN] [1484256994.613373326]: PR: request list timeout, retries left 1
[ WARN] [1484256995.617775147]: PR: request list timeout, retries left 0

If i try ubuntu@tegra-ubuntu:~$ rostopic echo mavros/state

I get just this one response then hangs...

header: 
  seq: 3
  stamp: 
    secs: 1484256996
    nsecs: 675819790
  frame_id: ''
connected: False
armed: False
guided: False
mode: ''
---

---- diagnostics -->

ubuntu@tegra-ubuntu:~$ rostopic echo -n1 /diagnostics
header: 
  seq: 262
  stamp: 
    secs: 1484257236
    nsecs: 494919849
  frame_id: ''
status: 
  - 
    level: 1
    name: mavros: FCU connection
    message: not connected
    hardware_id: /dev/ttyTHS2:921600
    values: 
      - 
        key: Received packets:
        value: 10
      - 
        key: Dropped packets:
        value: 0
      - 
        key: Buffer overruns:
        value: 0
      - 
        key: Parse errors:
        value: 0
      - 
        key: Rx sequence number:
        value: 8
      - 
        key: Tx sequence number:
        value: 119
      - 
        key: Rx total bytes:
        value: 635
      - 
        key: Tx total bytes:
        value: 97358
      - 
        key: Rx speed:
        value: 0.000000
      - 
        key: Tx speed:
        value: 277.000000
  - 
    level: 2
    name: mavros: GPS
    message: No satellites
    hardware_id: /dev/ttyTHS2:921600
    values: 
      - 
        key: Satellites visible
        value: 0
      - 
        key: Fix type
        value: 0
      - 
        key: EPH (m)
        value: Unknown
      - 
        key: EPV (m)
        value: Unknown
  - 
    level: 2
    name: mavros: Heartbeat
    message: No events recorded.
    hardware_id: /dev/ttyTHS2:921600
    values: 
      - 
        key: Heartbeats since startup
        value: 2
      - 
        key: Frequency (Hz)
        value: 0.000000
      - 
        key: Vehicle type
        value: Ground rover
      - 
        key: Autopilot type
        value: ArduPilotMega / ArduCopter
      - 
        key: Mode
        value: INITIALISING
      - 
        key: System status
        value: Calibrating
  - 
    level: 2
    name: mavros: System
    message: Sensor helth
    hardware_id: /dev/ttyTHS2:921600
    values: 
      - 
        key: Sensor present
        value: 0x0020DC07
      - 
        key: Sensor enabled
        value: 0x00208007
      - 
        key: Sensor helth
        value: 0x0020DC04
      - 
        key: 3D gyro
        value: Fail
      - 
        key: 3D accelerometer
        value: Fail
      - 
        key: 3D magnetometer
        value: Ok
      - 
        key: motor outputs / control
        value: Ok
      - 
        key: AHRS subsystem health
        value: Ok
      - 
        key: CPU Load (%)
        value: 0.0
      - 
        key: Drop rate (%)
        value: 0.0
      - 
        key: Errors comm
        value: 0
      - 
        key: Errors count #1
        value: 0
      - 
        key: Errors count #2
        value: 0
      - 
        key: Errors count #3
        value: 0
      - 
        key: Errors count #4
        value: 0
  - 
    level: 1
    name: mavros: Battery
    message: Low voltage
    hardware_id: /dev/ttyTHS2:921600
    values: 
      - 
        key: Voltage
        value: 0.00
      - 
        key: Current
        value: -0.0
      - 
        key: Remaining
        value: -1.0
  - 
    level: 2
    name: mavros: Time Sync
    message: No events recorded.
    hardware_id: /dev/ttyTHS2:921600
    values: 
      - 
        key: Timesyncs since startup
        value: 0
      - 
        key: Frequency (Hz)
        value: 0.000000
      - 
        key: Last dt (ms)
        value: 0.000000
      - 
        key: Mean dt (ms)
        value: 0.000000
      - 
        key: Last system time (s)
        value: 0.000000000
      - 
        key: Time offset (s)
        value: 0.000000000
---
ubuntu@tegra-ubuntu:~$

mhkabir, 2017-01-13:
It's is a bug in the Tegra serial driver. I have been able to reproduce this forever.

mhkabir, 2017-01-13:
But as far as Nvidia told me, the 16.04 release shouldn't have it. 

Did you disable the system serial console on UART0? Because it will cause the same issues as you're seeing unless the console is completely disabled. I don't remember the exact procedure, but you need to edit a few files to disable console output on UART0. 

I would suggest using UART1 however.

vooon, 2017-01-13:
Also maybe better to use USB-UART since nvidia does crap again...

Look at `dmesg`, try to run under `strace`, try lower baud rates (e.g. 57600).

To run under strace you must start `roscore`, then you may load configuration via `rosparam load /mavros ~/ws/src/mavros/mavros/launch/px4_config.yaml`
Then you been able to run `strace -fy ~/ws/devel/lib/mavros/mavros_node _fcu_url:=/dev/ttyTHS1`

skysingh, 2017-01-13:
Good news. mavros works with JetsonTX1 running ubuntu 16.04 L4T! Problem with earlier test using uart1 was a bad regulator so pixhawk was resetting on brownout. BTW I am using the connect-tech orbitty carrier board on the TX1. It is compact and excellent.

So for any one connecting pixhawk via mavros to Jetson TX1:
1. use UART1. Connect tele2 to UART1 rx and TX. Make sure baud rate is set in pixhawk. Check that you get pixhawk traffic using cat /dev/ttyTHS2. Under ubuntu L4T uart1 shows up as THS2
2. use the standard px4.launch file. set _fcu_ url = /dev/ttyTHS2:921600. Then roslaunch px4.launch
3. start rosservice (can put inside launch file):
<node pkg="rosservice" type="rosservice" name="mavros_node" args="call --wait /mavros/set_stream_rate 0 20 1"/>
4. node should start up with good messages
5. in another terminal try rostopic echo mavros/state. You should get good messages. Then you are all set.

Hope this helps someone. Thanks to mhkabir and vooon

closed 2017-01-13 by vooon

sunnye26, 2018-06-25:
How did you guys connect the Jetson TX1 to the pixhawk? Do you have a picture of the wiring done?