mavlink/mavros #646: Connexion problem with ardupilot

khancyr, 2017-01-13:
### Issue details

Connecting mavros on tcp on ardupilot sitl (rover and copter tested) on lastest master (https://github.com/ArduPilot/ardupilot/commit/4daa4e9df95f85911c36951352dc05ccc6e3e06c) make the connexion buggy when conneting on tcp : <arg name="fcu_url" default="tcp://localhost:5762" />

the problem isn't here with mavproxy : mavproxy.py --master="tcp:127.0.0.1:5762"  --console --mav10

cmd for launching sitl : sim_vehicle.py -v ArduCopter -m --mav10 --console --map -w  

on sitl console 
````
Mode(0x000000c0)> STABILIZE> Mode(0x000000c0)> STABILIZE> Mode(0x000000c0)> STABILIZE> Mode(0x000000c0)> STABILIZE> Mode(0x000000c0)> STABILIZE> Mode(0x000000c0)> STABILIZE> Mode(0x000000c0)> STABILIZE> Mode(0x000000c0)> STABILIZE> Mode(0x000000c0)> STABILIZE> Mode(0x000000c0)> STABILIZE> Mode(0x000000c0)> STABILIZE> Mode(0x000000c0)> STABILIZE> Mode(0x000000c0)> STABILIZE> Mode(0x000000c0)> STABILIZE> Mode(0x000000c0)
````
On mavproxy console : ARM , DISARM, ARM , DISARM MODE STABILIZE etc...

### MAVROS version and platform

Mavros: 0.18.5-0xenial-20161212-094034-0800
ROS: Kinetic
Ubuntu: 16.04

### Autopilot type and version

[X] ArduPilot
[  ] PX4

Version: master

### Node logs

```
─$ roslaunch apm.launch                                                    1 ↵
... logging to /home/khancyr/.ros/log/98c27694-d978-11e6-8be1-40e230b7507d/roslaunch-khancyr-N551JQ-21771.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://khancyr-N551JQ:36969/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 0.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 0.0
 * /mavros/fcu_url: tcp://localhost:5763
 * /mavros/gcs_url: 
 * /mavros/global_position/frame_id: fcu
 * /mavros/global_position/rot_covariance: 99999.0
 * /mavros/global_position/tf/child_frame_id: fcu_utm
 * /mavros/global_position/tf/frame_id: local_origin
 * /mavros/global_position/tf/send: False
 * /mavros/image/frame_id: px4flow
 * /mavros/imu/angular_velocity_stdev: 0.000349065850399
 * /mavros/imu/frame_id: fcu
 * /mavros/imu/linear_acceleration_stdev: 0.0003
 * /mavros/imu/magnetic_stdev: 0.0
 * /mavros/imu/orientation_stdev: 1.0
 * /mavros/local_position/frame_id: fcu
 * /mavros/local_position/tf/child_frame_id: fcu
 * /mavros/local_position/tf/frame_id: local_origin
 * /mavros/local_position/tf/send: False
 * /mavros/mission/pull_after_gcs: True
 * /mavros/mocap/use_pose: True
 * /mavros/mocap/use_tf: False
[... 333 lines cut ...]
    values: 
      - 
        key: Voltage
        value: 0.00
      - 
        key: Current
        value: 0.0
      - 
        key: Remaining
        value: 100.0
---
```

### Check ID

```
╰─$ rosrun mavros checkid
OK. I got messages from 1:1.

---
Received 33 messages, from 2 addresses
sys:comp   list of messages
 255:0     0, 66
  1:1     0, 22
```

vooon, 2017-01-13:
Please be more verbose, what wrong.
Because i see in that log, that mavros successfully establish connection with APM.
Perhaps you need to setup message stream rates (`rosrun mavros mavsys rate --all 10`)?

Or you try to connect mavproxy to same port? As i remember APM only accepts one TCP connection per "UART".
Also check param `~fcu_protocol` is unset or `v1.0`.

khancyr, 2017-01-13:
sitl is spam with arm , disarm and change mode to stabilize message .
I use only one connection on a uart
rosparam show no fcu_protocol

vooon, 2017-01-13:
No param == `v1.0`. I'm don't know why. Try to change heartbeat rate / setup SYSID_MYGCS.
Anyway i don't see problem on mavros side.

khancyr, 2017-01-13:
Yep, I try to modify fcu_protocol, sysid, but bug is still there. 
But it doesn't seems to be log neither by apm nor by mavproxy, this is quite strange but easily reproducible... 
I post it here because I only find this with mavros. I will try to do more test to find where exactly the issue is, it could totally be something on sitl side .

annesteenbeek, 2017-02-14:
@khancyr I've had the same problem, however it does not always occur. Haven't had it with SITL, only with actual copter, it's hard to reproduce, however it does only happen when using multiple mavlink sources/sinks. Did you ever figure out what caused it?

TSC21, 2017-04-04:
@khancyr is this solved to you?

khancyr, 2017-04-05:
@TSC21 Hum... it seems solve on lastest master and mavros on a fresh install ubuntu 16.04. So I will close it

closed 2017-04-05 by khancyr

TSC21, 2017-04-05:
Good

annesteenbeek, 2017-06-14:
I still have this problem with some of my equipment. 
I have a mavros link over Telem2 on an odroid and a normal GCS on Telem1.
Whenever mavros (0.18.5) is running, there seems to be a lot of arming/disarming and mode(hex) or Mode(0x000000c0) as described. 

However, when I stop the GCS on my laptop (mavproxy), then launch mavros, exit mavros, and reconnect mavproxy, it seems to function as expected.
I did a little bit of debugging, and it seems that mavros is broadcasting another heartbeat message, which is probably empty, causing the weird messages, as my GCS assumes the HB is from ardupilot.
I haven't been able to completely solve this, and I also have no idea why it stops when launching mavros from my laptop. Could this also be due to a ardupilot bug? (i'm on 3.4.2)

```
fwd msg 126 from chan1 on chan 2 sysid=-1 compid=-1
fwd HB from chan 1 on chan 2 from sysid=255 compid=0
...
fwd HB from chan 2 on chan 1 from sysid=1 compid=240
```

annesteenbeek, 2017-06-16:
I found a way to resolve this problem, maybe interesting for future people coming across this.
So this problem is caused by ardupilot forwarding the heartbeat message from Mavros to the GCS. Since the Mavros heartbeat message does not contain the state, the GCS freaks out. 
This can be solved by setting the Mavros systemID/componentID to be equal to that of the GCS (255 in most cases). 
The problem is also resolved if you run mavros on the channel and use the GCS bridge.

sulim-vladimir, 2017-09-08:
Having the same issue. @annesteenbeek , could you please explain in detail how to solve the issue? Thank you!