mavlink/mavros #648: How to do OFFBOARD to gps waypoint

skysingh, 2017-01-27:
First of all I am very impressed witb mavros. Nice job. I am running a Jetson TX1 connected over uart1 to a pixhawk. I can read mavros parameters but I am confused by two things even after spending a lot of time reading. Maybe somone can help a bit

1. Is there some example python node code to set mode. I need to switch to OFFBOARD.
2. I can read the gps fix but again I need an example of how to send it to the pixhawk. I am aware that I need to update better than 2hz to avoid failsafe timeout.

I am trying to build a rover that normally runs in AUTO and follows the preset way points. But when an obstruction is detected by the Jetson TX1 it should switch the pixhawk to GUIDED and send it a new temporary global gps waypoint. Once the temporary waypoint is reached the TX1 should switch the pixhawk back to auto. Thanks.

This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

_Please describe the problem, or desired feature_

### MAVROS version and platform

Mavros: ?0.18.4?
ROS: ?Kinetic?
Ubuntu: ?16.04?

### Autopilot type and version

[  ] ArduPilot
[  ] PX4

Version: ?3.7.1?

### Node logs

```
copy output of mavros_node. Usually console where you run roslaunch
```

### Diagnostics

```
place here result of:
rostopic echo -n1 /diagnostics
```

### Check ID

```
rosrun mavros checkid
```

TSC21, 2017-04-04:
I miss the point here? You want to work on OFFBOARD mode or in GUIDED mode? It seems to me that  you are messing the modes. From my evaluation, you will want a obstacle avoidance process that changes between modes. OFFBOARD won't send GPS waypoints, but only setpoints. So probably you want to navigate in auto and then, when an obstacle is detected, send setpoints to navigate in the local frame, till it gets again to a safe path to a waypoint that allows the rover to get to the next waypoint.

TSC21, 2017-04-07:
@AlexisTM do you have updates?

AlexisTM, 2017-04-07:
Got no update

TSC21, 2017-04-14:
@skysingh updates?

TSC21, 2017-06-09:
Closing as stalled. @skysingh please reopen if you still have problems.

closed 2017-06-09 by TSC21

debanik123, 2020-06-19:
https://github.com/Intelligent-Quads/iq_sim

debanik123, 2020-06-19:
@debanik123 see this tutorials and you will be good to go

debanik123, 2020-06-19:
https://github.com/PX4/avoidance?files=1