mavlink/mavros #649: libmavconn: Feature to select hardware flow control

vooon, 2017-01-28:
@sharvashish asked to support hw flow control (RTS/CTS lines).
That should be easy, but require API change (flag) and new URL proto (maybe `serial-hwfc://`?).

sharvashish, 2017-01-29:
Hi guys, 
I tried changing `		serial_dev.set_option(serial_port_base::flow_control(serial_port_base::flow_control::none));
`
to
`serial_dev.set_option(serial_port_base::flow_control(serial_port_base::flow_control::hardware));
`
in libmavconn/src/serial.cpp.

I could successfully compile the code as well. But I got a runtime error. I would post the error, next time I execute the same code.

vooon, 2017-02-24:
New proto was added, but i don't have tty device with support.
Or perhaps Boost.ASIO Linux backed do not support that option? Anyway it is hidden and do not require any changes of libmavconn code.

closed 2017-02-24 by vooon

mhkabir, 2018-03-08:
This was broken since implementation due to Boost bugs. Fix is here: https://github.com/mavlink/mavros/pull/971