mavlink/mavros #651: Imu corrdinate true direction

xiaopenghuang, 2017-02-08:
### Issue details

can anyone tell me the device imu raw direction , and mavros output imu direction .I am confused about it.

According to mpu6050 datasheet , the direction  is 
[image]

If  laid it flat on the ground , i think the acceleration is (0,0,-9.8), but the mavros output is (0,0,9.8).

use the positive direction define in mpu6050 datasheet , i draw the direction accrding to `/mavros/imu/data`

[image]

according to acceleration and gyro , i found the direction is reverse, why.

mhkabir, 2017-02-08:
Mavros follows the ENU convention according to ROS REP 103 (http://www.ros.org/reps/rep-0103.html). It is correct.

closed 2017-02-08 by mhkabir

xiaopenghuang, 2017-02-08:
@mhkabir can you tell me in pixhawk imu device direction , why when i laid flat ,get the acceleration (0,0,9.8) , not (0,0,-9.8),it really werid