mavlink/mavros #652: Mavros connection problem to APM

lyf44, 2017-02-08:
----

### Issue details
Hi, i am using Pixhack loaded with APM firmware with Rasberry Pi 3 as an onboard computer. 
It seems that I can't publish mavlink message (setpoint_raw) to APM using MAVROS and the only mavlink message I am able to get from APM is /mavros/state.  
Thanks a lot in advance

### MAVROS version and platform

Mavros: installed from binary pacakge. "sudo apt-get install ros-kinetic-mavros ros-kinetic-mavros-extras"
ROS: Kinetic
Ubuntu Mate: 16.04

### Autopilot type and version

ArduPilot 3.4

### Node logs
Command: `roslaunch mavros apm.launch fcu_url:=/dev/ttyAMA0:57600`
```
... logging to /home/ubuntu/.ros/log/309d55c8-edd9-11e6-9d3d-b827ebcda65a/roslaunch-ubuntu-8122.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.1.199  :34121/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 0.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 0.0
 * /mavros/fcu_url: /dev/ttyAMA0:57600
 * /mavros/gcs_url: 
 * /mavros/global_position/frame_id: fcu
 * /mavros/global_position/rot_covariance: 99999.0
 * /mavros/global_position/tf/child_frame_id: fcu_utm
 * /mavros/global_position/tf/frame_id: local_origin
 * /mavros/global_position/tf/send: False
 * /mavros/image/frame_id: px4flow
 * /mavros/imu/angular_velocity_stdev: 0.000349065850399
 * /mavros/imu/frame_id: fcu
 * /mavros/imu/linear_acceleration_stdev: 0.0003
 * /mavros/imu/magnetic_stdev: 0.0
 * /mavros/imu/orientation_stdev: 1.0
 * /mavros/local_position/frame_id: fcu
 * /mavros/local_position/tf/child_frame_id: fcu
 * /mavros/local_position/tf/frame_id: local_origin
 * /mavros/local_position/tf/send: False
 * /mavros/mission/pull_after_gcs: True
 * /mavros/mocap/use_pose: True
 * /mavros/mocap/use_tf: False
 * /mavros/plugin_blacklist: ['actuator_contro...
 * /mavros/plugin_whitelist: []
 * /mavros/px4flow/frame_id: px4flow
 * /mavros/px4flow/ranger_fov: 0.0
 * /mavros/px4flow/ranger_max_range: 5.0
 * /mavros/px4flow/ranger_min_range: 0.3
 * /mavros/safety_area/p1/x: 1.0
 * /mavros/safety_area/p1/y: 1.0
 * /mavros/safety_area/p1/z: 1.0
 * /mavros/safety_area/p2/x: -1.0
 * /mavros/safety_area/p2/y: -1.0
[... 288 lines cut ...]
    name: mavros: Battery
    message: No data
    hardware_id: /dev/ttyAMA0:57600
    values: 
      - 
        key: Voltage
        value: -1.00
      - 
        key: Current
        value: 0.0
      - 
        key: Remaining
        value: 0
```

### Check ID

```
OK. I got messages from 1:1.

---
Received 15 messages, from 1 addresses
sys:comp   list of messages
  1:1     0
```

vooon, 2017-02-08:
Check wiring! Looks like has broken Pi Tx -> Pix Rx.

lyf44, 2017-02-08:
Thanks @vooon , In the end I found out it was because we power the rasberry Pi and Pixhack separately and they do not share a common ground..... Stupid mistake. You can close this issue now.

closed 2017-02-08 by mhkabir

kungfrank, 2017-05-09:
Excuse me. I am a student working on the same project, but I got this error all the time. I reboot the Companion computer and APM many times, but still doesn't work(I have already configured baud rate of Telem1 in apmplanner).This problem confuse me for a week.

$ roslaunch mavros apm2.launch fcu_url:=/dev/ttyAMA0:57600

[ INFO] [1494332048.454563862]: GCS URL: udp://@
[ERROR] [1494332048.454735631]: serial0: receive: End of file
terminate called after throwing an instance of 'std::system_error'
  what():  Resource deadlock avoided
================================================================================REQUIRED process [mavros-1] has died!
process has died [pid 5719, exit code -6, cmd /opt/ros/kinetic/lib/mavros/mavros_node __name:=mavros __log:=/home/pi/.ros/log/a9df75fc-317b-11e7-ab17-b827ebf230df/mavros-1.log].
log file: /home/pi/.ros/log/a9df75fc-317b-11e7-ab17-b827ebf230df/mavros-1*.log
Initiating shutdown!
================================================================================

i use
ROS: Kinetic
Ubuntu Mate: 16.04
in raspberrypi3 with APM2.6

TSC21, 2017-05-09:
@kungfrank did you follow http://ardupilot.org/dev/docs/raspberry-pi-via-mavlink.html until the [Testing the connection topic](http://ardupilot.org/dev/docs/raspberry-pi-via-mavlink.html#testing-the-connection)? And please this in not related to the presented issue. Please open a new one

kungfrank, 2017-05-09:
@TSC21 I'm sorry.  I'm a newbie here and thanks for your suggestion. I will follow this tutorial first.