mavlink/mavros #653: Subcribing to /mavros/rc/out to get PWM inputs for System Identification of Tricopter

siddhu45, 2017-02-10:
Hello community,
in the past I've seeked help from you guys and you were really helpful. I try to contribute wherever I could. I'm working on a project "System Identification of a Tricopter" which has a Pixhawk onboard running the low level controller. I also have an Raspberry-Pi onboard which is my wireless link and I command trajectories and way-points via MAVROS. I'm at the end of my project and am performing System Identification of the tricopter [Talon Frame] and am seeking inputs [3 motor RPMs and 1 servo angle] and measurements [position(x, y, z), velocity(u, v, w), Euler angles (phi, theta, psi) & angular rates (p, q, r)] by commanding trajectories from ROS. I'm collecting this data by subscribing to MAVROS topics */mavros/local_position/pose* for position and Euler angles, */mavros/local_position/velocity* for velocity and angular rates and */mavros/rc/out* for Input PWM which I map to respective motor RPMs and servo tilt angle. I hope I'm making sense till now! 
After all this, once I've collected data, I use a System Identification [Output Error Method] MATLAB code which I wrote to evaluate the results, but I don't get any. The thing is forget the System Identification part, if I give the Inputs I collected to the system [mathematical model of tricopter which is already published by many authors] it doesn't generate any output (I mean it does, but it is garbage!). I use Runge-kutta 4th order numerical method for integration steps. What am I doing wrong here?

### Issue details

_1. The frequency at which MAVROS publishes data is ~50 Hz (approx 48/49 Hz) which may not be enough for me. I am sure Pixhawk's internal loop is running at quite higher frequency. Is there any way I can increase the subscribing frequency of MAVROS (if that is is going to help me at all)
2. Performing System Identification with this inputs or collecting data in this way does it make sense?? As the Tricopter is an unstable system is there something else I should be doing? The thing is, I tried giving a simulated input (I generate the Inputs using simulation in the same range as real inputs) and that does give me an output and even the System Identification code works fine. But I want to do all this in real time. I'm pretty sure my matlab code is correct because I have tested it on other systems and it gives me results. Still I'm not able to locate the problem I am at. Am I using the correct model (I've tried various authors and different combinations). 
3. Can any one please help me solve my issue as I'm stuck here for quite a long time now (a month and a half to be correct). I look at my code everyday and try to figure out what went wrong but I might need a third person view with a new vision to maybe find out my problem. If anyone's interested I will happy to share my codes with you and they are generalised so can used for any system. All you ned to do is change the model equations to your systems and the code will work all along. Any help or guidance on any of these things to solve my problem is really appreciated._

If you need any further info do let me know. I hope I was clear. Thank you so much! Really looking forward to the solution

### MAVROS version and platform

Mavros: ?0.18.4?
ROS: ?Indigo?
Ubuntu: ?14.04?

mhkabir, 2017-02-10:
You need to use the onboard logs which are recorded at full rate. Mavros running over a serial link won't be sufficient for this.

siddhu45, 2017-02-10:
Thank you so much for your response but will I be able to get all the required parameters from the logs?

mhkabir, 2017-02-10:
Yes, we already do system identification for our experimental setups. https://github.com/dronecrew/px4tools/blob/master/px4tools/logsysid.py

closed 2017-02-10 by mhkabir

siddhu45, 2017-02-11:
Thank you for that! I collected the data from the log files as you suggested. One thing which I observed from your sysid code is that you resample the data at 1ms for performing the system identification. Why is that necessary? I am still learning the process and understanding how you implement it. But as far as I analysed the log files, the data recorded is ~133 Hz if I'm not wrong. So why is the resampling necessary. Also, if you could answer that if it is possible to increase the subscribing default frequencies of the mavros topics. Thanks for all your help.