mavlink/mavros #654: REQUIRED process [mavros-1] has died!

cs147, 2017-02-13:
----

### Issue details

Can't launch mavros on an odroid-xu4 connected to pixhawk mini

### MAVROS version and platform

Mavros: 0.17.4
ROS: Indigo
Ubuntu: 14.04

### Autopilot type and version

PX4

Version: ?3.7.1?

### Node logs

```

$ roslaunch px4-mavros.launch 
... logging to /home/quadlab/.ros/log/a7af7b2e-f23d-11e6-ac47-001e063056a5/roslaunch-hedwig-4225.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://localhost:56493/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS                                                                                                                                                   [97/679]
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: ROLL_180
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: ROLL_180
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
=
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser                                                                                           [79/679]
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: ROLL_180
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: ROLL_180
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /mavros/fcu_url: serial:///dev/tty...
 * /mavros/gcs_url: 
 * /mavros/global_position/frame_id: fcu
 * /mavros/global_position/rot_covariance: 99999.0
 * /mavros/global_position/tf/child_frame_id: fcu_utm
 * /mavros/global_position/tf/frame_id: local_origin
 * /mavros/global_position/tf/send: False
 * /mavros/image/frame_id: px4flow
 * /mavros/imu/angular_velocity_stdev: 0.000349065850399
 * /mavros/imu/frame_id: fcu                                                                                                                                 [60/679]
 * /mavros/imu/linear_acceleration_stdev: 0.0003
 * /mavros/imu/magnetic_stdev: 0.0
 * /mavros/imu/orientation_stdev: 1.0
 * /mavros/local_position/frame_id: fcu
 * /mavros/local_position/tf/child_frame_id: fcu
 * /mavros/local_position/tf/frame_id: local_origin
 * /mavros/local_position/tf/send: False
 * /mavros/local_position/tf/send_fcu: False
 * /mavros/mission/pull_after_gcs: True
 * /mavros/mocap/use_pose: True
 * /mavros/mocap/use_tf: False
 * /mavros/plugin_blacklist: ['safety_area', '...
 * /mavros/plugin_whitelist: []
 * /mavros/px4flow/frame_id: px4flow
 * /mavros/px4flow/ranger_fov: 0.0
 * /mavros/px4flow/ranger_max_range: 5.0
 * /mavros/px4flow/ranger_min_range: 0.3
 * /mavros/safety_area/p1/x: 1.0
 * /mavros/safety_area/p1/y: 1.0
 * /mavros/safety_area/p1/z: 1.0
 * /mavros/safety_area/p2/x: -1.0
 * /mavros/safety_area/p2/y: -1.0
 * /mavros/safety_area/p2/z: -1.0                                                                                                                            [37/679]
 * /mavros/setpoint_accel/send_force: False
 * /mavros/setpoint_attitude/reverse_throttle: False
 * /mavros/setpoint_attitude/tf/child_frame_id: attitude
 * /mavros/setpoint_attitude/tf/frame_id: local_origin
 * /mavros/setpoint_attitude/tf/listen: False
 * /mavros/setpoint_attitude/tf/rate_limit: 10.0
 * /mavros/setpoint_position/tf/child_frame_id: setpoint
 * /mavros/setpoint_position/tf/frame_id: local_origin
 * /mavros/setpoint_position/tf/listen: False
 * /mavros/setpoint_position/tf/rate_limit: 50.0
 * /mavros/startup_px4_usb_quirk: True
 * /mavros/sys/disable_diag: False
 * /mavros/sys/min_voltage: 10.0
 * /mavros/target_component_id: 1
 * /mavros/target_system_id: 1
 * /mavros/tdr_radio/low_rssi: 40
 * /mavros/time/time_ref_source: fcu
 * /mavros/time/timesync_avg_alpha: 0.6                                                                                                                      [19/679]
 * /mavros/vibration/frame_id: vibration
 * /mavros/vision_pose/tf/child_frame_id: vision
 * /mavros/vision_pose/tf/frame_id: local_origin
 * /mavros/vision_pose/tf/listen: False
 * /mavros/vision_pose/tf/rate_limit: 10.0
 * /mavros/vision_speed/listen_twist: False
 * /rosdistro: indigo
 * /rosversion: 1.11.9

NODES
  /
    mavros (mavros/mavros_node)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[mavros-1]: started with pid [4244]
[ INFO] [1487028067.646523966]: FCU URL: serial:///dev/ttyACM0:9600
[ INFO] [1487028067.647170925]: device: /dev/ttyACM0 @ 9600 bps                                                                                               [0/679]
[ INFO] [1487028067.648900258]: GCS bridge disabled
[ INFO] [1487028067.822868591]: Plugin 3dr_radio loaded and initialized
[ INFO] [1487028067.833555425]: Plugin actuator_control loaded and initialized
[ INFO] [1487028067.838040133]: Plugin altitude loaded and initialized
[ INFO] [1487028067.856044883]: Plugin command loaded and initialized
[ INFO] [1487028067.879877258]: Plugin ftp loaded and initialized
[ INFO] [1487028067.908572425]: Plugin global_position loaded and initialized
[ INFO] [1487028067.910993258]: Plugin hil_controls loaded and initialized
[ INFO] [1487028067.911143133]: Plugin image_pub blacklisted
[ INFO] [1487028067.937327175]: Plugin imu_pub loaded and initialized
[ INFO] [1487028067.964244508]: Plugin local_position loaded and initialized
[ INFO] [1487028067.968569800]: Plugin manual_control loaded and initialized
================================================================================REQUIRED process [mavros-1] has died!
process has died [pid 4244, exit code -11, cmd /opt/ros/indigo/lib/mavros/mavros_node __name:=mavros __log:=/home/quadlab/.ros/log/a7af7b2e-f23d-11e6-ac47-001e06305$a5/mavros-1.log].
log file: /home/quadlab/.ros/log/a7af7b2e-f23d-11e6-ac47-001e063056a5/mavros-1*.log
Initiating shutdown!
================================================================================
[mavros-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

```

### Diagnostics

```
header: 
  seq: 1
  stamp: 
    secs: 1487028067
    nsecs: 896730675
  frame_id: ''
status: 
  - 
    level: 0
    name: mavros: GPS
    message: Node starting up
    hardware_id: ''
    values: []

```

### Check ID

```
Received 1 messages, from 1 addresses
sys:comp   list of messages
  1:1     0
```

vooon, 2017-02-14:
I suppose node fall with SIGSEGV or SIGABRT, try to run under strace, gdb.
But first try master built from source.

cs147, 2017-02-14:
@vooon I built mavros from source already. Is that what you mean?

vooon, 2017-02-14:
I mean MASTER, eg 0.18.x.

cs147, 2017-02-14:
@vooon Ok, will try that after the gdb thing.
How can I run gdb with `roslaunch mavros px4.launch`? Where does the launch-prefix go? Trying to follow [this](http://wiki.ros.org/roslaunch/Tutorials/Roslaunch%20Nodes%20in%20Valgrind%20or%20GDB), slightly confused.

cs147, 2017-02-14:
@vooon 

I changed [this line](https://github.com/mavlink/mavros/blob/master/mavros/launch/px4.launch#L11) in the launch file to be:
`<include file="$(find mavros)/launch/node.launch" launch-prefix="gdb -ex run --args">`
and I get the same error; it seems it never makes it to the node, or gdb might do something?

vooon, 2017-02-15:
You may do launch manually:
1. `$ roscore` (leave it running)
2. `$ rosparam load px4_config.yaml /mavros` (not really necessary)
3. `$ rosrun --prefix "strace -fy" mavros mavros_node _fcu_url:=/dev/ttyACM0`

cs147, 2017-02-15:
@vooon 

I tried to build mavros on master (0.18.6) from source, as per [these instructions](https://github.com/mavlink/mavros/blob/master/mavros/README.md#source-installation).
Doing `catkin build` on the workspace **never** results in a successful build. However, blowing away that catkin workspace and starting a new one from scratch, and building my repo (which depends on mavros) does indeed work.
Checking `package.xml` of mavros in `/opt/ros/indigo/share` reveals it is 0.17.4 still, and yet the code builds
Now, this time, when I launch `roslaunch mavros px4.launch`, it works with no error.
Do you have any insight into this weirdness?

vooon, 2017-02-15:
You expect that `build` will install it into /opt? Nope, source build is in workspace.
And i recommend to remove `ros-indigo-mavros*`, they may conflict with master.

cs147, 2017-02-23:
@vooon When I clone the master branch (0.18.6) and then do `cakin_make` I get this error:
```
/home/quadlab/catkin_ws/src/mavros/libmavconn/include/mavconn/mavlink_dialect.h:23:42: fatal error: mavlink/v2.0/common/common.hpp: No such file or directo
ry
 #include <mavlink/v2.0/common/common.hpp>
```

but when install the package `sudo apt-get install ros-indigo-mavros*` it's 0.17.4 and I don't have this compilation error...

vooon, 2017-02-24:
You should follow [installation instructions](https://github.com/mavlink/mavros/blob/master/mavros/README.md#source-installation). Use `--upstream-development` variation with `--rosdistro kinetic`.

bhairavmehta95, 2017-02-27:
Should someone who is using ROS Indigo still be specifying ```--rosdistro kinectic``

vooon, 2017-02-27:
Yes, if that someone want latest version.

Because &lt;distro&gt; will use branch:
- Hydro: hydro-devel
- Indigo: indigo-devel
- Jade: indigo-devel
- Kinetic: master

TSC21, 2017-04-04:
@cs147 is this solved to you?

TSC21, 2017-04-14:
Closing this as there is no updates. Reopen it if needed.

closed 2017-04-14 by TSC21