mavlink/mavros #656: Guided mode not working

skysingh, 2017-02-14:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

I am using ROS on JetsonTX1 to communicate over serial to pixhawk tele2. Mavros works fine. I can get data. I have also been able to change mode MANUAL AUTO LEARNING GUIDED using service and I see the mode change on MP. So far so good. Now I ask for GUIDED mode and it switches in MP. Now I publish the new lat/lon  to mavros/setpoint_raw/global and I verify that it is publishing correctly using echo. I am publishing at 10hz which I think is OK. BUT I do not seem to be able to get the new lat/lon correctly into the pixhawk - it gives me the distance to WP as 0. I do not any way to see what it thinks the lat/lon of the new WP.

Anybody done this ? Any help appreciated. I am getting close but this hurdle needs to be crossed.

_Please describe the problem, or desired feature_

### MAVROS version and platform

Mavros: ?0.18.4?
ROS: ?Kinetic?
Ubuntu: ?16.04?

### Autopilot type and version
Pixhawk 3DR
Ardurover 3.0.1

[  ] ArduPilot
[  ] PX4

Version: ?3.7.1?

### Node logs

```
copy output of mavros_node. Usually console where you run roslaunch
```

### Diagnostics

```
place here result of:
rostopic echo -n1 /diagnostics
```

### Check ID

```
rosrun mavros checkid
```

skysingh, 2017-02-14:
Some diagnostics that might help
Ros says FCU executing command ID#16   (which is go to nav)
Then CMD unexpected command 11, result 0
Then FCU: reached destination

vooon, 2017-02-15:
Please fill template, place node output (log) into \`\`\` block. Diag & Check ID not needed yet.

skysingh, 2017-02-18:
was away so did not reply for a while. Here is log output with my comments
-------------
Current postion lat/lon
40.7053276000 -74.7096261000
new WP to be GUIDED to....   (10 meters away)
40.7053641666 -74.7097343387
Mode is set to guided ...
[INFO] [1487424823.704678]: /vinorobo_node mode is: GUIDED

-------------   LOG FROM ROS ----

[ INFO] [1487424822.264716320]: FCU: Executing command ID #16
[ WARN] [1487424822.742772239]: CMD: Unexpected command 11, result 0
[ INFO] [1487424822.763489778]: FCU: Reached destination
[ INFO] [1487424822.763650922]: WP: reached #0

Looks like the mode is being set to GUIDED
The new WP is sent
But then FCU thinks is already there ?

I was thinking a workaround would be to modify the mission on the fly. For example say you are heading to WP#3. Now you want to go instead to a new WP. IOriginally, was going to switch to GUIDED, get to the new WP and then resume the mission with AUTO. But what if we insert the new WP into the mission list and send the whole new mission to the FCU. Then it it should go to the inserted new WP and when this is reached continue the mission. This would eliminate the need for GUIDED. I saw your mission.py example but I have not had a chance to try it. What do you think ?

AlexisTM, 2017-03-02:
try to use ``` to format your code...

TSC21, 2017-04-04:
@skysingh updates?

TSC21, 2017-04-14:
Closing this as there's no feedback. Please reopen if needed.

closed 2017-04-14 by TSC21