mavlink/mavros #657: Send custom message with mavros

pietrop88, 2017-02-14:
Hi all,
I want to send a custom message to the px4 firmware through mavros.
This is the content of the `led_color_ref.cpp` file in the */src/plugins/* directory of the mavros package:
```
#include <mavros/mavros_plugin.h>
#include <pluginlib/class_list_macros.h>

#include <mavros_msgs/LedColor.h>

namespace mavplugin {
/**
 * @brief led color and target reference plugin
 *
 * Send led color and led target to uav.
 */
class LedColorRefPlugin : public MavRosPlugin{
public:
	LedColorRefPlugin() :
		lc_nh("~led_color"),
		uas(nullptr)
	{};

	void initialize(UAS &uas_)
	{
		uas = &uas_;
		led_color_ref_sub = lc_nh.subscribe("", 10, &LedColorRefPlugin::led_color_ref_cb, this);
	}

	const message_map get_rx_handlers() {
		return { /* Rx disabled */ };
	}

private:
	ros::NodeHandle lc_nh;
	UAS *uas;

	ros::Subscriber led_color_ref_sub;

	/* -*- callbacks -*- */
	void led_color_ref_cb(const mavros_msgs::LedColor::ConstPtr &req) {

		ros::Time stamp = req->header.stamp;

		mavlink_message_t msg;
		mavlink_msg_led_color_ref_pack_chan(UAS_PACK_CHAN(uas), &msg,
				stamp.toNSec() / 1000000,
				req->target,
				req->R,
				req->G,
				req->B
				);
		UAS_FCU(uas)->send_message(&msg);
	}
};
};	// namespace mavplugin

PLUGINLIB_EXPORT_CLASS(mavplugin::LedColorRefPlugin, mavplugin::MavRosPlugin)
```
Then I added the plugin to the `CMakeLists.txt` file and to the `mavros_plugins.xml` file.
When I launch the mavros package I am able to see the new **led_color** topic.

Then I added the code to convert the new mavlink message to a uORB message. 
I followed [this guide](https://dev.px4.io/mavlink-messaging.html#receiving-custom-mavlink-messages):

I added
```
void handle_message_led_color_ref(mavlink_message_t *msg);
```
and 
```
orb_advert_t _led_color_ref_pub;
```
to the `mavlink_receiver.h` file.

In the `mavlink_receiver.cpp` file I implemented the `handle_message_led_color_ref` function and I added:
```
case MAVLINK_MSG_ID_LED_COLOR_REF:
      handle_message_led_color_ref(msg);
      break;
```
I am able to compile all the components but when I send a ROS message to the  **led_color** topic my code never goes in the `case MAVLINK_MSG_ID_LED_COLOR_REF:`.

The problem is between mavros and the px4 code that convert the mavlink message to a uORB message but I don't if the error is the mavros and thus the mavlink message is never sent or the problem is in the px4 side and the mavlink is (for some reason that i don't know) rejected.

Can you help me?

vooon, 2017-02-15:
You cannot to subscribe to "" topic! Use `nh("~")`, and `subscribe("led_color",...)`.
Next: you don't need to modify mavros sources, place your plugin in separate package (see mavros_extras for example).
And you use outdated version, i recommend to switch 0.18.x (plugin a bit different, but not much).

Where you placed your message? common.xml? Mavlink may silently drop unknown message (e.g. msg crc mismatch).

vooon, 2017-02-15:
I assume that firmware do not use your dialect (PX4 uses common.xml only, but you may include your file).

pietrop88, 2017-02-15:
I think that my firmware contains my mavlink dialect because I have these lines in the `mavlink_receiver.cpp` file:
```
case MAVLINK_MSG_ID_LED_COLOR_REF:
      handle_message_led_color_ref(msg);
      break;
```
and I am able to compile the firmware. If the firmware does not contain my dialect I think that the compiler is to able to find `MAVLINK_MSG_ID_LED_COLOR_REF`.

Is it right?

pietrop88, 2017-02-15:
With the same mavros plugin code I try to send a message that belongs to the *common* dialect:
instead
```
mavlink_msg_led_color_ref_pack_chan(UAS_PACK_CHAN(uas), &msg,
				stamp.toNSec() / 1000000,
				req->target,
				req->R,
				req->G,
				req->B
);
```
I wrote:
```
mavlink_msg_ping_pack_chan(UAS_PACK_CHAN(uas), &msg,
		stamp.toNSec() / 1000000,
		0,0,0
		);
```
that sends a **PING** message.
In this case px4 receives the message.

I think the problem is:

> Mavlink may silently drop unknown message (e.g. msg crc mismatch).

Can you help me?

vooon, 2017-02-15:
As i say, fw don't know your dialect. Fact that code compiles not guarantee that message crc array has entry of your one.
I assume that include your message set into common is easier, look at uAvionix.xml.

pietrop88, 2017-02-15:
I included my message in common.xml and it works. Tomorrow i will look at uAvionix.xml.
Thanks

pietrop88, 2017-02-16:
I looked at [uAvionix.xml](https://github.com/mavlink/c_library_v1/blob/master/message_definitions/uAvionix.xml).
My dialect is simpler but it includes the `common.xml` file.
This is the content of my dialect:
```
<?xml version="1.0"?>
<mavlink>
  <include>common.xml</include>
       <enums>
	</enums>
	<messages>
		<message id="150" name="LED_COLOR_REF">
			<description>
                                   This message encodes the led target and the led color.
                        </description>
                        <field type="uint32_t" name="time_boot_ms">Timestamp in milliseconds since system boot.</field>
			<field type="uint8_t" name="target">The led target.</field>
			<field type="uint8_t" name="R">Red.</field>
			<field type="uint8_t" name="G">Green.</field>
			<field type="uint8_t" name="B">Blue.</field>
		</message>
	</messages>
</mavlink>
```
I am not able to see the error in the definition of the dialect ( I followed [this guide](http://qgroundcontrol.org/mavlink/create_new_mavlink_message)).

vooon, 2017-02-16:
`common.xml` includes `uAvionix.xml`!

closed 2017-02-17 by pietrop88

lamhung81, 2018-02-28:
Hi pietrop88, could you please tell me how to create a new custom message with mavros to send from mavros to pixhawk px4?