mavlink/mavros #659: CMAKE ERROR  error: <type 'exceptions.ValueError'>: common dialect not listed!

yajingwang94, 2017-02-20:
Hi, I git the package mavlink/mavros and mavlink/mavros to support the catkin_make process of the [rotors_simulator](https://github.com/ethz-asl/rotors_simulator) package, but it reports the following error which I have never met:
[image]

I checked the related files mentioned above, finding that it failed to generate the mavros/libmavconn/include/mavconn/mavlink_dialect.h file, which is empty. Is there any files or package I have missed or anything I need to do?
Please give me some instructions. 
Thanks.

Mavros: 0.18.6
ROS: Jade
Ubuntu: 14.04

vooon, 2017-02-20:
https://github.com/mavlink/mavros/tree/master/mavros#source-installation

yajingwang94, 2017-02-21:
Thanks a lot. I followed the instruction in this page, but always failed at the step: `rosdep install --from-paths src --ignore-src -y`
It reports :
[image]
I returened to make my package again, and the error turns to be:
[image]

Dose this mean that I failed to install the mavros package? I noticed that the `mavros_msgs `package should be included by` mavros` package. Is there anything I have missed?

yajingwang94, 2017-02-21:
Thanks very much for your help @vooon . I have solved the problem above.

closed 2017-02-21 by yajingwang94