mavlink/mavros #661: Add option to disable timesync and pass on timestamp

mhkabir, 2017-02-24:
For many situations where autopilot stack is running on same board (e.g Snapdragon Flight, Raspberry Pi with Navio), the time-stamp of MAVLink messages does not need to be synchronised, since the time bases are the same.

@vooon What'd be the best way to implement this "passthrough" mode for timesync and not perform the sync?
FYI @rkintada @Seanmatthews

vooon, 2017-02-24:
Blacklist `sys_time`.

mhkabir, 2017-02-24:
No, that won't work.

mhkabir, 2017-02-24:
Actually. Scratch that. It should work. 

@Seanmatthews can you try it out?

mhkabir, 2017-03-07:
I pushed a preliminary branch here : https://github.com/mavlink/mavros/tree/uasys-upstreaming

vooon, 2017-03-09:
I think keeping it as PR little better.
Now i has limited debugging, because 16.10 not supported by Kinetic, so for releasing i use ROS Docker, which is not so handy when you need to debug something.

vooon, 2017-03-19:
@mhkabir i merged to upstream, but w/o much testing. Can you test?

TSC21, 2017-04-04:
@mhkabir is this working ok to you now?

closed 2017-05-04 by mhkabir