mavlink/mavros #662: Mavros Extras

bhairavmehta95, 2017-02-24:
### Issue details

I noticed that by running ```apt-get install ros-indigo-mavros-extras```, I do not get files such as:
mocap_pose_estimate.cpp

I do not want to build mavros from source, as that aptitude install works; is there a way I can just install mavros_extras from source? Or is there a way I can get all the source files and use them with my current mavros installation.

### MAVROS version and platform

Mavros: 0.18.4
ROS: Indigo
Ubuntu: 14.04

vooon, 2017-02-24:
Binary packages do not contain sources. You may download tar and copy extras to your ws src, that should work, but may be more buggy than "official source install" method.

bhairavmehta95, 2017-02-27:
Thank you for answering @vooon .

I have copied mocap_pose_estimate.cpp and built a node out of just that file (that is the only file from extras that I need), but I am having trouble understanding how to initialize the uas pointer for the data to be correctly sent to the FCU.

mhkabir, 2017-02-27:
Why are you even going to all that trouble? Just follow our source install guides and compile the package. 

I don't see the point of supporting a half-baked setup. If you want support, use the full source build.

bhairavmehta95, 2017-02-28:
Thanks, full source build worked for me now, closing this issue.

closed 2017-02-28 by bhairavmehta95