mavlink/mavros #663: Support ULog streaming for PX4

mhkabir, 2017-02-24:
We should support new FCs (e.g Intel Aero) which can only stream log data to the onboard computer, because they have no sdcard.

Reference :
https://github.com/PX4/Firmware/blob/master/Tools/mavlink_ulog_streaming.py

Till this is implemented, one can use the above script by adding a GCS proxy in mavros, and have the script connect to it.

cjeppe, 2017-02-27:
I am trying to follow your instruction as a solution to do logging while also having a mavros node on a Intel Aero drone. Can you please elaborate slightly more on your suggestion. I am unsure about how to add this GCS proxy in mavros.

cjeppe, 2017-03-01:
Finally managed to start a mavros with a GCS proxy with the command
`roslaunch mavros px4.launch fcu_url:=udp://@192.168.1.1:14550 gcs_url:=udp://127.0.0.1:14558@127.0.0.1:14610`
I am receiving all the data to mavros but I am not able to use the script mentioned by mhkabir
[https://github.com/PX4/Firmware/blob/master/Tools/mavlink_ulog_streaming.py](url)
I am able to make the script connect to the GCS proxy and the script is receiving all the messages meant for the mavros node but it never receives a COMMAND_ACK after the MAV_CMD_LOGGING_START is sent. This means the script is timing out after 4 seconds and terminating. Any ideas why the script never receives a COMMAND_ACK?
As a side note I can mention that I am able to use the mavlink_shell.py to enter the NuttX shell through the GCS proxy.

mindThomas, 2017-04-27:
Will this work with Mavros version 0.17.5? And if not, what is it that causes it not to work?

TSC21, 2017-04-27:
> Will this work with Mavros version 0.17.5? And if not, what is it that causes it not to work?

It's not implemented on MAVROS side yet.

TSC21, 2017-04-27:
Also it is a feature to include in next release. 0.17.5 will not receive more updates.

mindThomas, 2017-04-27:
I know it is not a Mavros feature, but if we enable the GCS proxy in Mavros 0.17.5 will we be able to start the Ulog script? Or will it be incompatible/not work due to incorrect Mavlink version selection.

TSC21, 2017-04-27:
> I know it is not a Mavros feature, but if we enable the GCS proxy in Mavros 0.17.5 will we be able to start the Ulog script? Or will it be incompatible/not work due to incorrect Mavlink version selection.

According to @mhkabir, it should (at least for the current implementation), but I never tried it myself. I don't know if this ULog is dependent on Mavlink 2.0

mindThomas, 2017-04-27:
Correct. Please also see this post: https://github.com/intel-aero/meta-intel-aero/issues/127#issuecomment-297643456

TSC21, 2017-04-27:
> Correct. Please also see this post: intel-aero/meta-intel-aero#127 (comment)

Yep. Basically we update the mavros version on `meta-ros` so this to happen. Is the only way it can work properly.

dgonano, 2019-02-21:
Sorry to awake an old thread but I guess this is a useful place to put it.
Can anyone say if serial baud of 921600 is reasonable for ulog streaming, MAVROS connection with vision messages, and forwarding to QGC?
Does PX4 have a serial usage monitor?

vooon, 2019-02-25:
@dgonano in `/diagnostics` you can see current data rate. On 921600 baud rate your throughput is 92160 Byte/s.

MaximeAubanel, 2020-08-31:
Is this still an ongoing task ?

As of today, if we are using `/dev/ttyPixhawk` to connect Mavros then how can we use the script at the same time ?

vooon, 2020-08-31:
@MaximeAubanel it is not implemented. You can use mavlink-router or script above.