mavlink/mavros #664: How to use distance_sensor? send distance Sensor data to FCU for use as altitude

Ahrovan, 2017-02-27:


AlexisTM, 2017-03-02:
You can unblacklist the plugin in the yaml configuration file

```
roscd mavros
cd launch
```

AlexisTM, 2017-03-02:
Also, if you buy a teraranger, you can connect directly to the pixhawk (:

Ahrovan, 2017-03-02:
plugin_blacklist:
# common
- actuator_control
- ftp
- safety_area
- setpoint_attitude
# extras
- image_pub
- 'mocap_*'
- px4flow
- 'vision_*'
- distance_sensor
- vibration
- altitude

plugin_whitelist: []
#- 'sys_*'

how to move - distance_sensor to plugin_whitelist?
 what means 'sys_*' ? after delete that occur error !

mhkabir, 2017-03-02:
Just remove it from blacklist. Delete the line. Don't change anything else.

closed 2017-03-02 by mhkabir

realdealneil, 2018-01-31:
It's probably bad form to add to a closed topic, but I learned something and wanted to share.  I have a lidarlite installed on my pixhawk, but I want to access it over mavros.  I checked that the mavlink packet was getting sent (sure enough, it is according to mavlink inspector in qgroundcontrol).  Then, I ran mavros.  I un-blacklisted the distance_sensor in the plugins list, and nothing happened.  (This is on ros-indigo on Ubuntu 14.04 on an Nvidia TK1 (parrot slam-dunk board)).

I happened to have built mavros on this board, so I tried sourcing my built version of mavros and unblacklisting the distance_sensor.  This time, 4 new topics showed up under /mavros/distance_sensor.  The /mavros/distance_sensor/lidarlite_pub was NOT doing anything, but the one called /mavros/distance_sensor/hrlv_ex4_pub showed the distance reading from the lidarlite.  I'm in business!  (If anyone knows why it shows up under a weird topic name, let me know).

AlexisTM, 2018-02-01:
@realdealneil 
Maybe you have to configure it?

The plugin configurations are defaulted: 

https://github.com/mavlink/mavros/blob/30ecc2087f58f2a2dc916308a372ceb9d4fab89b/mavros/launch/px4_config.yaml#L132-L154

realdealneil, 2018-02-01:
I figured it out.  I had installed the apt-get package for ros-indigo-mavros but not ros-indigo-mavros-extras.  I installed that and changed the id for lidarlite_pub from 1 to 0 (and set hrlv_ex4_pub to 1) and now it's being published when I use the built-in mavros package.  Thanks for your help @AlexisTM.

TSC21, 2018-02-01:
> I figured it out. I had installed the apt-get package for ros-indigo-mavros but not ros-indigo-mavros-extras.

That's not part of the solution I guarantee you.

> lidarlite_pub from 1 to 0 (and set hrlv_ex4_pub to 1)

Why don't you change the naming instead? If the LIDARLite is IDed as 0, just change https://github.com/mavlink/mavros/blob/master/mavros/launch/px4_config.yaml#L133 to `lidarlite_pub` and delete https://github.com/mavlink/mavros/blob/30ecc2087f58f2a2dc916308a372ceb9d4fab89b/mavros/launch/px4_config.yaml#L140-L154 which aren't doing nothing there. What you find on the config is an example, not a standard.

TSC21, 2018-02-01:
Also I am missing to update the current standard for orientation, where a downward facing sensor is now `PITCH_270` and not `ROLL_180` (https://github.com/PX4/Firmware/blob/master/msg/distance_sensor.msg#L18), where `MAV_SENSOR_ROTATION_NONE` is facing forward.

ChunJyeBehBeh, 2018-03-09:
plugin_blacklist:
# common
- actuator_control
- ftp
- safety_area
- hil
# extras
- altitude
- debug_value
- image_pub
- mocap_pose_estimate
- px4flow
- vibration
- 'vision_*'

plugin_whitelist: []
#- 'sys_*'

--------------------------------------------------------------------------
Does this mean that I have unblacklisted the distance sensor from my blacklist? So that my teraranger one that connect to pixfalcon has published to mavros already? How can I check it whether publish already or not?

TSC21, 2018-03-10:
Check above how to configure the plugin. I would check what is the sensor ID being published on `DISTANCE_SENSOR` msg on QGC/MissionPlanner so you can know what to put on the `id:` parameter.

TSC21, 2018-03-10:
> How can I check it whether publish already or not?

`rostopic echo /mavros/distance_sensor/<name_of_the_pub_you_set_on_config.yaml>`

ChunJyeBehBeh, 2018-03-13:
|| so you can know what to put on the id: parameter.
Where should I make this change? What code should I run for the teraranger one?

After I run "rostopic echo /mavros/distance_sensor/hrlv_ez4_pub" , there is nothing come out.

AlexWUrobot, 2022-12-20:
I get the similar issue and cannot the distance_sensor rostopic from the simulation.
However, I can get the distance value from the QGroundcontroller software.
I noticed. there are similar issue:
https://github.com/mavlink/mavros/issues/1387
https://github.com/mavlink/mavros/issues/745

AlexWUrobot, 2022-12-20:
I find the solution.
Just comment “distance sensors” and “range finders” on **px4_pluginlists.yaml** .

[image]

Just editing 
https://discuss.px4.io/t/how-to-read-distance-sensor-data-from-mavros-topic/21944