mavlink/mavros #665: yaw rate control not working

truhoang, 2017-02-27:
### Issue details

I am trying to send yaw rate commands to the 3DR Solo through setpoint_raw/local or setpont_raw/attitude. I have successfully send velocity commands using setpoint_raw/local and tried sending yaw rate commands to the other but nothing worked. Also tried non-raw setpoints but no response from the UAV. Please help.

### MAVROS version and platform

Mavros: mavros-indigo-devel
ROS: Indigo
Ubuntu: 14.04
Pixhawk firmware: APM:Copter3.3

Here is a snippet of my code and what I have tried:
```
command_pub = nh.advertise<mavros_msgs::PositionTarget>("/mavros/setpoint_raw/local", 10);
angular_pub = nh.advertise<mavros_msgs::AttitudeTarget>("/mavros/setpoint_raw/attitude", 10);
// tried publishing yaw rate commands to /mavros/setpoint_attitude/cmd_vel
//angular_pub = nh.advertise<geometry_msgs::TwistStamped>("/mavros/setpoint_attitude/cmd_vel", 10);
.
.
.
mavros_msgs::PositionTarget ctrl_msg;
mavros_msgs::AttitudeTarget angular_msg;
ctrl_msg.header.stamp = ros::Time::now();
ctrl_msg.coordinate_frame = mavros_msgs::PositionTarget::FRAME_BODY_NED;
ctrl_msg.type_mask = mavros_msgs::PositionTarget::IGNORE_PX |
                                     mavros_msgs::PositionTarget::IGNORE_PY |
                                     mavros_msgs::PositionTarget::IGNORE_PZ |
                                     mavros_msgs::PositionTarget::IGNORE_AFX |
                                     mavros_msgs::PositionTarget::IGNORE_AFY |
                                     mavros_msgs::PositionTarget::IGNORE_AFZ |
                                     mavros_msgs::PositionTarget::IGNORE_YAW;
                                     // uncomment if publishing to setpoint_raw/attitude
                                     // mavros_msgs::PositionTarget::IGNORE_YAW_RATE;
ctrl_msg.velocity.x = u;
ctrl_msg.velocity.y = v;
ctrl_msg.velocity.z = w;
ctrl_msg.yaw_rate = yaw_rate; // commented out if publishing to setpoint_raw/attitude

// /mavros/setpoint_raw/attitude
 angular_msg.type_mask = mavros_msgs::AttitudeTarget::IGNORE_ROLL_RATE | 
                                        mavros_msgs::AttitudeTarget::IGNORE_PITCH_RATE |
                                        mavros_msgs::AttitudeTarget::IGNORE_ATTITUDE;
angular_msg.body_rate.z = yaw_rate;
angular_msg.thrust = 0.5;
.
.
.
command_pub.publish(ctrl_msg);
angular_pub.publish(angular_msg);

````

vooon, 2017-02-28:
https://github.com/ArduPilot/ardupilot/blob/Copter-3.3/ArduCopter/GCS_Mavlink.cpp#L1476-L1550
As you see, it is unsupported in 3.3.

truhoang, 2017-02-28:
@vooon Do you have any recommendation for the firmware version so that setpoint_raw/local works with yaw rate commands as well?

vooon, 2017-02-28:
Latest version not support as well, but it support attitude target.

https://github.com/ArduPilot/ardupilot/blob/master/ArduCopter/GCS_Mavlink.cpp#L1534

truhoang, 2017-02-28:
@vooon one more question: it says that attitude and thrust must be used but I only want to send yaw rate commands. I understand the thrust value is to control the altitude during attitude change. Do I just not update the other variables to setpoint_raw/attitude?

vooon, 2017-03-01:
Nope, this didn't work, you should command attitude + thrust, else it won't fly.
You may try to use PX4 firmware, it has little better support.

truhoang, 2017-03-03:
Hi @vooon do you know where to check in the px4 firmware src files whether it supports yaw rate commands or not?

vooon, 2017-03-03:
https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_receiver.cpp#L802

truhoang, 2017-03-20:
I have determined that the ardupilot firmware for 3DR Solo does not have the implementation above. Issue is closed.

closed 2017-03-20 by truhoang