mavlink/mavros #668: can't get into

cs147, 2017-03-01:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

when trying to get into the nuttx shell via: `./Tools/mavlink_shell.py /dev/ttyACM0 --baudrate 9600`

while the mavros node is running (ie we've done `roslaunch mavros px4.launch`)

mavros instantly dies when we try to get into the shell. Or if I'm in the shell and then launch mavros/px4, then the shell dies.

### MAVROS version and platform

Mavros: 0.18.7
ROS: Indigo
Ubuntu: 14.04

### Autopilot type and version

[  ] ArduPilot
[X] PX4

Version: 1.5.5

### Node logs

```
[ERROR] [1488329637.890963490]: serial0: receive: End of file
terminate called after throwing an instance of 'std::system_error'
  what():  Resource deadlock avoided
================================================================================REQUIRED process [mavros-1] has died!
process has died [pid 7057, exit code -6, cmd /home/quadlab/ros_ws_kinetic_src/devel/lib/mavros/mavros_node /vicon/Pelican_Landing/Pelican_Landing:=/mavros/mocap/tf __name:=mavros __log:=/home/quadlab/.ros/log/ac66d0ea-fe18-11e6-a434-001e063056a5/mavros-1.log].
log file: /home/quadlab/.ros/log/ac66d0ea-fe18-11e6-a434-001e063056a5/mavros-1*.log
Initiating shutdown!
================================================================================
[mavros-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

```

### Diagnostics

```
header: 
  seq: 10
  stamp: 
    secs: 1488329955
    nsecs: 297649403
  frame_id: ''
status: 
  - 
    level: 0
    name: mavros: FCU connection
    message: connected
    hardware_id: /dev/ttyACM0:9600
    values: 
      - 
        key: Received packets:
        value: 345
      - 
        key: Dropped packets:
        value: 0
      - 
        key: Buffer overruns:
        value: 0
      - 
        key: Parse errors:
        value: 0
      - 
        key: Rx sequence number:
        value: 209
      - 
        key: Tx sequence number:
        value: 78
      - 
        key: Rx total bytes:
        value: 6650
      - 
        key: Tx total bytes:
        value: 1852
      - 
        key: Rx speed:
        value: 34.000000
      - 
        key: Tx speed:
        value: 397.000000
  - 
   level: 2
    name: mavros: GPS
    message: No satellites
    hardware_id: /dev/ttyACM0:9600
    values: 
      - 
        key: Satellites visible
        value: 0
      - 
        key: Fix type
        value: 0
      - 
        key: EPH (m)
        value: Unknown
      - 
        key: EPV (m)
        value: Unknown
  - 
    level: 0
    name: mavros: Heartbeat
    message: Normal
    hardware_id: /dev/ttyACM0:9600
    values: 
      - 
        key: Heartbeats since startup
        value: 7
      - 
        key: Frequency (Hz)
        value: 1.064927
      - 
        key: Vehicle type
        value: Quadrotor
      - 
        key: Autopilot type
        value: ArduPilotMega / ArduCopter
      - 
        key: Mode
        value: STABILIZE
      - 
        key: System status
        value: Standby
  - 
    level: 0
    name: mavros: System
    message: Normal
    hardware_id: /dev/ttyACM0:9600
    values: 
      - 
        key: Sensor present
        value: 0x00000000
      - 
        key: Sensor enabled
        value: 0x00000000
      - 
        key: Sensor helth
        value: 0x00000000
      - 
        key: CPU Load (%)
        value: 0.0
      - 
        key: Drop rate (%)
        value: 0.0
      - 
        key: Errors comm
        value: 0
      - 
        key: Errors count #1
        value: 0
      - 
        key: Errors count #2
        value: 0
      - 
        key: Errors count #3
        value: 0
      - 
        key: Errors count #4
        value: 0
 - 
    level: 2
    name: mavros: Battery
    message: No data
    hardware_id: /dev/ttyACM0:9600
    values: 
      - 
        key: Voltage
        value: -1.00
      - 
        key: Current
        value: 0.0
      - 
        key: Remaining
        value: 0.0
  - 
    level: 2
    name: mavros: Time Sync
    message: No events recorded.
    hardware_id: /dev/ttyACM0:9600
    values: 
      - 
        key: Timesyncs since startup
        value: 0
      - 
        key: Frequency (Hz)
        value: 0.000000
      - 
        key: Last dt (ms)
        value: 0.000000
      - 
        key: Mean dt (ms)
        value: 0.000000
      - 
        key: Last system time (s)
        value: 0.000000000
      - 
        key: Time offset (s)
        value: 0.000000000

```

### Check ID

```
OK. I got messages from 1:1.

---
Received 766 messages, from 1 addresses
sys:comp   list of messages
  1:1     0, 44, 77, 148, 22, 253

```

cs147, 2017-03-01:
when I'm in the shell and then I launch the px4 launchfile, this is the shell output:

```
nsh> Traceback (most recent call last):
  File "./Tools/mavlink_shell.py", line 219, in <module>
    main()
  File "./Tools/mavlink_shell.py", line 196, in main
    data = mav_serialport.read(4096)
  File "./Tools/mavlink_shell.py", line 80, in read
    self._recv()
  File "./Tools/mavlink_shell.py", line 70, in _recv
    timeout=0.03)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/pymavlink/mavutil.py", line 346, in recv_match
    m = self.recv_msg()
  File "/opt/ros/indigo/lib/python2.7/dist-packages/pymavlink/mavutil.py", line 309, in recv_msg
    s = self.recv(n)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/pymavlink/mavutil.py", line 801, in recv
    ret = self.port.read(n)
  File "/usr/lib/python2.7/dist-packages/serial/serialposix.py", line 453, in read
    buf = os.read(self.fd, size-len(read))
OSError: [Errno 11] Resource temporarily unavailable

```

cs147, 2017-03-01:
contents of the mavros log: 

```
[roslaunch][INFO] 2017-02-28 17:19:21,032: ... successfully launched [mavros-1]
[roslaunch][INFO] 2017-02-28 17:19:21,033: ... launch_nodes complete
[roslaunch.pmon][INFO] 2017-02-28 17:19:21,033: registrations completed <ProcessMonitor(ProcessMonitor-1, started daemon -1238096800)>
[roslaunch.parent][INFO] 2017-02-28 17:19:21,033: ... roslaunch parent running, waiting for process exit
[roslaunch][INFO] 2017-02-28 17:19:21,034: spin
[roslaunch][ERROR] 2017-02-28 17:19:33,168: ================================================================================REQUIRED process [mavros-1] has died!
process has died [pid 9675, exit code -6, cmd /home/username/ros_ws_kinetic_src/devel/lib/mavros/mavros_node /vicon/objectname/objectname:=/mavros/mocap/tf __name:=mavros __log:=/home/quadlab/.ros/log/ac66d0ea-fe18-11e6-a434-001e063056a5/mavros-1.log].
log file: /home/quadlab/.ros/log/ac66d0ea-fe18-11e6-a434-001e063056a5/mavros-1*.log
Initiating shutdown!
================================================================================
[roslaunch.pmon][INFO] 2017-02-28 17:19:33,269: ProcessMonitor._post_run <ProcessMonitor(ProcessMonitor-1, started daemon -1238096800)>
[roslaunch.pmon][INFO] 2017-02-28 17:19:33,270: ProcessMonitor._post_run <ProcessMonitor(ProcessMonitor-1, started daemon -1238096800)>: remaining procs are [<roslaunch.nodeprocess.LocalProcess object at 0xb6361530>]
[roslaunch.pmon][INFO] 2017-02-28 17:19:33,271: ProcessMonitor exit: killing mavros-1
[roslaunch][INFO] 2017-02-28 17:19:33,272: [mavros-1] killing on exit
[roslaunch.pmon][INFO] 2017-02-28 17:19:33,276: ProcessMonitor exit: cleaning up data structures and signals
[roslaunch.pmon][INFO] 2017-02-28 17:19:33,277: ProcessMonitor exit: pmon has shutdown
[roslaunch][INFO] 2017-02-28 17:19:33,356: process monitor is done spinning, initiating full shutdown
[roslaunch][INFO] 2017-02-28 17:19:33,357: runner.stop()
[roslaunch][INFO] 2017-02-28 17:19:33,357: shutting down processing monitor...
[roslaunch][INFO] 2017-02-28 17:19:33,358: shutting down processing monitor <ProcessMonitor(ProcessMonitor-1, stopped daemon -1238096800)>
[roslaunch.pmon][INFO] 2017-02-28 17:19:33,358: ProcessMonitor.shutdown <ProcessMonitor(ProcessMonitor-1, stopped daemon -1238096800)>
[roslaunch][INFO] 2017-02-28 17:19:33,358: ... shutting down processing monitor complete
[roslaunch][INFO] 2017-02-28 17:19:33,359: done
[roslaunch.pmon][INFO] 2017-02-28 17:19:33,359: ProcessMonitor.shutdown <ProcessMonitor(ProcessMonitor-1, stopped daemon -1238096800)>
[rospy.core][INFO] 2017-02-28 17:19:33,360: signal_shutdown [atexit]

```

mhkabir, 2017-03-01:
You can only have one process accessing the serial port. 

Add a GCS proxy and use it for the mavlink_shell script.

cs147, 2017-03-01:
FYI to any who might look this up later, I solved it.
Thanks for the tip @mhkabir 

I am able to run this:
`roslaunch mavros px4.launch fcu_url:=/dev/ttyACM0:921600 gcs_url:=udp://@<odroid-ip>`

Then to get the terminal of the nuttx shell, I can run this, and nothing dies:
`./Tools/mavlink_shell.py <odroid-ip>:14550 --baudrate 921600`