mavlink/mavros #669: Plugin distance_sensor loaded but Topic Not found!

MasterRos, 2017-03-02:
Hi..
[ INFO] [1488479645.789276243]: Plugin distance_sensor loaded
[ WARN] [1488479645.789853753]: DS: plugin not configured!
[ INFO] [1488479645.789883228]: Plugin distance_sensor initialized
rostopic list uploaded here [https://file.io/7QzdSx](url)
i removed that from this list (apm_pluginlists.yaml):

plugin_blacklist:
.# common
..
.# extras
...
distance_sensor
...

vooon, 2017-03-02:
That plugin require configuration like that: https://github.com/mavlink/mavros/blob/master/mavros/launch/px4_config.yaml#L116
But APM do not support DISTANCE_SENSOR message anyway.

Ahrovan, 2017-03-02:
I checked this (ardupilot-Copter-3.4.5 - GCS_Mavlink.cpp):
void GCS_MAVLINK_Copter::handleMessage(mavlink_message_t* msg)
{
...
    case MAVLINK_MSG_ID_DISTANCE_SENSOR:
    {
        result = MAV_RESULT_ACCEPTED;
        copter.rangefinder.handle_msg(msg);
        break;
    }
..
}
and ( ardupilot-Copter-3.4.5 - AP_RangeFInder_Mavlink.cpp):
void AP_RangeFinder_MAVLink::handle_msg(mavlink_message_t *msg)
{
    mavlink_distance_sensor_t packet;
    mavlink_msg_distance_sensor_decode(msg, &packet);

    last_update_ms = AP_HAL::millis();
    distance_cm = packet.current_distance;
}
//
last line show APM can set distance_cm from DISTANCE_SENSOR message!

Ahrovan, 2017-03-02:
what mean  DS: plugin not configured!
I check mavros with PX4 stack too, but this warning Remained.

vooon, 2017-03-03:
Ok, looks like APM now implemented that. Update to master, it added config in #670.

Ahrovan, 2017-03-03:
that's work.. but parameter for select pub or sub do not work well.

TSC21, 2017-04-04:
@Ahrovan any updates? Did you check the px4_config.yaml for how it works?

Ahrovan, 2017-04-04:
No, just i checked apm_pluginlists.yaml

TSC21, 2017-04-04:
Apply the configs the same way as you find on px4_config. See if you want to publish the data comming from the FCU or you want to subscribe data coming from a sensor on your computer and send it to the FCU.

TSC21, 2017-04-14:
@Ahrovan updates?

TSC21, 2017-06-09:
Closing as staled. @MasterRos please reopen if you still have problems.

closed 2017-06-09 by TSC21

khancyr, 2017-06-09:
distance_sensor is now full implemented on ardupilot side for sensor output and as input

TSC21, 2017-06-09:
> distance_sensor is now full implemented on ardupilot side for sensor output and as input

Thanks for confirming @khancyr.