mavlink/mavros #672: Question on UDP in GCS_Bridge - is it a client or server ?

Tutorgaming, 2017-03-06:
For the explanation of this line in help section of
GCS_BRIDGE 
UDP: udp://[bind_host][:port]@[remote_host][:port][/?ids=sysid,compid]

1.
I want to know that GCS_Bridge act as a server or client?
and what is the meaning of bind_host and remote_host?

2.
I planned to have a drone monitoring system which has MAV as a payload message.
So at the drones (Computer+ROS powered - not the APM), 
GCS_Bridge must act like client which send the status to the server 
and wait to receive the command (and waypoints) from server as well 

Do you have any idea about this?

vooon, 2017-03-06:
Read about BSD sockets and UDP.
Bind - where program will listen incoming messages.
Remote - where it will send messages.

TSC21, 2017-04-04:
@Tutorgaming is this answered?

Tutorgaming, 2017-04-07:
I think this question is answered. 

By reading through codes, I found that GCS_Bridge acts like both client and server in order to be the listener on the ROS Side and Send it via network  

About the meaning of the bind and remote, Vooon had already answered that. 

For my monitoring system, I implemented both UDP client and server on the drones to send and receive the MAV command from and to my Custom Mission planner, It works pretty well on the prototype.  :) 
 . . But to make it ready for the large amounts of drones, I still have no idea to manage such a network thing. 

BTW Many thanks

TSC21, 2017-04-07:
> But to make it ready for the large amounts of drones, I still have no idea to manage such a network thing.

Try broadcasting.

Closing this as solved.

closed 2017-04-07 by TSC21