mavlink/mavros #675: Plugin setpoint_raw/attitude does not publish if I set the bitmask to IGNORE_ATTITUDE

MAVREEF, 2017-03-10:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

I am controlling a quad using the setpoint_raw/attitude plugin.  I would like to send rate commands instead of quaternions but when I set the bitmask to IGNORE_ATTITUDE the loopback topic "/mavros/setpoint_raw/target_attitude" stops publishing.  Here is how I'm coding it:

mavros_msgs::AttitudeTarget att_setp;

att_setp.type_mask = mavros_msgs::AttitudeTarget::IGNORE_ATTITUDE;
.
.
.
//orientation and rate data for pixhawk commands
        att_setp.orientation.x = manual_output_att.x;
        att_setp.orientation.y = manual_output_att.y;
        att_setp.orientation.z = manual_output_att.z;
        att_setp.orientation.w = manual_output_att.w;
        att_setp.body_rate.x = .1;
        att_setp.body_rate.y = .1;
        att_setp.body_rate.z = .1;

// use for OFFBOARD autonomous control
        att_setp.thrust = auto_thrust_out;

//publishes control message
        att_setp.header.stamp = `rosTimeCurrent;`
        att_pub.publish(att_setp);

Any help is very much appreciated!

Respectfully,

MAVREEF
     

### MAVROS version and platform

Mavros: ?0.17.5?
ROS: ?Indigo?
Ubuntu: ?14.04?

### Autopilot type and version

[  ] ArduPilot
[ X ] PX4

Version: ?1.5.5?

### Node logs

```
SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: ROLL_180
[... 274 lines cut ...]
        value: -1.427935
      - 
        key: Mean dt (ms)
        value: 0.001462
      - 
        key: Last system time (s)
        value: 8325.988326000
      - 
        key: Time offset (s)
        value: 1489175972.040629864

```

### Check ID

```
rosrun mavros checkid

OK. I got messages from 1:1.

---
Received 7710 messages, from 1 addresses
sys:comp   list of messages
  1:1     0, 1, 2, 132, 140, 141, 147, 30, 31, 32, 36, 65, 70, 74, 76, 87, 230, 231, 105, 111, 241, 245
```

vooon, 2017-03-11:
I don't see any problems on mavros side.
Look at receiver: https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_receiver.cpp#L1175
And loopback sender: https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_messages.cpp#L2671

And here your problem on line 2675: `if (_att_sp_sub->update(&_att_sp_time, &att_sp)) {`

TSC21, 2017-04-04:
@MAVREEF updates?

TSC21, 2017-04-14:
Closing this as there are no updates. Reopen if needed.

closed 2017-04-14 by TSC21

fabrizioschiano, 2020-11-02:
@MAVREEF any news on this? I know it is a long time ago but I would be interested to know if you advanced on your issue.