mavlink/mavros #677: Custom mavlink message not received by mavros

Ramomrito, 2017-03-15:
### Issue details

Hello,

For the past three days I have been trying with no success to send my custom mavlink message to my ros network via mavros. I have looked at similiar issues #657  #166 , but they have not solved my problem. I have tried everything from adding my message to the normal common dialect, to creating my own dialect, but for some reason the handler function in my plugin is not executed when it takes my custom message as an argument.

If I however use another message in the common dialect (under 255 id ... mavlink 1.0?), my handle function works all fine. The compiler is not complaining about my message because I have the message definition in the mavlink package common.xml.

I have verified that my message is being sent from the pixhawk, and for clarification I used the dev.px4.io guide to set this all up.

Right now I am truly clueless as to why my message is not being received by mavros and would appreciate any help

### MAVROS version and platform

Mavros: 0.18.7
ROS: Kinetic
Ubuntu: 16.04

### Autopilot type and version

[  ] ArduPilot
[ x ] PX4
### Node logs

Launches fine and loads my plugin!

### Diagnostics

```
header: 
  seq: 19
  stamp: 
    secs: 1489599977
    nsecs: 396413265
  frame_id: ''
status: 
  - 
    level: 0
    name: mavros: FCU connection
    message: connected
    hardware_id: udp://:14540@127.0.0.1:14557
    values: 
      - 
        key: Received packets:
        value: 7821
      - 
        key: Dropped packets:
        value: 0
      - 
        key: Buffer overruns:
        value: 0
      - 
        key: Parse errors:
        value: 0
      - 
        key: Rx sequence number:
        value: 156
      - 
        key: Tx sequence number:
        value: 100
      - 
        key: Rx total bytes:
        value: 337240
      - 
        key: Tx total bytes:
        value: 9239
      - 
        key: Rx speed:
        value: 41952.000000
      - 
        key: Tx speed:
        value: 835.000000
  - 
    level: 0
    name: mavros: GPS
    message: 3D fix
    hardware_id: udp://:14540@127.0.0.1:14557
    values: 
      - 
        key: Satellites visible
        value: 10
      - 
        key: Fix type
        value: 3
      - 
        key: EPH (m)
        value: 0.00
      - 
        key: EPV (m)
        value: 0.00
  - 
    level: 0
    name: mavros: Heartbeat
    message: Normal
    hardware_id: udp://:14540@127.0.0.1:14557
    values: 
      - 
        key: Heartbeats since startup
        value: 15
      - 
        key: Frequency (Hz)
        value: 1.000013
      - 
        key: Vehicle type
        value: Quadrotor
      - 
        key: Autopilot type
        value: PX4 Autopilot
      - 
        key: Mode
        value: AUTO.RTL
      - 
        key: System status
        value: Standby
  - 
    level: 0
    name: mavros: System
    message: Normal
    hardware_id: udp://:14540@127.0.0.1:14557
    values: 
      - 
        key: Sensor present
        value: 0x00000000
      - 
        key: Sensor enabled
        value: 0x00000000
      - 
        key: Sensor helth
        value: 0x00000000
      - 
        key: CPU Load (%)
        value: 0.0
      - 
        key: Drop rate (%)
        value: 0.0
      - 
        key: Errors comm
        value: 0
      - 
        key: Errors count #1
        value: 0
      - 
        key: Errors count #2
        value: 0
      - 
        key: Errors count #3
        value: 0
      - 
        key: Errors count #4
        value: 0
  - 
    level: 0
    name: mavros: Battery
    message: Normal
    hardware_id: udp://:14540@127.0.0.1:14557
    values: 
      - 
        key: Voltage
        value: 12.15
      - 
        key: Current
        value: -1.0
      - 
        key: Remaining
        value: 100.0
  - 
    level: 0
    name: mavros: Time Sync
    message: Normal
    hardware_id: udp://:14540@127.0.0.1:14557
    values: 
      - 
        key: Timesyncs since startup
        value: 156
      - 
        key: Frequency (Hz)
        value: 10.000138
      - 
        key: Last dt (ms)
        value: 0.019910
      - 
        key: Mean dt (ms)
        value: -0.000020
      - 
        key: Last system time (s)
        value: 16.850916000
      - 
        key: Time offset (s)
        value: 1489599960.528765917
---
```

### Check ID

OK. I got messages from 1:1.
```
---
Received 6089 messages, from 1 addresses
sys:comp   list of messages
  1:1     0, 1, 2, 140, 141, 147, 24, 30, 31, 32, 33, 74, 76, 83, 85, 87, 230, 231, 105, 111, 241, 242, 245, 253
```
No Surprise... my ID is not there

vooon, 2017-03-16:
Show me the code! :)
If you use v2 ID (>255), then you might need to enable v2 protocol, but it only required to send msg.
Receive should automatically switch. Also you can see all received messages on `/mavlink/from`, even if receiver don't know msg crc, but framing should be correct.

Ramomrito, 2017-03-16:
It is true that the message ID is over 255 because everything else was occupied.
I actually also tried replacing one of the under 255 IDs but it caught me up with a ton of errors so I am refraining from it for now. 

I should also note that I changed a param in mavros.cpp (line 49)
```
nh.param<std::string>("fcu_protocol", fcu_protocol, "v2.0");
```

For completeness here is the code from the pixhawk side:
```
#include <uORB/topics/voliro_ao.h>
#include <v2.0/uAvionix/mavlink_msg_voliro_ao.h>

class MavlinkStreamVoliroAo : public MavlinkStream
{
public:
    const char *get_name() const
    {
        return MavlinkStreamVoliroAo::get_name_static();
    }
    static const char *get_name_static()
    {
        return "VOLIRO_AO";
    }

		static uint16_t get_id_static()
		{
			return MAVLINK_MSG_ID_VOLIRO_AO;
		}

		uint16_t get_id()
		{
			return get_id_static();
		}

    static MavlinkStream *new_instance(Mavlink *mavlink)
    {
        return new MavlinkStreamVoliroAo(mavlink);
    }
    unsigned get_size()
    {
        return MAVLINK_MSG_ID_VOLIRO_AO_LEN + MAVLINK_NUM_NON_PAYLOAD_BYTES;
    }

private:
    MavlinkOrbSubscription *_sub;
    uint64_t _voliro_ao_time;

    /* do not allow top copying this class */
    MavlinkStreamVoliroAo(MavlinkStreamVoliroAo &);
    MavlinkStreamVoliroAo& operator = (const MavlinkStreamVoliroAo &);

protected:
    explicit MavlinkStreamVoliroAo(Mavlink *mavlink) : MavlinkStream(mavlink),
        _sub(_mavlink->add_orb_subscription(ORB_ID(voliro_ao))),  // make sure you enter the name of your uorb topic here
        _voliro_ao_time(0)
    {}

    void send(const hrt_abstime t)
    {
			//Not sure about with _struct_ or not /Omar
      struct voliro_ao_s _voliro_ao;    //make sure ca_traj_struct_s is the definition of your uorb topic

      if (_sub->update(&_voliro_ao_time, &_voliro_ao)) {
        mavlink_voliro_ao_t _msg_voliro_ao;  //make sure mavlink_ca_trajectory_t is the definition of your custom mavlink message

        //_msg_voliro_ao.timestamp = _voliro_ao.timestamp;
        _msg_voliro_ao.voliro_alpha[0] = _voliro_ao.voliro_alpha[0];
				_msg_voliro_ao.voliro_alpha[1] = _voliro_ao.voliro_alpha[1];
				_msg_voliro_ao.voliro_alpha[2] = _voliro_ao.voliro_alpha[2];
				_msg_voliro_ao.voliro_alpha[3] = _voliro_ao.voliro_alpha[3];
				_msg_voliro_ao.voliro_alpha[4] = _voliro_ao.voliro_alpha[4];
				_msg_voliro_ao.voliro_alpha[5] = _voliro_ao.voliro_alpha[5];

        _msg_voliro_ao.voliro_omega[0]  = _voliro_ao.voliro_omega[0];
				_msg_voliro_ao.voliro_omega[1]  = _voliro_ao.voliro_omega[1];
				_msg_voliro_ao.voliro_omega[2]  = _voliro_ao.voliro_omega[2];
				_msg_voliro_ao.voliro_omega[3]  = _voliro_ao.voliro_omega[3];
				_msg_voliro_ao.voliro_omega[4]  = _voliro_ao.voliro_omega[4];
				_msg_voliro_ao.voliro_omega[5]  = _voliro_ao.voliro_omega[5];

        mavlink_msg_voliro_ao_send_struct(_mavlink->get_channel(), &_msg_voliro_ao);
      }
    }
};

```

This is my plugin, I took imu_pub.cpp as reference. If I change the message from 
```
#include <mavros/mavros_plugin.h>
#include <mavros_msgs/voliro_ao.h>

namespace mavros {
namespace std_plugins {

/**
 * @brief IMU data publication plugin
 */
class VOLIROPubPlugin : public plugin::PluginBase {
public:
	VOLIROPubPlugin() : PluginBase(),
		imu_nh("~voliro"),
		has_voliro(false)
	{ }

	void initialize(UAS &uas_)
	{
		PluginBase::initialize(uas_);

		voliro_pub = imu_nh.advertise<mavros_msgs::voliro_ao>("voliro_ao",10);

		// reset has_* flags on connection change
		enable_connection_cb();
	}

	Subscriptions get_subscriptions() {
		return {
           make_handler(&VOLIROPubPlugin::handle_voliro),
		};
	}

private:
	ros::NodeHandle imu_nh;
	ros::Publisher voliro_pub;

	bool has_voliro;

	void handle_voliro(const mavlink::mavlink_message_t *msg, mavlink::common::msg::VOLIRO_AO &volao)
	{

		ROS_INFO_COND_NAMED(!has_voliro, "imu", "VOLIRO: VOLIRO ALPHA/OMEGA detected!");
		has_voliro = true;

		ROS_INFO("VOLIRO WAS HERE");

		// mavros_msgs::voliro_ao volomsg;
		// volomsg.voliro_alpha[0] = volao.voliro_alpha[0];
		// volomsg.voliro_alpha[1] = volao.voliro_alpha[1];
		// volomsg.voliro_alpha[2] = volao.voliro_alpha[2];
		// volomsg.voliro_alpha[3] = volao.voliro_alpha[3];
		// volomsg.voliro_alpha[4] = volao.voliro_alpha[4];
		// volomsg.voliro_alpha[5] = volao.voliro_alpha[5];
		//
		// volomsg.voliro_omega[0] = volao.voliro_omega[0];
		// volomsg.voliro_omega[1] = volao.voliro_omega[1];
		// volomsg.voliro_omega[2] = volao.voliro_omega[2];
		// volomsg.voliro_omega[3] = volao.voliro_omega[3];
		// volomsg.voliro_omega[4] = volao.voliro_omega[4];
		// volomsg.voliro_omega[5] = volao.voliro_omega[5];

		// voliro_pub.publish(volomsg);

	}

	void connection_cb(bool connected) override
	{
    has_voliro = false;
	}

};
}	// namespace std_plugins
}	// namespace mavros

#include <pluginlib/class_list_macros.h>
PLUGINLIB_EXPORT_CLASS(mavros::std_plugins::VOLIROPubPlugin, mavros::plugin::PluginBase)

```

vooon, 2017-03-16:
Plugin code looks correct. Try to enable debugging messages:

```
#
#   rosconsole will find this file by default at $ROS_ROOT/config/rosconsole.config
#
#   You can define your own by e.g. copying this file and setting
#   ROSCONSOLE_CONFIG_FILE (in your environment) to point to the new file
#
log4j.logger.ros=INFO
log4j.logger.ros.roscpp.superdebug=WARN

log4j.logger.ros.mavros=DEBUG
log4j.logger.ros.rosconsole_bridge.mavconn=DEBUG

log4j.logger.ros.mavros_extras=DEBUG
log4j.logger.rosout=DEBUG
```

But also i recommend to write simple python sender (connect to TCP & encode and send random) to check where problem occurs.

Ramomrito, 2017-03-16:
Hello Vooon,

Thanks for the reply, I think the problem has to do with the Mavlink version somehow. Because mavros only receives messages with an ID < 255 right now, even though I changed the fcu_protocol parameter to v2.0 in mavros.cpp. 
Today I found out about the following doc: https://github.com/mavlink/mavlink/blob/master/doc/MAVLink2.md

It says I have to enable Mavlink protocol 2.0 from px4 side (which I did) as well as enable signing. Enabling the protocol 2.0 is not sufficient however. I am trying to enable this signing feature at the moment, but I am about to test the other direction of communication like you said.

vooon, 2017-03-16:
Mavros do not support signing yet. So you should enable v2 on PX4 side, but without signing.

vooon, 2017-03-16:
As i remember i checked v2 on handling msgid 256 (1 bit more than byte may store).

Ramomrito, 2017-03-16:
Hmm ok. The doc says:
 "if signing is not enabled and MAVLink2 is enabled then the vehicle may choose to start by sending MAVLink1 and switch to MAVLink2 on a link when it first receives a MAVLink2 message on the link".

So does this mean that I have to send something from ros to px4 first to enable mavlink2?
This is what I am working on at the moment anyway.

Ramomrito, 2017-03-16:
Well, after all this time a colleague of mine found out what was not working. It turned out to be from the Pixhawk side. Sadly, the PX4 documentation does not state anything about this change, but I'll write it down here for reference. Apart from changing the mavlink_messages.cpp file to create the stream class and add it to the list, you have to change the mavlink_main.cpp file. 
There is a switch at almost line 2000~ that looks like this
```
switch (_mode) {
	case MAVLINK_MODE_NORMAL:
		configure_stream("SYS_STATUS", 1.0f);
		configure_stream("EXTENDED_SYS_STATE", 1.0f);
		configure_stream("HIGHRES_IMU", 1.5f);
		configure_stream("ATTITUDE", 20.0f);
		configure_stream("RC_CHANNELS", 5.0f);
		configure_stream("SERVO_OUTPUT_RAW_0", 1.0f);
		configure_stream("ALTITUDE", 1.0f);
		configure_stream("GPS_RAW_INT", 1.0f);
		configure_stream("ADSB_VEHICLE", 2.0f);
		configure_stream("COLLISION", 2.0f);
		configure_stream("DISTANCE_SENSOR", 0.5f);
		configure_stream("OPTICAL_FLOW_RAD", 1.0f);
		configure_stream("VISION_POSITION_NED", 1.0f);
		configure_stream("ESTIMATOR_STATUS", 0.5f);
		configure_stream("NAV_CONTROLLER_OUTPUT", 1.5f);
		configure_stream("GLOBAL_POSITION_INT", 5.0f);
		configure_stream("LOCAL_POSITION_NED", 1.0f);
		configure_stream("POSITION_TARGET_GLOBAL_INT", 1.5f);
		configure_stream("ATTITUDE_TARGET", 2.0f);
		configure_stream("HOME_POSITION", 0.5f);
		configure_stream("NAMED_VALUE_FLOAT", 1.0f);
		configure_stream("VFR_HUD", 4.0f);
		configure_stream("WIND_COV", 1.0f);
		configure_stream("VOLIRO_AO",10.0f);
		break;
```
I needed to append a configure_stream of my message to get it working. (I did it in all of the cases for now)

Thank you very much, vooon, for looking at my problem. I appreciate the help.

closed 2017-03-16 by Ramomrito

tianzhuqiao, 2023-10-25:
Hi @Ramomrito, I got the same issue (looks like PX4 doesn't receive the message I added). I want to send a mavlink message to PX4 (so the message is only one way from outside to PX4, and I confirm the message is sent). Do I still need to add the "configure_stream" part? The following doc doesn't mention it. 
https://docs.px4.io/main/en/middleware/mavlink.html#receiving-custom-mavlink-messages