mavlink/mavros #680: TF_OLD_DATA ignoring data mavros nodes

zer1988, 2017-03-24:
**### Issue details**
Experienced a problem relative to run two remote machines (2 UAVs), which is installed mavros inside and one pc using roslaunch on master pc. I could change the node name of 2 mavros packages in each UAVs. However, everytime when I run roslaunch code, the mavros's topics are displayed in Rqt_console, some topics of mavros1 and mavros2 are loaded successfully  as well as all of nodes are displayed in launch screen as in logfile, and then  the warn messages are displayed every 0.1 second: "TF_OLD_DATA ignoring data from the past for frame base_link..." for all of topics of mavros1(uav1) and mavros2 (uav2). Since this problem occurred, all of systems experienced the lag time. I saw a stop of few seconds in the flow message of Rqt_console sometimes. While I reduced the number of nodes in two UAVs. The speed of network is better, but still slow. Any help is highly appriciate.

### MAVROS version and platform
Mavros: ?0.18.4?
ROS: Indigo
Ubuntu: 14.04 LTS

**### Autopilot type and version**
PX4
Version: 1.5.5

**### Logfile**
[logfile.docx](https://github.com/mavlink/mavros/files/866525/logfile.docx)

**### launch file**
[launchfile.pdf](https://github.com/mavlink/mavros/files/866537/launchfile.pdf)

**###Error**
[image]

vooon, 2017-03-24:
As i understand both nodes uses `/tf` (and `/tf_static`) and both node has identical TF tree.
Previous ROS releases has tf_prefix option, look how OSRF deal in UCTF project.

zer1988, 2017-03-24:
Hi voon. This warning just appear while i run 2 mavros nodes using 1 launch file. It no longer shows this when i only run 1 mavros node. And the problem is that this message  is displayed every 0.1 seconds and make a overloadt of the bandwidth of my network or slow down my network. Do you know how i can disable this mess or how i can solve  this prolem. Thanks.

vooon, 2017-03-25:
Your problem is that node1 and n2 publish same transformation e.g map → base_link.
Machine time is not perfectly synchronized, network delay may wary, so TF lib got msg from n1 with T=2, then from n2 with T=1 => report error.

`tf_prefix` was removed (actually it do not work). You need to modify config.yaml for each instance that you want to run under same `roscore`.

zer1988, 2017-03-25:
Hi voon,
Oh I see now. Thank you for your explanation. To solve for the same /tf. This way is one solution?
<group ns="drone1">
    <node machine="odroid1" name="mavros1" pkg="mavros" type="mavros_node"
output="screen">
</group>
so that i will have /drone1/tf

vooon, 2017-03-25:
As i remember, tf lib uses `/tf`, so because of `/` (absolute path) that topic shouldn't go into namespace. But you may try.

zer1988, 2017-03-25:
Hi voon, I will check this on Monday and tell you my result. But this one I saw from the UCTF project. 
<node
    name="px4_@(vehicle_name)"
    pkg="uctf"
    type="px4.sh"
    args="@(controller_config_path)"
    cwd="node"
    ns="/@(vehicle_name)"
    output="screen" />
@[ if launch_mavros ]@
@# TODO not fully tested adding this back.
  <include file="$(find mavros)/launch/px4.launch" ns="/@(vehicle_name)">
    <arg name="fcu_url" value="udp://:@(ros_interface_port4)@@@(local_ip):@(ros_interface_port3)" />
    <arg name="tgt_system" value="@(mav_sys_id)" />
  </include>
@[end if]

zer1988, 2017-03-27:
Hi voon, I tried with that way but it is not sucessful. Actually I just use tf::transformation to convert my drone's angles from quaternion to eurler. Currently, i can program to convert directly without using this library more. So how can I disable /tf library? and when I do that, it will affect to another nodes of mavros like mavros/ imu_msg ? Thanks.

vooon, 2017-03-27:
By default TF is not used, see `<plugin>/tf/send` and `<plugin>/tf/listen` should be `false`.
But library itself cannot be disabled, as it always used by UAS class (so subscription to `/tf` is always present.).
You may give different names to `frame_id`, or remap `/tf` to `/ns/tf`.

zer1988, 2017-03-28:
Hi voon, can you guide me more clearly a little how to add frame_id or remap for /tf to /ns/tf because the last time while I add group ns, it is not seccessful. This message is still displayed. Thanks.

vooon, 2017-03-28:
https://github.com/mavlink/mavros/blob/master/mavros/launch/px4_config.yaml `frame_id`.

http://wiki.ros.org/roslaunch/XML/remap

TSC21, 2017-04-04:
@zer1988 updates?

TSC21, 2017-04-14:
Closing this as there's no feedback. Please reopen if needed.

closed 2017-04-14 by TSC21

001zwzwzw, 2023-06-07:
@zer1988 do you solve this problem? i meet the same question