mavlink/mavros #682: Send a new custom mavlink message from PX4 to mavros

zhangyongwei93, 2017-03-29:
Hi,  
I want to send a custom mavlink message to mavros, and transform to a ros topic form.  
**In pixhawk firmware (v1.5.2dev)** 

 - I have added a new message in `mavlink/v1.0/message_definitions/common.xml` with id=220, and have used `python -m mavgenerate` to generate custom message head file, then copy them into the `/mavlink/v1.0` directory.   
 - In `mavlink_messages.cpp`,I have added a new MavlinkStream class to receive a ORB topic and send the message with mavlink protocol, and have added Stream info in `StreamListItem *streams_list[]`.    
 - In `mavlink_main.cpp`, I have configured the mavlink stream through `configure_stream("custom_stream",30.0f)`.   
    
**In mavros package (indigo-devel)**  

 - I have updated the `/opt/ros/indigo/include/mavlink` with mavlink folder in pixhawk firmware.   
 - In `mavros_msgs/msg` added a new msg file
 - In `mavros/src/plugins`added a new plugins cpp file, and a new class in `mavros\mavros_plugins.xml`.  

**Currently problem**  
 I can read the custom mavlink message with the correct ID(220) and message LEN from serial port assistant in computer by connect the TELEM2 on PX4.   
In mavros, after `roslaunch mavros px4.launch`,terminal outputed that the custom plugin has loaded and initialized.  
But after `rosrun mavros checkid`,terminal outputed
`OK. I got messages from 1:1.
Received 596 messages, from 1 addresses
sys:comp   list of messages
  1:1     0, 65, 1, 230, 70, 33, 105, 74, 141, 111, 241, 242, 83, 147, 87, 36, 32, 30, 245`
there is no message id with 220.

vooon, 2017-03-29:
Did you recompile whole mavros? Known messages array compilled in libmavconn, and may be in other parts.

zhangyongwei93, 2017-03-31:
@vooon Sorry for replying late. I have recompiled whole mavros, and I tracked into the `MAVConnSerial::async_read_end` function in `libmavconn/src/serial.cpp`; at the mavlink message parse part `mavlink_parse_char(channel, rx_buf[i], &message, &status)`, I found the function body in `mavlink/v1.0/mavlink_helpers.h`, and I confirmed it is the CRC error, then I disabled CRC just for test, the custom mavlink message really appeared. But why the CRC cause error?

vooon, 2017-04-01:
Because each message at source gen stage generate additional byte computed from message definition. It is stored in message crc array and used by parse function.
Note that msg crc do not send via wire, just added to packet CRC, so any unknown message will be reported as CRC error.

Mavros v0.18 do not process messages with errors, but passes trough `/mavlink/from`.

zyq1993, 2017-04-11:
hello,@zhangyongwei93

>  I have updated the /opt/ros/indigo/include/mavlink with mavlink folder in pixhawk firmware.

Can you tell me how to updated the directory /opt/ros/indigo/include/mavlink? 
I put my custom_messages.xml in mavros/src/mavlink/message_definitions, and rebuild, but there is no custom_messages in /opt/ros/indigo/include/mavlink/v1.0 and v2.0.

vooon, 2017-04-11:
@zyq1993 that instruction for you: https://github.com/mavlink/mavros/blob/master/mavros/README.md#source-installation
Do not forget to remove binary installed mavros and mavlink. Build result will be in workspace devel dir.

zyq1993, 2017-04-11:
Now，I have removed binary installed mavros and mavlink, and the mavlink directory in /opt/ros/indigo/include/ is removed too. After I rebuild the mavros package, there is still no mavlink directory in /opt/ros/indigo/include/. Dose it matter?

zyq1993, 2017-04-11:
If I need my custom_messages directory in  /opt/ros/indigo/include/mavlink, what should I do?

vooon, 2017-04-11:
Look at `~/catkin_ws/devel/include`; Place your xml into `~/catkin_ws/src/mavlink/message_defenitions/v1.0`

zyq1993, 2017-04-11:
I place the mavlink package and my custom_messages.xml into ~/catkin_ws/src, and do catkin build, I can find the generated headers in ~/catkin_ws/devel, **but no mavlink directory in /opt/ros/indigo/include/**.
What should I do if I want there is a mavlink directory in the path /opt/ros/indigo/include/?

zhangyongwei93, 2017-04-11:
I think you can copy the generated mavlink directory in pixhawk firmware to the path `/opt/ros/indigo/include/`

zyq1993, 2017-04-11:
Oh, I solved it. Thanks a lot for the help of both of you! @vooon @zhangyongwei93

zyq1993, 2017-04-11:
Oh，I  copied the generated mavlink directory to path /opt/ros/indigo/include/, but there is a new problem.
I have a package, and that package needs generated mavlink header files, but I don't want to put the mavlink package into my package to compile together.  I need a mavlink directory in /opt/ros/indigo/include/, so I don't need to put the mavlink package in my package. But the binary installed mavlink package dosen't contain my custom messages. Although I copied the mavlink package(contains custom messages) into /opt/ros/indigo/include/, but I can't use the copied mavlink headers in my code. It implied that there is no mavlink headers.
How to solve my problem? Thanks a lot.

vooon, 2017-04-11:
@zhangyongwei93 how thing going? Is that issue fixed?

@zyq1993 relying on particular directory is bad idea, you have **cmake**! And ros release provide config for build system.
Also after build headers not heed anymore. And your package may build in same workspace as mavlink.
To solve your task, i think (a bit dirty) solution is to make fork of `release/kinetic/mavlink` branch and commit here your dialect xml.

zhangyongwei93, 2017-04-12:
@vooon  I modified the `MAVLINK_MESSAGE_CRCS`, and the issue has fixed. Thanks a lot.

zhangyongwei93, 2017-04-12:
@zyq1993 can you commit here your error message which implied that there is no mavlink headers?

zyq1993, 2017-04-12:
The mavlink msg id of my custom message is MAVLINK_MSG_ID_SENSOR_COMBINED_RAW, and when I compiles, the error is " 'MAVLINK_MSG_ID_SENSOR_COMBINED_RAW' was not declared in this scope".
So，I gave up using headers in /opt/ros/indigo/include/ .  I think only binary installed package can put their headers directories to that path.  I finally compile the mavlink package and my package together and it works well.
@vooon  @zhangyongwei93  Thanks for your help~~~

closed 2017-04-12 by zhangyongwei93

lamhung81, 2017-09-22:
Hi all, 
We would like to create a new custom mavlink message to mavros, and transform to a ros topic form, for the purpose of transferring data from a new external pressure sensor in px4 to an on-board computer.
We have followed all the steps that you have mentioned. Our platform includes Ubuntu 14.04, ROS indigo, PX4 v1.6.5
However, there some issues that we do need your help:

1. We follows the tutorial to generate new mavlink header file
http://qgroundcontrol.org/mavlink/create_new_mavlink_message
However, in the generator interface, if  "validate" option is chosen, then there are some errors preventing generate common.xml; If without "validate", there some warnings.
If possible, could you tell us which generator (and which version) we should use? Do we need "validate" option?

2. We found another mavlink generator in mavlink github. 
https://github.com/mavlink/mavlink
Should we use this one?

3. During PX4 compilation, we found that there compiler was searching for one id name (declared in mavlink v1.0 (after generating)) in v2.0. So,  do we need to generate only v1.0 or also v2.0 of mavlink?

3. There are two directories that contain mavlink v1.0 header files, one is in ROS ~/catkin_ws/build/mavlink/include/v1.0 and the other one is in the OS system ~/opt/ros/indigo/include/mavlink/v1.0; So what you have updated is that only for the OS system?

TSC21, 2017-09-22:
@lamhung81:
1. https://github.com/ArduPilot/pymavlink - that's the generator. Though, you will require the full spec of the Mavlink package, so I advise you to generate the mavlink headers from the pymavlink version that comes from https://github.com/mavlink/mavlink;
2. That's not another. pymavlink it is what generates your headers and it is included on the Mavlink repo.
3. That depends if your you want to use Mavlink 2.0 capabilities or not. Since we already parse Mavlink 2.0 msgs on MAVROS, you can generate 2.0 msgs already;
4. The first is generated from building the packages on your catkin_ws. The other one is the headers from the ROS mavlink package. And I don't understand your question.
I advise to see what @JoonmoAhn did in https://github.com/mavlink/mavros/issues/783. Do not continue commenting on this issue, since it is closed already. If you have any issues, just open a new issue so it can be tracked.