mavlink/mavros #683: Setting up mavros node to run inside 2 UAVs

zer1988, 2017-03-29:
### Issue details
Hi everybody. Currently, I was successful in controlling 1 drone using mavros. However, to run 2 drones at the same time using launch file, I do not know the right way to edit since we cannot run the mavros node in 2 drones with the same node. I tried to change the name of mavros node to mavros 1 and mavros 2 in  mavros_node.cpp. It can arm motors now as well as i can read the topics of mavros, but it is unstable and cannot take off. 

### MAVROS version and platform
Mavros: ?0.18.4?
ROS: Indigo
Ubuntu: 14.04 LTS

### Autopilot type and version

[  ] PX4
Version: 1.5.5

vooon, 2017-03-29:
You may use namespaces and better assign different SYSTEM_ID to FCU (and mavros).

zer1988, 2017-03-29:
Hi Voon,
I tried to add namespace but my motor cannot arm as usual. Further more, I assigned different system_id to fcu, but how about the mavros?. Thanks

zer1988, 2017-03-30:
Hi voon, it works nơ. Thank you for your help. The problem is that I added one group ns but i do not add the ns prefix for each node of mavros while I use. I think we can close this topic now. Thanks.

closed 2017-03-30 by vooon

michael1874888, 2017-08-21:
Hi , We are trying to do the same thing as you , but we face some problems . Can you please show me your launch file ? It would be very helpful to us .

cnpcshangbo, 2017-08-23:
I think we need to open a new issue to ask.