mavlink/mavros #686: Error when download/cat files using mavftp: IOError [Errno 110] Connection timed out

fionachui, 2017-04-04:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

When trying to download a .px4log through mavftp, I get the following error:
```
ubuntu@ubuntu:~$ rosrun mavros mavftp download /fs/microsd/log/2016-11-01/21_43_29.px4log
Prigressbar disabled. install python-progressbar
Traceback (most recent call last):
  File "/opt/ros/indigo/lib/mavros/mavftp", line 296, in <module>
    main()
  File "/opt/ros/indigo/lib/mavros/mavftp", line 292, in main
    args.func(args)
  File "/opt/ros/indigo/lib/mavros/mavftp", line 152, in do_download
    buf = from_fd.read(FTP_PAGE_SIZE)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/mavros/ftp.py", line 107, in read
    _check_raise_errno(ret)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/mavros/ftp.py", line 39, in _check_raise_errno
    raise IOError(ret.r_errno, os.strerror(ret.r_errno))
IOError: [Errno 110] Connection timed out
```
So I'm able to open the file, as the main roslaunch mavros window displays:
```
[ INFO] [1491328759.740312395]: FTP:Open /fs/microsd/log/2016-11-01/21_43_29.px4log: success, session 0, size 263247
```
But I get the [Errno 110] Connection timed out error on reading the file. 

### MAVROS version and platform

Mavros: 0.17.4
ROS: Indigo
Ubuntu: 14.04

### Autopilot type and version

[  ] ArduPilot
[X] PX4

Version: 1.1

### Node logs

```
ubuntu@ubuntu:~$ roslaunch mavros px4.launch fcu_url:="/dev/ttyUSB0:921600"
... logging to /home/ubuntu/.ros/log/4bfe2358-1960-11e7-a396-54ee75c35d17/roslaunch-ubuntu-10479.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ubuntu:40096/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: ROLL_180
[... 299 lines cut ...]
      - 
        key: Last dt (ms)
        value: -1.344876
      - 
        key: Mean dt (ms)
        value: 0.008980
      - 
        key: Last system time (s)
        value: 1210.658179000
      - 
        key: Time offset (s)
        value: 1491328273.698000193
---
```

### Check ID

```
OK. I got messages from 1:1.

---
Received 4327 messages, from 1 addresses
sys:comp   list of messages
  1:1     32, 1, 2, 36, 147, 105, 74, 140, 141, 111, 76, 241, 83, 245, 251, 44, 30, 0
```

vooon, 2017-04-04:
Flight logs often too big for FTP, also for some reason it is stable only with USB connection (usb on pihxawk - ttyACM%d).
On UART it only suitable when you need to modify extras rc or mixer file.
Also after file download script tell FCU to calculate CRC32 of entire file and here also may be timeout.

Note: FTP is slow, but QGC download implementation twice faster than mavros.

TSC21, 2017-04-14:
> Note: FTP is slow, but QGC download implementation twice faster than mavros.

Where may be the difference @vooon?

vooon, 2017-04-14:
QGC support streaming download, it require only one request message to transfer whole file.
While mine uses older request buffer, so to transfer file i should request bunch of 239-byte responses.

I'm not going to implement that mode, just use QGC.

TSC21, 2017-06-24:
> I'm not going to implement that mode, just use QGC.

I think this is a need, given that we offer the FTP possibility on MAVROS. I will look into it and get back with news (and a PR).

vooon, 2017-06-25:
@TSC21 because FTP stable only via USB, and main usage of mavros via TELEMx there no reason to implement that mode, and it is not compatible with current fuse-like interface.

So:
1. No strong reason for implementation
2. Logging may be done via mavlink and intel's router

TSC21, 2017-06-25:
@vooon as you wish. I think one of the main usages of MAVROS to some people is also use it as a replacement of QGC. Still, we are lacking support for ULog stream.

mhkabir, 2017-06-26:
It's a usability thing. File transfer is ridiculously slow over telemetry, and pulling a log of any decent size (10-15 MB) will take too long even with an optimised implementation. 

Regarding ULog streaming - Yes, it's a nice-to-have, but you can just use `mavlink-router`. (https://github.com/01org/mavlink-router).

TSC21, 2017-06-26:
> Regarding ULog streaming - Yes, it's a nice-to-have, but you can just use mavlink-router. (https://github.com/01org/mavlink-router).

True. Still, I think it's our wish as MAVROS developers and maintainers to have it full-featured.

vooon, 2017-06-26:
Actually for newer drone i will setup router between tty and mavros.
Because TELEM1 will be occupied by RFD and TELEM2. So i need router inside FCU too.

And i don't thinking that mavros is replacement for any GCS, it is set of API to write your autonomous control software.
Anyway you need GCS to tune firmware.