mavlink/mavros #687: Pixhawk position setpoint code makes quadcopter takeoff higher when send x, y values

vinhk, 2017-04-05:
Dear all, 

I am working on ROS code for making my quadcopter takeoff and do basic movements such as forward, back, left, right. I am unable to make it do so at this point using PX4Flow. The drone keeps going up when I send the x and y values. I based my code from https://dev.px4.io/ros-mavros-offboard.html documentation. Therefore, I added more lines to work on getting it to move. Here is my code: 

`/**`
` * @file offb_node.cpp`
` * @brief offboard example node, written with mavros version 0.14.2, px4 flight`
` * stack and tested in Gazebo SITL`
 `*/`
`#include <time.h>`
`#include <ros/ros.h>`
`#include <geometry_msgs/PoseStamped.h>`
`#include <mavros_msgs/CommandBool.h>`
`#include <mavros_msgs/SetMode.h>`
`#include <mavros_msgs/State.h>`

`mavros_msgs::State current_state;`
`void state_cb(const mavros_msgs::State::ConstPtr& msg){`
    `current_state = *msg;`
`}`

`int main(int argc, char **argv)`
`{`
    `time_t start = time(0);`
    `ros::init(argc, argv, "offb_node");`
    `ros::NodeHandle nh;`

    ros::Subscriber state_sub = nh.subscribe<mavros_msgs::State>
            ("mavros/state", 10, state_cb);
    ros::Publisher local_pos_pub = nh.advertise<geometry_msgs::PoseStamped>
            ("mavros/setpoint_position/local", 10);
    ros::Publisher att_pub = nh.advertise<geometry_msgs::PoseStamped>
            ("mavros/setpoint_attitude/attitude", 10);
    ros::ServiceClient arming_client = nh.serviceClient<mavros_msgs::CommandBool>
            ("mavros/cmd/arming");
    ros::ServiceClient set_mode_client = nh.serviceClient<mavros_msgs::SetMode>
            ("mavros/set_mode");

    //the setpoint publishing rate MUST be faster than 2Hz
    ros::Rate rate(10.0);

    // wait for FCU connection
    while(ros::ok() && current_state.connected){
        ros::spinOnce();
        rate.sleep();
    }

    geometry_msgs::PoseStamped pose;
    pose.pose.position.x = 0;
    pose.pose.position.y = 0;
    pose.pose.position.z = 1;

    //send a few setpoints before starting
    for(int i = 100; ros::ok() && i > 0; --i){
        local_pos_pub.publish(pose);
        ros::spinOnce();
        rate.sleep();
    }

    mavros_msgs::SetMode offb_set_mode;
    offb_set_mode.request.custom_mode = "OFFBOARD";

    mavros_msgs::CommandBool arm_cmd;
    arm_cmd.request.value = true;

    ros::Time last_request = ros::Time::now();

    while(ros::ok()){
        if( current_state.mode != "OFFBOARD" &&
            (ros::Time::now() - last_request > ros::Duration(5.0))){
            if( set_mode_client.call(offb_set_mode) &&
                offb_set_mode.response.success){
                ROS_INFO("Offboard enabled");
            }
            last_request = ros::Time::now();
        } else {
            if( !current_state.armed &&
                (ros::Time::now() - last_request > ros::Duration(5.0))){
                if( arming_client.call(arm_cmd) &&
                    arm_cmd.response.success){
                    ROS_INFO("Vehicle armed");
                }
                last_request = ros::Time::now();
            }
        }

	double seconds_since_start = difftime( time(0), start);
	printf("Dur [ms]: %.1f \n", seconds_since_start);
	if(seconds_since_start == 22){
		ROS_INFO("22secs");
		pose.pose.position.z = 1;
		ros::spinOnce();
	}

	if(seconds_since_start == 28){
		ROS_INFO("28secs");
		pose.pose.position.x = -.5;
                pose.pose.position.z = 1; 
		ros::spinOnce();
	}
        if(seconds_since_start == 35){
                ROS_INFO("35secs");
                pose.pose.position.y = -.5;
                pose.pose.position.z = 1;
                ros::spinOnce();
        }
	local_pos_pub.publish(pose);
        ros::spinOnce();
        rate.sleep();
	//send a few setpoints before starting
   
    }

    
    return 0;
`}`

Here is my log data: http://logs.uaventure.com/view/E3EGxVJBSATSjzjVN3KYGZ

vinhk, 2017-04-05:
if there is some other ways to send movements commands to the drone? Any help is appreticated.
@vooon @TSC21

TSC21, 2017-04-05:
```c++
double seconds_since_start = difftime( time(0), start);
printf("Dur [ms]: %.1f \n", seconds_since_start);
if(seconds_since_start == 22){
	ROS_INFO("22secs");
	pose.pose.position.z = 1;
	ros::spinOnce();
}

if(seconds_since_start == 28){
	ROS_INFO("28secs");
	pose.pose.position.x = -.5;
            pose.pose.position.z = 1; 
	ros::spinOnce();
}
    if(seconds_since_start == 35){
            ROS_INFO("35secs");
            pose.pose.position.y = -.5;
            pose.pose.position.z = 1;
            ros::spinOnce();
    }
local_pos_pub.publish(pose);
    ros::spinOnce();
    rate.sleep();
//send a few setpoints before starting

}
```
This needs to be inside the `while(ros::ok())`. Also, why are you always initiating `double seconds_since_start`? Why not use `ros::Time`? The problem is in your control code and not in the FW or MAVROS.

vinhk, 2017-04-05:
I'm not looking to always initializing 'seconds_since_start'. It is there to check when x,y pose commands are sent.

TSC21, 2017-04-05:
Taking a look at your logs, your LOCAL.POSITION.Z doesn't leave 1 meter. Do you get a position lock win your drone?

vinhk, 2017-04-05:
For position lock I am unable to get to to work as shown when I have it in position control mode. The quadcopter drifted when in position hold mode. That can be an issue when using position set points in ros I guess.

vinhk, 2017-04-05:
In terminal yes, at round 15 seconds it switch to offboard mode.

TSC21, 2017-04-05:
No. Position lock is having the solid green on your pixhawk. Means that the FCU knows it's position.
> That can be an issue when using position set points in ros I guess.

Position setpoints in offboard mode nothing have to do position hold mode.

TSC21, 2017-04-05:
I think you must push this issue to PX4 Firmware repo, has this is does not seem to be MAVROS related. Closing this.

TSC21, 2017-04-05:
Also, try posting this on gitter. Add a log to it

closed 2017-04-05 by TSC21