mavlink/mavros #692: How to get Pixhawk with Oboard Computer to connect to QGC remote PC

vinhk, 2017-04-08:
I have a Pixhawk with Onboard computer that has IP of 192.168.1.21 and I have a QGroundControl station that has IP address of 192.168.1.8. Is there a way to connect to QGroundControl using this command? 
`roslaunch mavros px4.launch fcu_url:=/dev/ttyACM1:92100`

I know there is a `gcs_url` parameter but I can not seem to connect my drone to QGroundcontrol.

Thank you

vinhk, 2017-04-08:
@vooon @TSC21 do you guys know how to make this work?

vinhk, 2017-04-08:
it's working now. Based on this: http://rosindex.github.io/p/mavros/ 
`rosrun mavros mavros_node fcu_url:=/dev/ttyACM1:921600 gcs_url:=udp://@192.168.1.8`

closed 2017-04-08 by vinhk

MuhammadBilal1, 2024-09-24:
> it's working now. Based on this: http://rosindex.github.io/p/mavros/ `rosrun mavros mavros_node fcu_url:=/dev/ttyACM1:921600 gcs_url:=udp://@192.168.1.8`

The URL you provided isn't working.