mavlink/mavros #698: failed build mavros in raspberry pi 2

gibranmfw, 2017-04-29:
### Issue details

Hello, i was following the step for installing mavros from sources in this guide 
[https://github.com/mavlink/mavros/blob/master/mavros/README.md#installation](url)
Then i got an error saying about missing dependencies. Then i tried to install all of the dependencie using rosinstall until all. But when i tried to build using 

`catkin build`

i got errors 

> [  1%] ^[[32mBuilding CXX object src/CMakeFiles/orocos-kdl.dir/articulatedbodyinertia.cpp.o^[[0m^M
[3%] ^[[32mBuilding CXX object src/CMakeFiles/orocos-kdl.dir/chain.cpp.o^[[0m^M
In file included from /usr/include/c++/4.9/vector:69:0,^M
                 from /home/pi/ros_catkin_ws/src/orocos_kinematics_dynamics/orocos_kdl/src/segment.hpp:29,^M
                 from /home/pi/ros_catkin_ws/src/orocos_kinematics_dynamics/orocos_kdl/src/chain.hpp:25,^M
                 from /home/pi/ros_catkin_ws/src/orocos_kinematics_dynamics/orocos_kdl/src/chain.cpp:22:^M
/usr/include/c++/4.9/bits/vector.tcc: In member function ‘void std::vector<_Tp, _Alloc>::_M_fill_insert(std::vector<_Tp, _Alloc>::iterator, std::vector<_Tp, _Alloc>::size_type, const value_type&) [with _Tp = KDL::Segment; _Alloc = std::allocator<KDL::Segment>; std::vector<_Tp, _Alloc>::iterator = __gnu_cxx::__normal_iterator<KDL::Segment*, std::vector<KDL::Segment> >; typename std::_Vector_base<_Tp, _Alloc>::pointer = KDL::Segment*; std::vector<_Tp, _Alloc>::size_type = unsigned int; std::vector<_Tp, _Alloc>::value_type = KDL::Segment]’:^M
/usr/include/c++/4.9/bits/vector.tcc:449:5: internal compiler error: in build_int_cst_wide, at tree.c:1200^M
     vector<_Tp, _Alloc>::^M
     ^^M
Please submit a full bug report,^M
with preprocessed source if appropriate.^M
See <file:///usr/share/doc/gcc-4.9/README.Bugs> for instructions.^M
Preprocessed source stored into /tmp/ccYTzBvr.out file, please attach this to your bugreport.^M
src/CMakeFiles/orocos-kdl.dir/build.make:86: recipe for target 'src/CMakeFiles/orocos-kdl.dir/chain.cpp.o' failed^M
make[2]: *** [src/CMakeFiles/orocos-kdl.dir/chain.cpp.o] Error 1^M
make[2]: *** Waiting for unfinished jobs....^M
/home/pi/ros_catkin_ws/src/orocos_kinematics_dynamics/orocos_kdl/src/articulatedbodyinertia.cpp: In constructor ‘KDL::ArticulatedBodyInertia::ArticulatedBodyInertia(const KDL::RigidBodyInertia&)’:^M
/home/pi/ros_catkin_ws/src/orocos_kinematics_dynamics/orocos_kdl/src/articulatedbodyinertia.cpp:30:3: internal compiler error: Segmentation fault^M
   ArticulatedBodyInertia::ArticulatedBodyInertia(const RigidBodyInertia& rbi)^M
   ^^M
Please submit a full bug report,^M
with preprocessed source if appropriate.^M
See <file:///usr/share/doc/gcc-4.9/README.Bugs> for instructions.^M
Preprocessed source stored into /tmp/ccAp7MyB.out file, please attach this to your bugreport.^M
src/CMakeFiles/orocos-kdl.dir/build.make:62: recipe for target 'src/CMakeFiles/orocos-kdl.dir/articulatedbodyinertia.cpp.o' failed^M
make[2]: *** [src/CMakeFiles/orocos-kdl.dir/articulatedbodyinertia.cpp.o] Error 1^M
CMakeFiles/Makefile2:135: recipe for target 'src/CMakeFiles/orocos-kdl.dir/all' failed^M
make[1]: *** [src/CMakeFiles/orocos-kdl.dir/all] Error 2^M
Makefile:127: recipe for target 'all' failed^M
make: *** [all] Error 2^M

### MAVROS version and platform

ROS: Kinetic
Raspbian : Jessie

TSC21, 2017-04-29:
This is not related to mavros. `orocos` is not a dependency of mavros. Hint: try to update your GCC version

closed 2017-04-29 by TSC21

vooon, 2017-04-29:
@gibranmfw perhaps you out of memory, c++ templates eats tons of RAM at compilation. Try build with `-j2` or even `-j1`.