mavlink/mavros #701: Does anyone have code for Px4 using Mavros to send basic movements such as takeoff, forward, backward, left, and right without GPS?

vinhk, 2017-05-04:
I am using Px4flow with Pixhawk. Would like to see if there is a position control code for this case.

Thanks

TSC21, 2017-05-04:
The src for position estimation doesn't influence the position control. So the same code sending setpoints works if you use only GPS, only flow or both, cause it is the estimation modules that rely on those, not the position controllers. So https://dev.px4.io/en/ros/mavros_offboard.html will do the job.

closed 2017-05-04 by TSC21

vinhk, 2017-05-04:
@TSC21 How about left right movements?

TSC21, 2017-05-04:
Considering ENU frame, that would be changing `y` values, to positive or negative depending on the `local_origin`. Back and forward would be the `x` values.

vinhk, 2017-05-04:
if i want to change y values, I just need to do ` pose.pose.position.y = 2;` inside the `while` loop?

TSC21, 2017-05-04:
That means you are setpointing the vehicle 2 meters to the right of the local origin and it will stick in that position if you don't send a different stream of setpoints. But you can try it yourself with SITL for example

vinhk, 2017-05-04:
Okay, thanks. I think my issue is with altitude and position hold.

vinhk, 2017-05-04:
But in terms of testing in Stil I have that part.

TSC21, 2017-05-04:
Offboard mode is as it states, a mode. The same goes to altitude hold and position hold, which are also other modes. Both of them do not depende on external setpoints and only work based of the feedback of the onboard position control and estimation.