mavlink/mavros #703: Backport for Indigo

Seanmatthews, 2017-05-05:
### Feature request details

mavros has corrected a slew of errors between Indigo and Kinetic. At least one of the popular flight platforms, Snapdragon Flight, operates on Linaro 14.04 using ROS Indigo. Is it possible to create a backport of the current version of code? Or are there existing instructions to do so? I run into problems with mavlink and libmavconn compatibility when trying to install mavros from the most recent release. Attempting to install mavlink or libmavconn from source also yields similar problems.

vooon, 2017-05-05:
Did you tried [source installation][1] instructions?
Uninstall mavlink and mavros packages first, then clone kinetic versions to workspace.

Backporting is not possible because of high change count, but Kinetic version still compiles in Indigo.
That regularly tested by Travis CI.

[1]: https://github.com/mavlink/mavros/tree/master/mavros#source-installation

Seanmatthews, 2017-05-05:
@vooon Those instructions work very well, but they don't seem to put the version further than 0.17.5. And in particular, some of the message differences between Indigo and Kinetic still remain (the State message, for example). Any suggestions?

Seanmatthews, 2017-05-05:
I found that, if just before performing the `rosdep` step, you change the mavros version listed in `.rosinstall` to 0.19.0, the build will work on Indigo. This probably negates some of the wstool steps, but there it is for simplicity's sake.

closed 2017-05-05 by Seanmatthews

vooon, 2017-05-06:
You just need to add `--rosdistro kinetic` parameter to 3-rd step. But it's should give same 0.19.0 tag.