mavlink/mavros #704: Disable unnecessary topics of mavros

zer1988, 2017-05-05:
### Issue details
I would like to disable the unnecessary topics of mavros for my project such as mavros/local_position/pose while run launch file (as in attached file) because there are many topics I did not use while run 2 drones at the same time but they occupy a large amount of network load.
Thanks,

### MAVROS version and platform
[launch.docx](https://github.com/mavlink/mavros/files/980385/launch.docx)
Mavros: ?0.18.4?
ROS: Indigo
Ubuntu: 14.04

### Autopilot type and version

PX4
Version: 1.5.5

TSC21, 2017-05-06:
Put them on the `plugin_blacklist` of `px4_pluginlists.yaml`.

zer1988, 2017-05-06:
Hi TSC 21,
I found and had a look in that file but I do not understand how we can disable one necessary topic. We just type the name of topic right? Can you please give me one example how? 
Thanks.

TSC21, 2017-05-06:
No. You disable the plugin. If you want to eliminate a certain topic just remove it from code when building from src.

TSC21, 2017-05-06:
For eliminating the topics as I said you need to remove them from code in the plugins and rebuild from src as by default they are created in each plugin.

zer1988, 2017-05-06:
Hi tcs21,
Thanks for your help. I can delete unnecessary topics now. I think we can close this issue now.

closed 2017-05-06 by TSC21