mavlink/mavros #709: mavros installation on Raspberry Pi 3

b-rohit, 2017-05-11:
I tried to install mavros packages using apt-get > 
`pi@mushroom3:~ $ sudo apt-get install ros-kinetic-mavros
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package ros-kinetic-mavros
`
output of apt-get update
`pi@mushroom3:~ $ sudo apt-get update
Hit http://mirrordirector.raspbian.org jessie InRelease
Hit http://mirrordirector.raspbian.org testing InRelease                                   
Hit http://archive.raspberrypi.org jessie InRelease                                        
Hit http://mirrordirector.raspbian.org jessie/main armhf Packages                          
Hit http://packages.ros.org jessie InRelease                                           
Hit http://mirrordirector.raspbian.org jessie/contrib armhf Packages                   
Hit http://mirrordirector.raspbian.org jessie/non-free armhf Packages
Hit http://mirrordirector.raspbian.org jessie/rpi armhf Packages
Hit http://archive.raspberrypi.org jessie/main armhf Packages   
Hit http://archive.raspberrypi.org jessie/ui armhf Packages     
Hit http://mirrordirector.raspbian.org testing/main Sources                                     
Hit http://mirrordirector.raspbian.org testing/contrib Sources                                  
Hit http://mirrordirector.raspbian.org testing/non-free Sources                                 
Hit http://mirrordirector.raspbian.org testing/rpi Sources                                          
Hit http://packages.ros.org jessie/main armhf Packages                                              
Ign http://archive.raspberrypi.org jessie/main Translation-en                                       
Ign http://mirrordirector.raspbian.org jessie/contrib Translation-en                                
Ign http://archive.raspberrypi.org jessie/ui Translation-en       
Ign http://mirrordirector.raspbian.org jessie/main Translation-en 
Ign http://mirrordirector.raspbian.org jessie/non-free Translation-en
Ign http://mirrordirector.raspbian.org jessie/rpi Translation-en
Ign http://packages.ros.org jessie/main Translation-en
Reading package lists... Done                         `

os info
`pi@mushroom3:~ $ sudo apt-get update
Hit http://mirrordirector.raspbian.org jessie InRelease
Hit http://mirrordirector.raspbian.org testing InRelease                                   
Hit http://archive.raspberrypi.org jessie InRelease                                        
Hit http://mirrordirector.raspbian.org jessie/main armhf Packages                          
Hit http://packages.ros.org jessie InRelease                                           
Hit http://mirrordirector.raspbian.org jessie/contrib armhf Packages                   
Hit http://mirrordirector.raspbian.org jessie/non-free armhf Packages
Hit http://mirrordirector.raspbian.org jessie/rpi armhf Packages
Hit http://archive.raspberrypi.org jessie/main armhf Packages   
Hit http://archive.raspberrypi.org jessie/ui armhf Packages     
Hit http://mirrordirector.raspbian.org testing/main Sources                                     
Hit http://mirrordirector.raspbian.org testing/contrib Sources                                  
Hit http://mirrordirector.raspbian.org testing/non-free Sources                                 
Hit http://mirrordirector.raspbian.org testing/rpi Sources                                          
Hit http://packages.ros.org jessie/main armhf Packages                                              
Ign http://archive.raspberrypi.org jessie/main Translation-en                                       
Ign http://mirrordirector.raspbian.org jessie/contrib Translation-en                                
Ign http://archive.raspberrypi.org jessie/ui Translation-en       
Ign http://mirrordirector.raspbian.org jessie/main Translation-en 
Ign http://mirrordirector.raspbian.org jessie/non-free Translation-en
Ign http://mirrordirector.raspbian.org jessie/rpi Translation-en
Ign http://packages.ros.org jessie/main Translation-en
Reading package lists... Done                         `

apt-get source works fine 
`apt-get source ros-kinetic-mavros
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Need to get 146 kB of source archives.
Get:1 http://packages.ros.org/ros/ubuntu/ jessie/main ros-kinetic-mavros 0.18.7-0jessie (dsc) [1,433 B]
Get:2 http://packages.ros.org/ros/ubuntu/ jessie/main ros-kinetic-mavros 0.18.7-0jessie (tar) [123 kB]
Get:3 http://packages.ros.org/ros/ubuntu/ jessie/main ros-kinetic-mavros 0.18.7-0jessie (diff) [20.8 kB]
Fetched 146 kB in 0s (170 kB/s)              
dpkg-source: warning: extracting unsigned source package (ros-kinetic-mavros_0.18.7-0jessie.dsc)
dpkg-source: info: extracting ros-kinetic-mavros in ros-kinetic-mavros-0.18.7
dpkg-source: info: unpacking ros-kinetic-mavros_0.18.7.orig.tar.gz
dpkg-source: info: unpacking ros-kinetic-mavros_0.18.7-0jessie.debian.tar.xz
`

vooon, 2017-05-11:
Don't know why.

http://repositories.ros.org/status_page/ros_kinetic_default.html?q=mav - status page
http://repositories.ros.org/debian/main/pool/main/r/ros-kinetic-mavros/ - debian repo - armhf exists

But i'm not sure does armhf from ROS is suitable with Raspbian, since they rebuild everything to support ARMv6 and call that armhf.
While normal debian armhf is ARMv7 with hardware floating point (VFPv3?).

Even Pi3 uses same raspbian, so no benefits from ARMv8 of Cortex A53 cores (except that 64-bit is not needed on 1GiB RAM).

b-rohit, 2017-05-11:
@vooon  is not possible to install using apt-get on RPi, should I follow source installation guide ?

vooon, 2017-05-12:
@b-rohit probably that best choice. But perhaps you have to build whole ROS from source, that will take long-long time.
I recommend to setup `distcc` to speedup compilation and use good fast USB thumb-stick.

Or you may switch to Ubuntu armhf (worked for me at Pi2) or Debian AArch64.

b-rohit, 2017-05-16:
@vooon thanks a lot . I chose to work on your second option. I installed ubuntu mate 16.04 on Rpi3 and now successfully install both the mavros packages.

closed 2017-05-16 by b-rohit

HencyChen, 2017-07-28:
@b-rohit:
Hi, first thanks for your problem since I've the same problem. But when I install ubuntu mate 16.04 on my pi3, and run the command "sudo apt-get install ros-kinetic-mavros", I still get the same error message: "E: Unable to locate package ros-kinetic-mavros". So I'm wondering how can you make it run normally on ubuntu 16.04@pi3?

TSC21, 2017-07-28:
@b-rohit did you correctly configure your Ubuntu repositories -
http://wiki.ros.org/kinetic/Installation/Ubuntu. If you did, try following the src install - https://github.com/mavlink/mavros/tree/master/mavros#installation.

b-rohit, 2017-08-03:
@HencyChen please follow this [tutorial](http://wiki.ros.org/kinetic/Installation/Ubuntu). you have to add and configure the ros repository in your Ubuntu package list. you should be able to download using `apt`-get after that. 
@TSC21  I think you wanted to tag HencyChen.

TSC21, 2017-08-03:
@b-rohit yes, sorry for that.