mavlink/mavros #710: Mavros DeviceError:serial:open: No such file or directory

samuelguesnier, 2017-05-11:
Hello,

----

### Issue details

mavros crashes when running the command "roslaunch mavros apm.launch" and getting the error 

> serial0: device: /dev/ttyACM0 @ 57600 bps
[FATAL] [1494513763.016715558]: FCU: DeviceError:serial:open: No such file or directory

### MAVROS version and platform

ros-kinetic-mavros  version 0.18.7
Ubuntu: 16.04.2 LTS

### Autopilot type and version

[ ArduCopter ] ArduPilot

Version: ?3.7.1?

### Node logs

```
[rosmaster.main][INFO] 2017-05-11 16:54:39,508: initialization complete, waiting for shutdown
[rosmaster.main][INFO] 2017-05-11 16:54:39,508: Starting ROS Master Node
[xmlrpc][INFO] 2017-05-11 16:54:39,509: XML-RPC server binding to 0.0.0.0:11311
[xmlrpc][INFO] 2017-05-11 16:54:39,510: Started XML-RPC server [http://guesnier-HP-ProBook-650-G1:11311/]
[rosmaster.master][INFO] 2017-05-11 16:54:39,510: Master initialized: port[11311], uri[http://guesnier-HP-ProBook-650-G1:11311/]
[xmlrpc][INFO] 2017-05-11 16:54:39,510: xml rpc node: starting XML-RPC server
[rosmaster.master][INFO] 2017-05-11 16:54:39,553: +PARAM [/run_id] by /roslaunch
[rosmaster.master][INFO] 2017-05-11 16:54:39,554: +PARAM [/roslaunch/uris/host_guesnier_hp_probook_650_g1__42603] by /roslaunch
[rosmaster.master][INFO] 2017-05-11 16:54:39,603: +PARAM [/mavros/local_position/tf/frame_id] by /roslaunch
[rosmaster.master][INFO] 2017-05-11 16:54:39,603: +PARAM [/mavros/safety_area/p1/z] by /roslaunch
[rosmaster.master][INFO] 2017-05-11 16:54:39,604: +PARAM [/mavros/safety_area/p1/y] by /roslaunch
[rosmaster.master][INFO] 2017-05-11 16:54:39,604: +PARAM [/mavros/safety_area/p1/x] by /roslaunch
[rosmaster.master][INFO] 2017-05-11 16:54:39,605: +PARAM [/mavros/conn/timesync_rate] by /roslaunch
[rosmaster.master][INFO] 2017-05-11 16:54:39,605: +PARAM [/mavros/global_position/tf/send] by /roslaunch
[rosmaster.master][INFO] 2017-05-11 16:54:39,606: +SERVICE [/rosout/get_loggers] /rosout http://guesnier-HP-ProBook-650-G1:41790/
[rosmaster.master][INFO] 2017-05-11 16:54:39,606: +PARAM [/mavros/vision_pose/tf/child_frame_id] by /roslaunch
[rosmaster.master][INFO] 2017-05-11 16:54:39,606: +PARAM [/mavros/px4flow/frame_id] by /roslaunch
[rosmaster.master][INFO] 2017-05-11 16:54:39,606: +PARAM [/mavros/px4flow/ranger_min_range] by /roslaunch
[rosmaster.master][INFO] 2017-05-11 16:54:39,607: +PARAM [/mavros/setpoint_accel/send_force] by /roslaunch
[rosmaster.master][INFO] 2017-05-11 16:54:39,607: +SERVICE [/rosout/set_logger_level] /rosout http://guesnier-HP-ProBook-650-G1:41790/
[rosmaster.master][INFO] 2017-05-11 16:54:39,607: +PARAM [/mavros/vibration/frame_id] by /roslaunch
[rosmaster.master][INFO] 2017-05-11 16:54:39,608: +PARAM [/mavros/vision_pose/tf/listen] by /roslaunch
[rosmaster.master][INFO] 2017-05-11 16:54:39,608: +PARAM [/mavros/safety_area/p2/z] by /roslaunch
[rosmaster.master][INFO] 2017-05-11 16:54:39,608: +PARAM [/mavros/image/frame_id] by /roslaunch
[rosmaster.master][INFO] 2017-05-11 16:54:39,608: +PARAM [/mavros/imu/frame_id] by /roslaunch
[rosmaster.master][INFO] 2017-05-11 16:54:39,608: +PARAM [/mavros/setpoint_attitude/tf/listen] by /roslaunch
[rosmaster.master][INFO] 2017-05-11 16:54:39,609: +PARAM [/mavros/imu/magnetic_stdev] by /roslaunch
[rosmaster.master][INFO] 2017-05-11 16:54:39,610: +PUB [/rosout_agg] /rosout http://guesnier-HP-ProBook-650-G1:41790/
[rosmaster.master][INFO] 2017-05-11 16:54:39,611: +PARAM [/mavros/px4flow/ranger_max_range] by /roslaunch
[rosmaster.master][INFO] 2017-05-11 16:54:39,612: +PARAM [/mavros/setpoint_position/tf/listen] by /roslaunch
[rosmaster.master][INFO] 2017-05-11 16:54:39,612: +PARAM [/mavros/setpoint_attitude/tf/child_frame_id] by /roslaunch
[rosmaster.master][INFO] 2017-05-11 16:54:39,612: +PARAM [/mavros/sys/disable_diag] by /roslaunch
[rosmaster.master][INFO] 2017-05-11 16:54:39,612: +PARAM [/mavros/local_position/tf/child_frame_id] by /roslaunch
[rosmaster.master][INFO] 2017-05-11 16:54:39,613: +PARAM [/mavros/setpoint_attitude/tf/rate_limit] by /roslaunch
[rosmaster.master][INFO] 2017-05-11 16:54:39,613: +PARAM [/mavros/target_component_id] by /roslaunch
[rosmaster.master][INFO] 2017-05-11 16:54:39,613: +PARAM [/mavros/imu/orientation_stdev] by /roslaunch
[rosmaster.master][INFO] 2017-05-11 16:54:39,613: +PARAM [/mavros/imu/linear_acceleration_stdev] by /roslaunch
[rosmaster.master][INFO] 2017-05-11 16:54:39,614: +PARAM [/mavros/mission/pull_after_gcs] by /roslaunch
[rosmaster.master][INFO] 2017-05-11 16:54:39,614: +PARAM [/mavros/setpoint_position/tf/child_frame_id] by /roslaunch
[rosmaster.master][INFO] 2017-05-11 16:54:39,614: +PARAM [/mavros/target_system_id] by /roslaunch
[rosmaster.master][INFO] 2017-05-11 16:54:39,614: +PARAM [/mavros/startup_px4_usb_quirk] by /roslaunch
[rosmaster.master][INFO] 2017-05-11 16:54:39,615: +PARAM [/mavros/fcu_url] by /roslaunch
[rosmaster.master][INFO] 2017-05-11 16:54:39,615: +PARAM [/mavros/cmd/use_comp_id_system_control] by /roslaunch
[rosmaster.master][INFO] 2017-05-11 16:54:39,617: +PARAM [/mavros/setpoint_position/tf/rate_limit] by /roslaunch
[rosmaster.master][INFO] 2017-05-11 16:54:39,618: +PARAM [/mavros/mocap/use_pose] by /roslaunch
[... 84 lines cut ...]
[rosmaster.main][INFO] 2017-05-11 16:54:40,135: stopping master...
[rospy.core][INFO] 2017-05-11 16:54:40,135: signal_shutdown [atexit]
```

### Diagnostics

```
place here result of:
ERROR: Unable to communicate with master!

```

### Check ID

```
Unable to register with master node [http://localhost:11311]: master may not be running yet. Will keep trying.
^CTraceback (most recent call last):
  File "/opt/ros/kinetic/lib/mavros/checkid", line 108, in <module>
    main()
  File "/opt/ros/kinetic/lib/mavros/checkid", line 102, in main
    rospy.init_node("checkid", anonymous=True)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/client.py", line 336, in init_node
    raise rospy.exceptions.ROSInitException("Failed to initialize time. Please check logs for additional details")
rospy.exceptions.ROSInitException: Failed to initialize time. Please check logs for additional details
```

vooon, 2017-05-12:
Do you have `/dev/ttyACM0` device? Select your serial device by adding `fcu_url:=/dev/ttyUSB0` or similar.

samuelguesnier, 2017-05-12:
Yes I tried but it still not working. It is really weird because two days ago it was working perfectly fine. I don't see what it could be :-/

vooon, 2017-05-12:
Try to open it with terminal program `miniterm.py /dev/ttyACM0`, also try to reboot that machine.
Does you have `modem-manager` (shipped with network-manager by default)? Sometimes it lock port.

samuelguesnier, 2017-05-12:
I uninstalled modem-manager two days ago when I saw that it could cause problem.
I tried your solution but I get the error 

> No such file or directory

The only avaible port is ttyS0 , so I also tried fcu_url to connect with this port and it runs but then , I can't connect with Jakopter ( https://github.com/Jakopter/Jakopter ) 

This is the log for mavros : 

```
[roslaunch][INFO]

 2017-05-12 13:03:28,659: Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
[roslaunch][INFO] 2017-05-12 13:03:28,664: Done checking log file disk usage. Usage is <1GB.
[roslaunch][INFO] 2017-05-12 13:03:28,664: roslaunch starting with args ['/opt/ros/kinetic/bin/roslaunch', 'mavros', 'fcu_url:=/dev/ttyS0', 'apm.launch']
[roslaunch][INFO] 2017-05-12 13:03:28,664: roslaunch env is {'MANDATORY_PATH': '/usr/share/gconf/ubuntu.mandatory.path', 'ROS_DISTRO': 'kinetic', 'ROS_LOG_FILENAME': '/home/guesnier/.ros/log/a0dad35c-3702-11e7-b5a2-18cf5e11378c/roslaunch-guesnier-HP-ProBook-650-G1-5296.log', 'XDG_GREETER_DATA_DIR': '/var/lib/lightdm-data/guesnier', 'GNOME_DESKTOP_SESSION_ID': 'this-is-deprecated', 'PKG_CONFIG_PATH': '/opt/ros/kinetic/lib/pkgconfig', 'UPSTART_EVENTS': 'xsession started', 'LESSOPEN': '| /usr/bin/lesspipe %s', 'XDG_SESSION_TYPE': 'x11', 'ROSLISP_PACKAGE_DIRECTORIES': '', 'QT_IM_MODULE': 'ibus', 'LOGNAME': 'guesnier', 'WINDOWID': '62915170', 'PATH': '/usr/lib/ccache:/opt/ros/kinetic/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/home/guesnier/.local/bin:/home/guesnier/jsbsim/src:/home/guesnier/ardupilot/Tools/autotest', 'XDG_VTNR': '7', 'GNOME_KEYRING_CONTROL': '', 'CMAKE_PREFIX_PATH': '/opt/ros/kinetic', 'LD_LIBRARY_PATH': '/opt/ros/kinetic/lib', 'LANG': 'fr_FR.UTF-8', 'TERM': 'xterm-256color', 'SHELL': '/bin/bash', 'XDG_SESSION_PATH': '/org/freedesktop/DisplayManager/Session0', 'XAUTHORITY': '/home/guesnier/.Xauthority', 'LANGUAGE': 'fr_FR', 'SESSION_MANAGER': 'local/guesnier-HP-ProBook-650-G1:@/tmp/.ICE-unix/1591,unix/guesnier-HP-ProBook-650-G1:/tmp/.ICE-unix/1591', 'SHLVL': '1', 'QT_LINUX_ACCESSIBILITY_ALWAYS_ON': '1', 'QT_QPA_PLATFORMTHEME': 'appmenu-qt5', 'CLUTTER_IM_MODULE': 'xim', 'DISPLAY': ':0', 'UPSTART_INSTANCE': '', 'QT_ACCESSIBILITY': '1', 'SESSION': 'ubuntu', 'SESSIONTYPE': 'gnome-session', 'XMODIFIERS': '@im=ibus', 'ROS_ETC_DIR': '/opt/ros/kinetic/etc/ros', 'GPG_AGENT_INFO': '/home/guesnier/.gnupg/S.gpg-agent:0:1', 'HOME': '/home/guesnier', 'QT4_IM_MODULE': 'xim', 'GTK2_MODULES': 'overlay-scrollbar', 'XDG_SESSION_DESKTOP': 'ubuntu', 'COMPIZ_BIN_PATH': '/usr/bin/', 'XDG_RUNTIME_DIR': '/run/user/1000', 'INSTANCE': '', 'PYTHONPATH': '/opt/ros/kinetic/lib/python2.7/dist-packages', 'COMPIZ_CONFIG_PROFILE': 'ubuntu', 'SSH_AUTH_SOCK': '/run/user/1000/keyring/ssh', 'VTE_VERSION': '4205', 'ROS_ROOT': '/opt/ros/kinetic/share/ros', 'GDMSESSION': 'ubuntu', 'IM_CONFIG_PHASE': '1', 'UPSTART_JOB': 'unity7', 'LESSCLOSE': '/usr/bin/lesspipe %s %s', 'GNOME_KEYRING_PID': '', 'XDG_SEAT_PATH': '/org/freedesktop/DisplayManager/Seat0', 'ROS_PACKAGE_PATH': '/opt/ros/kinetic/share', 'XDG_CURRENT_DESKTOP': 'Unity', 'XDG_SESSION_ID': 'c2', 'DBUS_SESSION_BUS_ADDRESS': 'unix:abstract=/tmp/dbus-DkorivoDCG', '_': '/opt/ros/kinetic/bin/roslaunch', 'DEFAULTS_PATH': '/usr/share/gconf/ubuntu.default.path', 'GTK_IM_MODULE': 'ibus', 'DESKTOP_SESSION': 'ubuntu', 'UPSTART_SESSION': 'unix:abstract=/com/ubuntu/upstart-session/1000/1346', 'XDG_CONFIG_DIRS': '/etc/xdg/xdg-ubuntu:/usr/share/upstart/xdg:/etc/xdg', 'GTK_MODULES': 'gail:atk-bridge:unity-gtk-module', 'XDG_SEAT': 'seat0', 'GDM_LANG': 'fr_FR', 'NODE_PATH': '/usr/lib/nodejs:/usr/lib/node_modules:/usr/share/javascript', 'XDG_DATA_DIRS': '/usr/share/ubuntu:/usr/share/gnome:/usr/local/share/:/usr/share/:/var/lib/snapd/desktop', 'PWD': '/home/guesnier', 'JOB': 'unity-settings-daemon', 'ROS_MASTER_URI': 'http://localhost:11311', 'XDG_MENU_PREFIX': 'gnome-', 'LS_COLORS': 'rs=0:di=01;34:ln=01;36:mh=00:pi=40;33:so=01;35:do=01;35:bd=40;33;01:cd=40;33;01:or=40;31;01:mi=00:su=37;41:sg=30;43:ca=30;41:tw=30;42:ow=34;42:st=37;44:ex=01;32:*.tar=01;31:*.tgz=01;31:*.arc=01;31:*.arj=01;31:*.taz=01;31:*.lha=01;31:*.lz4=01;31:*.lzh=01;31:*.lzma=01;31:*.tlz=01;31:*.txz=01;31:*.tzo=01;31:*.t7z=01;31:*.zip=01;31:*.z=01;31:*.Z=01;31:*.dz=01;31:*.gz=01;31:*.lrz=01;31:*.lz=01;31:*.lzo=01;31:*.xz=01;31:*.bz2=01;31:*.bz=01;31:*.tbz=01;31:*.tbz2=01;31:*.tz=01;31:*.deb=01;31:*.rpm=01;31:*.jar=01;31:*.war=01;31:*.ear=01;31:*.sar=01;31:*.rar=01;31:*.alz=01;31:*.ace=01;31:*.zoo=01;31:*.cpio=01;31:*.7z=01;31:*.rz=01;31:*.cab=01;31:*.jpg=01;35:*.jpeg=01;35:*.gif=01;35:*.bmp=01;35:*.pbm=01;35:*.pgm=01;35:*.ppm=01;35:*.tga=01;35:*.xbm=01;35:*.xpm=01;35:*.tif=01;35:*.tiff=01;35:*.png=01;35:*.svg=01;35:*.svgz=01;35:*.mng=01;35:*.pcx=01;35:*.mov=01;35:*.mpg=01;35:*.mpeg=01;35:*.m2v=01;35:*.mkv=01;35:*.webm=01;35:*.ogm=01;35:*.mp4=01;35:*.m4v=01;35:*.mp4v=01;35:*.vob=01;35:*.qt=01;35:*.nuv=01;35:*.wmv=01;35:*.asf=01;35:*.rm=01;35:*.rmvb=01;35:*.flc=01;35:*.avi=01;35:*.fli=01;35:*.flv=01;35:*.gl=01;35:*.dl=01;35:*.xcf=01;35:*.xwd=01;35:*.yuv=01;35:*.cgm=01;35:*.emf=01;35:*.ogv=01;35:*.ogx=01;35:*.aac=00;36:*.au=00;36:*.flac=00;36:*.m4a=00;36:*.mid=00;36:*.midi=00;36:*.mka=00;36:*.mp3=00;36:*.mpc=00;36:*.ogg=00;36:*.ra=00;36:*.wav=00;36:*.oga=00;36:*.opus=00;36:*.spx=00;36:*.xspf=00;36:', 'USER': 'guesnier'}
[roslaunch][INFO] 2017-05-12 13:03:28,664: starting in server mode
[... cut ...]
[roslaunch.pmon][INFO] 2017-05-12 13:12:48,539: ProcessMonitor.shutdown <ProcessMonitor(ProcessMonitor-1, stopped daemon 139630299662080)>
[roslaunch][INFO] 2017-05-12 13:12:48,539: ... shutting down processing monitor complete
[roslaunch][INFO] 2017-05-12 13:12:48,539: done
[roslaunch.pmon][INFO] 2017-05-12 13:12:48,539: ProcessMonitor.shutdown <ProcessMonitor(ProcessMonitor-1, stopped daemon 139630299662080)>
[rospy.core][INFO] 2017-05-12 13:12:48,540: signal_shutdown [atexit]
```

But if it was working 2 days ago , I think I should reinstall all of this , I will keep you in touch

samuelguesnier, 2017-05-15:
I reinstalled my OS and it's working , I don't know why but it's working so thank you for your help :+1:

closed 2017-05-15 by samuelguesnier

Oliver-ss, 2019-04-22:
chmod 777 /dev/ttyACM0 
I solved it with this.

Mubashir-alam, 2024-05-05:
the problem solves by giving the following argument `fcu_url:="udp://:14540@127.0.0.1:14557` just make sure dependant packages are installed and run `roslaunch mavros px4.launch fcu_url:="udp://:14540@127.0.0.1:14557"`. so by default the is `/dev/ttyACM0` which mean the launch file expect an `flight controller` is connected to it. but we are taking the data from the simulation, not the physical hardware.