mavlink/mavros #711: Handle frame_id and child_id

jgoppert, 2017-05-13:
### Issue details
I would like to think about working toward using tf2 lookup as in odom2 branch. Right now we are assuming the child_id and frame_id which is really breaking the contract from the user who sent the message. Based on the child_id and frame_id, we should use tf2 to lookup the tranformations relative to the NED frame as I do here: https://github.com/mavlink/mavros/blob/odom2/mavros_extras/src/plugins/odom.cpp#L70

I believe many of the issues of users with frame confusion on mavros is that they don't know we are breaking the contract and assuming the frame_id and child_id. We should really extend this to cover all of mavros including set points etc.

Also, we should be publishing local_origin_ned as discussed here:
http://www.ros.org/reps/rep-0103.html

For outdoor systems where it is desireable to work under the north east down [6] (NED) convention, define an appropriately transformed secondary frame with the "_ned" suffix:

X north
Y east
Z down

I realize there is a concern about overhead of these lookups. I have test flown odom2 branch and didn't notice any impact. I was using tf2_ros static publisher for most transforms which should require little overhead, they fixed the issue with tf 1 static_publisher where they actually had to republish them. The only real lookup was between base_link and local_origin_ned.

@vooon, @TSC21 , what do you think?

vooon, 2017-05-13:
Really don't know. That transforms always was a painful thing...

By the way UAS already has tf2 listener: https://github.com/mavlink/mavros/blob/odom2/mavros/include/mavros/mavros_uas.h#L206

TSC21, 2017-05-13:
Maybe we can bet in that approach and see how it goes. It was always an hard time to get a propper frame transform work WRT body frames, as it is always dependent on the atittude of the vehicle. So unless we always get the the current vehicle atittude into the mix (as I did on the odom plugin) we probably want to explore the usage of tf2 and the way it allows easily transform frames (at least for body frames).

jgoppert, 2017-05-16:
@vooon OK, will use the tf2 mixin.

According to the ROS REP's 103 and 105  I think we should be publishing frames as follows:

* map -> odom (we shouldn't be publishing this, it is typically published by SLAM etc.)
* odom_ned  -> baselink_ned
* odom_ned -> odom (static)
* base_link_end -> base_link (static)

We should also support easily renaming all of the above with a prefix such as mav1, mav2 so the tf tree doesn't clash with multiple vehicles.

AlexisTM, 2017-10-13:
NOTE: tf2_static_broadcaster do not allow multiple frames to be published.

If you start a node after mavros, it will subscribe to /tf_static and only the last message of every node implementing tf2_static_broadcaster will be sent. It is simply a latched topic. 

https://github.com/ros/geometry2/blob/5d34903e1f13aa2ab69e5964db38b58810f00a63/tf2_ros/src/static_transform_broadcaster.cpp#L40-L51