mavlink/mavros #712: SITL Guided mode rejects nav commands: result 3

fbalaban, 2017-05-15:
### Issue details

ROSpy script to change to guided mode and then send nav commands to coordinates. It changes to guided mode (SetMode) but all commands with either CommandLong or CommandInt are rejected APM Planner msg: (FAILURE: Unsupported CMD: 16) 
mavlink msg:  Got MAVLink msg: COMMAND_ACK {command : 16, result : 3}

### MAVROS version and platform

Mavros: 0.18.7
ROS: Kinetic
Ubuntu: 16.04

### Autopilot type and version

[ x] ArduPilot
[  ] PX4

ArduPlane
Version: 3.6.0

### Node logs

```
no message on mavros console 
```

### Diagnostics

```
header: 
  seq: 94
  stamp: 
    secs: 1494875298
    nsecs: 810423669
  frame_id: ''
status: 
  - 
    level: 0
    name: mavros1: FCU connection
    message: connected
    hardware_id: udp://localhost:14552@/
    values: 
      - 
        key: Received packets:
        value: 12818
      - 
        key: Dropped packets:
        value: 0
      - 
        key: Buffer overruns:
        value: 0
      - 
        key: Parse errors:
        value: 0
      - 
        key: Rx sequence number:
        value: 106
      - 
        key: Tx sequence number:
        value: 126
      - 
        key: Rx total bytes:
        value: 434673
      - 
        key: Tx total bytes:
        value: 2263
      - 
        key: Rx speed:
        value: 3497.000000
      - 
        key: Tx speed:
        value: 17.000000
  - 
    level: 0
    name: mavros1: GPS
    message: 3D fix
    hardware_id: udp://localhost:14552@/
    values: 
      - 
        key: Satellites visible
        value: 10
      - 
        key: Fix type
        value: 3
      - 
        key: EPH (m)
        value: 1.21
      - 
        key: EPV (m)
        value: 2.00
  - 
    level: 0
    name: mavros1: Heartbeat
    message: Normal
    hardware_id: udp://localhost:14552@/
    values: 
      - 
        key: Heartbeats since startup
        value: 116
      - 
        key: Frequency (Hz)
        value: 0.999939
      - 
        key: Vehicle type
        value: Fixed wing aircraft
      - 
        key: Autopilot type
        value: ArduPilotMega / ArduCopter
      - 
        key: Mode
        value: GUIDED
      - 
        key: System status
        value: Active
  - 
    level: 2
    name: mavros1: System
    message: Sensor helth
    hardware_id: udp://localhost:14552@/
    values: 
      - 
        key: Sensor present
        value: 0x0061FC3F
      - 
        key: Sensor enabled
        value: 0x0061FC3F
      - 
        key: Sensor helth
        value: 0x0031FC3F
      - 
        key: 3D gyro
        value: Ok
      - 
        key: 3D accelerometer
        value: Ok
      - 
        key: 3D magnetometer
        value: Ok
      - 
        key: absolute pressure
        value: Ok
      - 
        key: differential pressure
        value: Ok
      - 
        key: GPS
        value: Ok
      - 
        key: 3D angular rate control
        value: Ok
      - 
        key: attitude stabilization
        value: Ok
      - 
        key: yaw position
        value: Ok
      - 
        key: z/altitude control
        value: Ok
      - 
        key: x/y position control
        value: Ok
      - 
        key: motor outputs / control
        value: Ok
      - 
        key: rc receiver
        value: Ok
      - 
        key: AHRS subsystem health
        value: Ok
      - 
        key: Terrain subsystem health
        value: Fail
      - 
        key: CPU Load (%)
        value: 0.0
      - 
        key: Drop rate (%)
        value: 0.0
      - 
        key: Errors comm
        value: 0
      - 
        key: Errors count #1
        value: 0
      - 
        key: Errors count #2
        value: 0
      - 
        key: Errors count #3
        value: 0
      - 
        key: Errors count #4
        value: 0
  - 
    level: 0
    name: mavros1: Battery
    message: Normal
    hardware_id: udp://localhost:14552@/
    values: 
      - 
        key: Voltage
        value: 12.44
      - 
        key: Current
        value: 10.5
      - 
        key: Remaining
        value: 87.0

```

### Check ID

```
WARNING: /mavros/target_system_id not set. Used default value: 1
WARNING: /mavros/target_component_id not set. Used default value: 1
NOTE: Target parameters may be unset, but that may be result of incorrect --mavros-ns option.Double check results!
OK. I got messages from 1:1.

---
Received 1492 messages, from 1 addresses
sys:comp   list of messages
  1:1     0, 1, 2, 133, 193, 136, 152, 22, 150, 24, 27, 29, 30, 32, 33, 35, 164, 165, 168, 42, 174, 178, 182, 62, 65, 74, 163, 87, 36, 241, 116, 125

```

fbalaban, 2017-05-17:
It seems that I cannot send waypoint related commands. I've also tested it by not renaming the mavros node (it was mavros1). Mavlink sends the rejected response but the node doesn't produce a Warning. 

Can someone reproduce? Maybe I'm doing something wrong? Isn't this supposed to work? Send command 16 in guided mode?

vooon, 2017-05-17:
Perhaps what you need is topic `~setpoint_raw/global` of type `GlobalPositionTarget.msg`.

fbalaban, 2017-05-17:
Can't I use a service instead? If not, I will give it a try. BTW I've also tried by using mavcmd with the same results.
Finally, is there a problem for Ardupilot SITL (and not PX4), because the documentation has the PX4 in parenthesis.

vooon, 2017-05-17:
mavcmd or direct call of `~cmd/command` is not matter, since autopilot responds that command is not supported.
To see what commands are actually have support you should look into sources, and that may take a while.
Possible that mission interface can handle NAV_WAYPOINT (cmd=16), but not via command if.

And what you want is like old GOTO waypoint, what is replaced by global setpoint.

vooon, 2017-05-17:
PX4 - legacy :) Current APM also support all of that commands, except it don't send loopback, so published topics stay silent.

fbalaban, 2017-05-17:
Tried that also. No response, no loopback (as you've said). 
Published GlobalPositionTarget to mavros/setpoint_raw/global but no effect. 
Am I missing something? Should I send another command after this message?

I start at auto mode with a wp mission loaded, change to guided and try to send GOTO commands..

Yes, I'm actually trying to implement a simple GOTO coordinates script. I publish constantly, but an 1Hz wouldn't be that bad.

vooon, 2017-05-17:
Hmm, [ArduPlane GCS_Mavlink.cpp L2064][1] SET_POSITION_TARGET_GLOBAL_INT looks strange!
[ArduCopter GCS_Mavlink.cpp L1693][2] looks have proper code.

Perhaps that was issue?

[1]: https://github.com/ArduPilot/ardupilot/blob/master/ArduPlane/GCS_Mavlink.cpp#L2064
[2]: https://github.com/ArduPilot/ardupilot/blob/master/ArduCopter/GCS_Mavlink.cpp#L1693

fbalaban, 2017-05-17:
Is this the one related to the mavros command? I'm not sure if that's the problem, I haven't studied the ardupilot code thoroughly. Do you think I should escalate it there?
If there is nothing more we can check (or another way to do what I have in mind - maybe change to a Copter instance - but this exceeds my needs), I can do it.

fbalaban, 2017-05-18:
FYI 
SET_POSITION_TARGET_GLOBAL_INT is indeed NOT supported in Plane.
Waiting for a resolution in ardupilot hub

fbalaban, 2017-05-19:
As @khancyr said in the [ardupilot](https://github.com/ArduPilot/ardupilot/issues/6279) issue:  * to use MAV_CMD_NAV_WAYPOINT in GUIDED mode you need to set the current parameter to 2*. How is this possible with the message definition of the GlobalPositionTarget? (Hint: it's not)

Is there any way to post lat,lon coordinates in GUIDED mode for Plane?

TSC21, 2017-06-15:
> Is there any way to post lat,lon coordinates in GUIDED mode for Plane?

You can convert global to local coordinates and send them to the FCU. Or you propose a PR to Ardupilot (and MAVROS) to add support for `SET_POSITION_TARGET_GLOBAL_INT`.

vooon, 2017-06-15:
APM:Plane do not support local setpoint at all.

TSC21, 2017-06-16:
> APM:Plane do not support local setpoint at all

Ok so that's another issue for you @fbalaban.

TSC21, 2017-06-22:
Closing this since it is an issue on the Firmware side.

closed 2017-06-22 by TSC21