mavlink/mavros #713: Simple mavlink message receiver problem

L4ncelot, 2017-05-16:
### MAVROS version and platform

Mavros: 0.19.0
ROS: Kinetic
Ubuntu: 16.04

### Issue details

Hi,

I'm currently trying to write mavlink message receiver using **libmavconn** library but unfortunately I cannot compile my code for some reason. Here's my code:

**mavlink_receiver_test.h:**
```
#ifndef PROJECT_MAVLINK_RECEIVER_TEST_H
#define PROJECT_MAVLINK_RECEIVER_TEST_H

#include <ros/ros.h>
#include <mavconn/interface.h>
#include <mavlink/v2.0/common/mavlink_msg_set_attitude_target.h>

using mavlink::mavlink_message_t;

class MavlinkReceiverTest {
public:
    MavlinkReceiverTest(const std::string& fcu_url);

protected:
    void handle_message(const mavlink_message_t* message, const mavconn::Framing framing);

    void handle_msg_set_attitude_target(const mavlink_message_t* message);

    mavconn::MAVConnInterface::Ptr link_;
};

#endif //PROJECT_MAVLINK_RECEIVER_TEST_H

```

**mavlink_receiver_test.cpp:**
```
#include "aft_controller/mavlink_receiver_test.h"

MavlinkReceiverTest::MavlinkReceiverTest(const std::string& fcu_url) {
    link_ = mavconn::MAVConnInterface::open_url(fcu_url);

    link_->message_received_cb = std::bind(&MavlinkReceiverTest::handle_message, this, std::placeholders::_1,
                                           std::placeholders::_2);
}

void MavlinkReceiverTest::handle_message(const mavlink_message_t* message, const mavconn::Framing framing) {
    switch (message->msgid) {
        case MAVLINK_MSG_ID_SET_ATTITUDE_TARGET:
            ROS_INFO_STREAM("it's working");
            handle_msg_set_attitude_target(message);
            break;
        default:
            break;
    }
}

void MavlinkReceiverTest::handle_msg_set_attitude_target(const mavlink_message_t* message) {
    (void) message;
}

int main(int argc, char** argv) {
    ros::init(argc, argv, "mavlink_receiver_test");
    ros::NodeHandle nh;

    MavlinkReceiverTest mavlink("udp://localhost:14557@localhost:14565");
[... 261 lines cut ...]
/home/l4ncelot/mavros/devel/include/mavlink/v2.0/common/mavlink_msg_set_attitude_target.h: At global scope:
/home/l4ncelot/mavros/devel/include/mavlink/v2.0/common/mavlink_msg_set_attitude_target.h:388:65: error: ‘mavlink_message_t’ does not name a type
 static inline void mavlink_msg_set_attitude_target_decode(const mavlink_message_t* msg, mavlink_set_attitude_target_t* set_attitude_target)
                                                                 ^
aft_copter/aft_controller/CMakeFiles/mavlink_receiver_test.dir/build.make:62: recipe for target 'aft_copter/aft_controller/CMakeFiles/mavlink_receiver_test.dir/src/mavlink_receiver_test.cpp.o' failed
make[2]: *** [aft_copter/aft_controller/CMakeFiles/mavlink_receiver_test.dir/src/mavlink_receiver_test.cpp.o] Error 1
CMakeFiles/Makefile2:4837: recipe for target 'aft_copter/aft_controller/CMakeFiles/mavlink_receiver_test.dir/all' failed
make[1]: *** [aft_copter/aft_controller/CMakeFiles/mavlink_receiver_test.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Base path: /home/l4ncelot/aft_copter
Source space: /home/l4ncelot/aft_copter/src
Build space: /home/l4ncelot/aft_copter/build
Devel space: /home/l4ncelot/aft_copter/devel
Install space: /home/l4ncelot/aft_copter/install
####
#### Running command: "make cmake_check_build_system" in "/home/l4ncelot/aft_copter/build"
####
####
#### Running command: "make -j4 -l4" in "/home/l4ncelot/aft_copter/build"
####
Invoking "make -j4 -l4" failed
```

It seems like some declarations are missing, but if I try to include for example `mavlink.h` or `mavlink_types.h` files I get ton of errors during compilation process. Am I missing something? Should I link some libraries against my example? I couldn't find how to use libmavconn library precisely and I'm stuck currently...

vooon, 2017-05-16:
That issue because you included "mavlink/v2.0/common/mavlink_msg_set_attitude_target.h", while libmavconn 0.19 witched to C++11 library with mavros.
You should not include anything else than `mavconn/interface.h` - it automatically include MAVLink headers.

Instead of `MAVLINK_MSG_ID_SET_ATTITUDE_TARGET` you should use `mavlink::common::msg::SET_ATTITUDE_TARGET::MSG_ID`.
Note that serialization done by libmavconn (`MAVConnInteface::send_message()`), but deserialization not (done in helper lambda done by `PluginBase::make_handler()`).

Anyway why you need to use it outside of mavros?

L4ncelot, 2017-05-16:
Oh, I didn't know that. So how could I deserialize attitude_message? 
Normally I would something like this:
```
void MavlinkReceiverTest::handle_msg_set_attitude_target(const mavlink_message_t* message) {
        mavlink_set_attitude_target_t set_attitude_target;
	mavlink_msg_set_attitude_target_decode(message, &set_attitude_target);    
}
```

Is there some example of doing such a thing?

I need to write my own mavlink receiver because I'm developing code for both **pixhawk** based copters and **DJI** copters. My program uses **MAVROS** to communicate with autopilots. The problem is that **DJI** copter doesn't use **MAVLINK** as a communication protocol. So I need to write **MAVLINK** message receiver and use DJI's ROS onboard API for controlling the copter.

vooon, 2017-05-16:
Look at `make_handler()`, perhaps some sort of that you may need (if you want to automate message handler switch).

Instead of decode you should write:
```cpp
mavlink::MsgMap map(message);
mavlink::common::msg::SET_ATTITUDE_TARGET s;
s.decode(map);

for (auto v in s.q) std::cout << v << ", ";
std::cout << std::endl;

L4ncelot, 2017-05-17:
Thank you very much for your help. Everything's working fine now.

closed 2017-05-17 by L4ncelot

alsaibie, 2018-05-09:
@vooon The answers here are very useful and not present anywhere else that I've searched. The ros node examples in PX4 stack are not up to date with the libmavconn changes discussed here. So thanks.