mavlink/mavros #717: add support for GPS_STATUS

dswk, 2017-05-17:
### Issue details

mavros does not appear to support the GPS_STATUS mavlink message, which contains info about the GPS satellites in view:

https://pixhawk.ethz.ch/mavlink/#GPS_STATUS

It looks like the ROS topic that would need to be populated is gps_common/GPSStatus.msg:

http://docs.ros.org/api/gps_common/html/msg/GPSStatus.html

### MAVROS version and platform

Mavros: 0.19
ROS: Kinetic
Ubuntu: 16.04

### Autopilot type and version

[  ] ArduPilot
[ X ] PX4

Version: 1.6.0

vooon, 2017-05-17:
Proposed message is not in standard message set, it is from gpsd_umd.
PX4 also does not support it: [mavlink_messages.cpp][1] does not have class for GPS_STATUS.

[1]: https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_messages.cpp

dswk, 2017-05-17:
Thanks vooon. I hadn't noticed gps_common wasn't in the "standard" set, bummer. It does seem to me like this mavlink message should have a way to make it through to ROS, even if in the worst case it is simply passed along as a String or something.

Even bigger bummer about not including the GPS_STATUS message in the PX4 firmware. I'm not sure how best to attack that one. Argh. Thanks for the heads-up, I hadn't gotten that far yet!

mhkabir, 2017-05-17:
As far as I know most of the standard GPS protocols do not expose all that information to the flight controller.

dswk, 2017-05-17:
Oh, no? It's in the $GPGSV message in NMEA 0183. Most standard GPS receivers that support NMEA 0183 messages support GSV, I've used it on several ones including my Android phone and OEM modules. Will have to look into how the PX4 gets it, would hate to have to carry around a second GPS receiver or something just because the protocol wasn't passing all the data through.

vooon, 2017-05-17:
Most AP's prefer to use proprietary binary protocols of receiver module because NMEA does not have proper 3D velocity info.
So NMEA often is a fallback for exotic modules.

dswk, 2017-05-17:
Thanks for the info, it does indeed appear that for the uBlox neo-n8m, while supporting GSV in NMEA mode as isual, does not seem to support it in the UBX protocol. That's a shame, seems I may have no choice but to carry an additional GPS receiver in NMEA mode. I appreciate your time.

mhkabir, 2017-05-17:
What's you use-case for the satellite data?

dswk, 2017-05-17:
There are some interesting things you can do when you know where all the satellites are in the sky and the SNR to each. A couple of examples:

http://www.cc.gatech.edu/~summetj/papers/kim_iswc_08.pdf
https://tia.ucsb.edu/about-tia/success-stories/shadowmaps/
ftp://www.ngs.noaa.gov/pub/abilich/LarsonSnow_paperOverview.ppt

mhkabir, 2017-05-17:
Interesting! 👍

dswk, 2017-05-18:
Actually it appears that for the uBlox Neo-M8N, the UBX-RXM-RAWX message does contain the C/N0 info for each satellite, among many other things, provided the receiver has protocol version 17 (some not about "Time Sync products only", not sure what that means yet). Will have to keep digging to make sure that applies to my receiver etc, and then of course there's the issue of getting support in PX4.

TSC21, 2017-05-28:
@dswk you are welcome to make a PR with the support in the `global_position` plugin so to be reviewed. But in that case you need to propose some changes on PX4 side so this can be tested on the FCU.

TSC21, 2017-06-03:
@dswk any updates?

dswk, 2017-06-03:
@TSC21 sorry, I'm new here and not exactly sure what is being asked of me. You're saying I need to go open up a feature request on the PX4 github repo to add support for the GPS_STATUS mavlink message? I see a bunch of related code in there now, e.g.:

https://github.com/PX4/Firmware/blob/master/Tools/mavlink_px4.py#L776

Unfortunately I'm not in a good position right now to dig into this and make code changes myself.

TSC21, 2017-06-03:
> You're saying I need to go open up a feature request on the PX4 github repo to add support for the GPS_STATUS mavlink message?

Yes. Explain what you are trying to achieve, what are the benefits and for tips on how to do it.

> I see a bunch of related code in there now, e.g.:

> https://github.com/PX4/Firmware/blob/master/Tools/mavlink_px4.py#L776

That's an autogenerated file for the mavlink implementation. The support needed is to add the handlers in `mavlink_messages.cpp`, `mavlink_receiver.cpp`, implementation on the `gps` driver, support on uORB msg to the new data, usage on the estimators and controllers, among others.

> Unfortunately I'm not in a good position right now to dig into this and make code changes myself.

Sometimes we just need to dig ourselves into stuff to get the features we want. But you can always ask for tips and someone will help you for sure.

TSC21, 2017-06-09:
@dswk are you up to add this feature or not?

dswk, 2017-06-09:
@TSC21 I've opened an issue over at PX4. We'll see what they say. Unfortunately I'm really swamped right now so it's probably unrealistic to take on making any of these changes myself.

TSC21, 2017-06-09:
> Unfortunately I'm really swamped right now so it's probably unrealistic to take on making any of these changes myself.

Well I can't really guarantee that someone is going to implement that besides you, unless there's someone interested on the same or someone really willing to help. I'm in the middle of other stuff right now but that is something interesting to add as a feature so probably in the future I can consider aiding you. But could be that it gets into the repo sooner than we think.

dswk, 2017-06-10:
Understood, I definitely recognize that this is a niche feature and as such is pretty far down on the priority list. I do appreciate everyone's time that has helped walk me through this so far.

TSC21, 2017-06-15:
> Understood, I definitely recognize that this is a niche feature and as such is pretty far down on the priority list. I do appreciate everyone's time that has helped walk me through this so far.

@dswk Still you are welcomed to propose a change on our side first (MAVROS). We can work further on pushing this to the Firmware side then

TSC21, 2017-06-29:
@dswk I'm closing this issue for now. Let us know if you have intentions and bandwidth to implement this and we can help you.

closed 2017-06-29 by TSC21