mavlink/mavros #718: Path generation on px4

ash-anand, 2017-05-17:
I'm working on autonomous quadcopter with PixHawk 2 autopilot and RPi 3.
I want my quad to go to a certain point but follow a certain curve, like a circle or ellipse, and not directly.
I searched about waypoints but quad stops at each waypoint , turns and then goes to the next point.
Is there any method to make quad follow the waypoints without stopping ?

Thanks

vooon, 2017-05-18:
You may stream setpoints to follow path.

amanchandra333, 2017-05-18:
@vooon the generated path has to be smooth. Can't px4 itself generate waypoints to follow a particular curve between two points?

vooon, 2017-05-18:
That is question for firmware devs.

ash-anand, 2017-05-19:
@vooon Can we use waypoint msg for local setpoints instead of GPS lat and lon ?

ash-anand, 2017-05-20:
@vooon Also is there a way to publish GPS data in OFFBOARD mode ?
I have tried publishing data on /mavros/setpoint_raw/global and /mavros/setpoint_raw/target_global but nothing happens and it gives warning CMD: Unexpected command 176, result 1.
Quad goes in AUTO.RTL mode soon after it

vooon, 2017-05-20:
Look at firmware docs. But i don't think that mission with frame=LOCAL_NED cmd=NAV_WAYPOINT are supported.
Read about OFFBOARD mode! And read [setpoint_raw][1] docs.

[1]: http://wiki.ros.org/mavros#mavros.2BAC8-Plugins.setpoint_raw

TSC21, 2017-06-09:
@ash-anand look at https://github.com/mavlink/mavros/issues/728. Maybe the answer to your issue is there.

closed 2017-06-09 by TSC21