mavlink/mavros #719: rosrun mavros mavsafety arm Request failed.

kungfrank, 2017-05-24:
Hello everyone I'm connect AMP2.6 with RPi3
After running ~$ roslaunch mavros apm2.launch fcu_url:=/dev/ttyAMA0:57600
I only got this error
**[ERROR] [1495533549.836750474]: WP: timed out.**

but I'm flying indoor without GPS,so it seem not a problem isn't it?

I have tried arming the apm with :~$ rosrun mavros mavsafety arm
result is **Request failed. Check mavros logs**

diagnostics:
I run ~$ rostopic echo -n1 /diagnostics | tee /tmp/diag.yml

```
> header: 
>   seq: 855
>   stamp: 
>     secs: 1495533498
>     nsecs: 543132484
>   frame_id: ''
> status: 
>   - 
>     level: 0
>     name: mavros: FCU connection
>     message: connected
>     hardware_id: /dev/ttyAMA0:57600
>     values: 
>       - 
>         key: Received packets:
>         value: 34700
>       - 
>         key: Dropped packets:
>         value: 0
>       - 
>         key: Buffer overruns:
>         value: 0
>       - 
>         key: Parse errors:
>         value: 0
>       - 
>         key: Rx sequence number:
>         value: 47
>       - 
>         key: Tx sequence number:
>         value: 79
>       - 
>         key: Rx total bytes:
>         value: 928321
>       - 
>         key: Tx total bytes:
>         value: 18935
>       - 
>         key: Rx speed:
>         value: 1310.000000
>       - 
>         key: Tx speed:
>         value: 34.000000
>   - 
>     level: 2
>     name: mavros: GPS
>     message: No satellites
>     hardware_id: /dev/ttyAMA0:57600
>     values: 
>       - 
>         key: Satellites visible
>         value: 0
>       - 
>         key: Fix type
>         value: 0
>       - 
>         key: EPH (m)
>         value: 0.00
>       - 
>         key: EPV (m)
>         value: Unknown
>   - 
>     level: 0
>     name: mavros: Heartbeat
>     message: Normal
>     hardware_id: /dev/ttyAMA0:57600
>     values: 
>       - 
>         key: Heartbeats since startup
>         value: 1085
>       - 
>         key: Frequency (Hz)
>         value: 1.000011
>       - 
>         key: Vehicle type
>         value: Quadrotor
>       - 
>         key: Autopilot type
>         value: ArduPilotMega / ArduCopter
>       - 
>         key: Mode
>         value: STABILIZE
>       - 
>         key: System status
>         value: Critical
>   - 
>     level: 2
>     name: mavros: System
>     message: Sensor helth
>     hardware_id: /dev/ttyAMA0:57600
>     values: 
>       - 
>         key: Sensor present
>         value: 0x0021FC0F
>       - 
>         key: Sensor enabled
>         value: 0x00219C0F
>       - 
>         key: Sensor helth
>         value: 0x0021FC0B
>       - 
>         key: 3D gyro
>         value: Ok
>       - 
>         key: 3D accelerometer
>         value: Ok
>       - 
>         key: 3D magnetometer
>         value: Fail
>       - 
>         key: absolute pressure
>         value: Ok
>       - 
>         key: 3D angular rate control
>         value: Ok
>       - 
>         key: attitude stabilization
>         value: Ok
>       - 
>         key: yaw position
>         value: Ok
>       - 
>         key: motor outputs / control
>         value: Ok
>       - 
>         key: rc receiver
>         value: Ok
>       - 
>         key: AHRS subsystem health
>         value: Ok
>       - 
>         key: CPU Load (%)
>         value: 60.2
>       - 
>         key: Drop rate (%)
>         value: 0.0
>       - 
>         key: Errors comm
>         value: 0
>       - 
>         key: Errors count #1
>         value: 0
>       - 
>         key: Errors count #2
>         value: 0
>       - 
>         key: Errors count #3
>         value: 0
>       - 
>         key: Errors count #4
>         value: 0
>   - 
>     level: 0
>     name: mavros: Battery
>     message: Normal
>     hardware_id: /dev/ttyAMA0:57600
>     values: 
>       - 
>         key: Voltage
>         value: 16.59
>       - 
>         key: Current
>         value: 28.3
>       - 
>         key: Remaining
>         value: -104.0
>
```
I got
key: 3D magnetometer
        value: Fail

But I can arm with RC transmitter before, doesn't it mean my pre-arm safety check is finish without problem?

Do I lost anything? 

ROS: Kinetic
Ubuntu Mate: 16.04 on rpi3

vooon, 2017-05-24:
Paste here log output of `roslaunch` command.

kungfrank, 2017-05-27:
I already fixed this problem!

 It work after I disable battery failsafe. 
I found this in log output of roslaunch command.
[ERROR] [1495612351.768071776]: FCU: Low Battery!

But I feel strange ,I set the “Low Battery"  threshold voltage 10.5 volts, and“Reserved MAH " 1000 with using 5200mAh battery. Then, I measured my battery which still have 14 volt. But it still didn't work until I disabled battery failsafe. 

```
pi@pi-desktop:~$ roslaunch mavros apm2.launch fcu_url:=/dev/ttyAMA0:57600
... logging to /home/pi/.ros/log/d1916586-4055-11e7-abcb-b827ebf230df/roslaunch-pi-desktop-1801.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://pi-desktop:40418/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 0.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 0.0
 * /mavros/fcu_url: /dev/ttyAMA0:57600
 * /mavros/gcs_url: 
 * /mavros/global_position/frame_id: fcu
 * /mavros/global_position/rot_covariance: 99999.0
 * /mavros/global_position/tf/child_frame_id: fcu_utm
 * /mavros/global_position/tf/frame_id: local_origin
 * /mavros/global_position/tf/send: False
 * /mavros/image/frame_id: px4flow
 * /mavros/imu/angular_velocity_stdev: 0.000349065850399
 * /mavros/imu/frame_id: fcu
 * /mavros/imu/linear_acceleration_stdev: 0.0003
 * /mavros/imu/magnetic_stdev: 0.0
 * /mavros/imu/orientation_stdev: 1.0
 * /mavros/local_position/frame_id: fcu
 * /mavros/local_position/tf/child_frame_id: fcu
 * /mavros/local_position/tf/frame_id: local_origin
 * /mavros/local_position/tf/send: False
 * /mavros/mission/pull_after_gcs: True
 * /mavros/mocap/use_pose: True
 * /mavros/mocap/use_tf: False
 * /mavros/plugin_blacklist: ['actuator_contro...
 * /mavros/plugin_whitelist: []
 * /mavros/px4flow/frame_id: px4flow
 * /mavros/px4flow/ranger_fov: 0.0
 * /mavros/px4flow/ranger_max_range: 5.0
 * /mavros/px4flow/ranger_min_range: 0.3
 * /mavros/safety_area/p1/x: 1.0
 * /mavros/safety_area/p1/y: 1.0
 * /mavros/safety_area/p1/z: 1.0
 * /mavros/safety_area/p2/x: -1.0
 * /mavros/safety_area/p2/y: -1.0
 * /mavros/safety_area/p2/z: -1.0
 * /mavros/setpoint_accel/send_force: False
 * /mavros/setpoint_attitude/reverse_throttle: False
 * /mavros/setpoint_attitude/tf/child_frame_id: attitude
 * /mavros/setpoint_attitude/tf/frame_id: local_origin
 * /mavros/setpoint_attitude/tf/listen: False
 * /mavros/setpoint_attitude/tf/rate_limit: 10.0
 * /mavros/setpoint_position/tf/child_frame_id: setpoint
 * /mavros/setpoint_position/tf/frame_id: local_origin
 * /mavros/setpoint_position/tf/listen: False
[... 133 lines cut ...]
[ INFO] [1495612306.111319140]: Built-in MAVLink package version: 2017.4.2
[ INFO] [1495612306.111770209]: Known MAVLink dialects: common ardupilotmega ASLUAV autoquad matrixpilot paparazzi slugs standard uAvionix ualberta
[ INFO] [1495612306.111965109]: MAVROS started. MY ID 1.240, TARGET ID 1.1
[ WARN] [1495612306.280312446]: GP: No GPS fix
[ INFO] [1495612306.326543800]: CON: Got HEARTBEAT, connected. FCU: ArduPilotMega / ArduCopter
[ WARN] [1495612306.514607159]: TM: Wrong FCU time.
[ WARN] [1495612308.335600070]: VER: broadcast request timeout, retries left 4
[ WARN] [1495612309.335777886]: VER: broadcast request timeout, retries left 3
[ WARN] [1495612310.357539801]: VER: unicast request timeout, retries left 2
[ WARN] [1495612311.357793895]: VER: unicast request timeout, retries left 1
[ WARN] [1495612312.358746842]: VER: unicast request timeout, retries left 0
[ WARN] [1495612313.363746183]: VER: your FCU don't support AUTOPILOT_VERSION, switched to default capabilities
[ERROR] [1495612316.348175087]: FCU: ArduCopter V3.2.1 (36b405fb)
[ERROR] [1495612316.357824261]: FCU: Frame: QUAD
[ INFO] [1495612321.340958487]: WP: mission received
[ INFO] [1495612328.586283481]: PR: parameters list received
[ WARN] [1495612336.728100968]: GP: No GPS fix
[ERROR] [1495612351.689583104]: FCU: Calibrating barometer
[ERROR] [1495612351.699051050]: FCU: barometer calibration complete
[ERROR] [1495612351.719475370]: FCU: Initialising APM...
[ERROR] [1495612351.768071776]: FCU: Low Battery!
[ WARN] [1495612366.624137832]: TM: Wrong FCU time.
[ WARN] [1495612367.239471082]: GP: No GPS fix
[ WARN] [1495612397.395583511]: GP: No GPS fix
```

mhkabir, 2017-05-27:
Great!

closed 2017-05-27 by mhkabir

yezishu2181345, 2019-09-06:
hi, my battery is good, but this problem is "Request failed. Check mavros logs", how do you deal with it ? thanks

TSC21, 2019-09-06:
@yezishu2181345 don't necro-bump in issues. If you have a problem on your own, please open a new issue, follow the issue report template, and wait for feedback. Thank you