mavlink/mavros #722: Connect 2 FCU's to same MAVROS instance

L4ncelot, 2017-05-25:
### Issue details

Hi,

I'm currently working with **DJI** copters and **ROS**. The problem which I'm facing right now is that **DJI** doesn't use **MAVLINK** as a main communication protocol. But I've been developing my code for **px4**, so I need to use **MAVLINK** with **DJI** as well. 

My idea is to send messages to **MAVROS**, then handle those messages in separated **ROS** node and send those messages to **DJI** using `dji_sdk/dji_drone.h` header file. I've already tested this scenario and it's working. The problem is that my **ROS** node is connected to **MAVROS** through **UDP** connection and it's behaving like **FCU**. But I also need to send  heartbeat etc. of my copter to **MAVROS** with DJI's **dji2mav** module.

How could I achieve this? Is there a way to connect 2 **FCU's** to one **MAVROS** instance. Isn't it a little bit weird to have some module to behave like another **FCU**?

Thank you for any advice.

### MAVROS version and platform

Mavros: 0.19.0
ROS: Kinetic
Ubuntu: 16.04

vooon, 2017-05-25:
No, MAVROS is not designed to support multiple FCU by same node.
But same system may be connected via TCP server. E.g. you use `fcu_url:=tcp-s://` and then connect FCU parts as clients.

vooon, 2017-05-25:
Other possible way is to use Intel's mavlink router.

L4ncelot, 2017-05-25:
Hmm... I will look at TCP then and give you a feedback. Thank you very much.

L4ncelot, 2017-05-25:
How do I specify if given connection is client or server? Using tcp-s prefix for server and tcp-c for client?

vooon, 2017-05-25:
http://wiki.ros.org/mavros#Connection_URL sorry, i wrong, it should be `tcp-l://`

L4ncelot, 2017-05-25:
So in the end I couldn't use the **TCP** protocol, because **DJI's** mavlink mode is written as **UDP** and I don't really want to implement my own **TCP** control. So now I'm just subscribing to whatever topic I need from **MAVROS** and act on that. But it feels weird, because I'm subscribing to the same topic as my code is publishing to.

L4ncelot, 2017-05-29:
Ok, so I've edited the DJI's mavlink module so that it supports TCP communication. Then I just connected 2 TCP clients to 1 TCP server (MAVROS node). And it works. Thank you very much again for this idea.

closed 2017-05-29 by L4ncelot

MatejHrlec, 2018-03-27:
I know, this is an old topic, but I would like to ask you, what did you have to edit to make the dji2mav module use TCP communication?
Thank you for any help.

L4ncelot, 2018-03-27:
@MatejHrlec It's almost 1 year ago since I did it. The problem is that DJI has new SDK (ROS SDK as well) and I had not check it yet cause I'm not working with DJI right now so I don't know if this information is still valid.

I did not have to modify the DJI SDK. Basically I created mavlink receiver using **mavconn** library and `dji_drone.h` header file for sending commands to DJI drone. When I receive MAVLink command message from MAVROS I decode it and send it to DJI. This way I can command DJI with MAVROS like other MAVLink copters.

MatejHrlec, 2018-03-29:
Thank you for your reply!
Are you using the mavconn receiver to only send out heartbeats, or do you have to do more stuff there?