mavlink/mavros #723: Mavros : setpoint_position/local

b-rohit, 2017-05-30:
I want to send mocap position data to pixhawk/px4 using mavros. I use the topic setpoint_position/local to publish x,y,z. I have tested it in gazebo simulation environment it works fine. I want to know following points> 
1. at what rate I should publish the coordinates ? 
2. how fast does drone fly between two published position and is it possible to change the speed ?
3. Which path does drone take between two published position?

mhkabir, 2017-05-30:
1. More than 2Hz, recommended would be 30Hz.
2. Yes, it depends on the position controller velocity parameter.
3. Minimum snap. (in theory)

closed 2017-05-30 by mhkabir

YCH188, 2019-03-08:
@mhkabir @b-rohit Hello, I want to use setpoint_position/local, but I don't know how it operate the UAV, does it changes position through subscribe to local_position/pose? And if I want to use optical position for X, Y and local_position for Z, how should I publish and subscribe the topic to mavros? Is there any instructions website for this ? Thanks a lot!