mavlink/mavros #724: Altitude problem with /mavros/cmd/takeoff

Ahrovan, 2017-06-02:
Takeoff work with mavproxy, and "takeoff 5" command -> result about 5meter altitude, but :

```
ros::ServiceClient takeoff_cl = n.serviceClient<mavros_msgs::CommandTOL>("/mavros/cmd/takeoff");
mavros_msgs::CommandTOL srv_takeoff;
srv_takeoff.request.altitude = 5;
srv_takeoff.request.latitude = 0;
srv_takeoff.request.longitude = 0;
srv_takeoff.request.min_pitch = 0;
srv_takeoff.request.yaw = 0;
if(takeoff_cl.call(srv_takeoff)){
ROS_ERROR("srv_takeoff send ok %d", srv_takeoff.response.success);
}else{
ROS_ERROR("Failed Takeoff");
}
```

result of this code with mavros is not 5meter altitude, but about 1meter.

for example plot show - takeoff 5 (in GUIDED+mavproxy) and takeoff 5 (in LOITER+mavros)
[image]

ArduPilot(Copter-3.5.0-rc7 ), ros-kinetic-mavros(0.18.7), Gazebo8 uctf SITL

vooon, 2017-06-03:
Try:
```
rosrun mavros mavsys mode -c GUIDED
rosrun mavros mavcmd takeoff 0 0 0 0 5
```

TSC21, 2017-06-09:
@Ahrovan you still have problems?

Ahrovan, 2017-06-10:
@vooon  rosrun mavros mavsys mode -c GUIDED
or
    ros::ServiceClient cl = n.serviceClient<mavros_msgs::SetMode>("/mavros/set_mode");
    mavros_msgs::SetMode srv_setMode;
    srv_setMode.request.base_mode = 0;
    srv_setMode.request.custom_mode = "LOITER";
    if(cl.call(srv_setMode)){
        ROS_ERROR("setmode send ok %d value:", srv_setMode.response.success);
    }else{
        ROS_ERROR("Failed SetMode");
        return -1;
    }
result in mavros terminal this:
CMD: Unexpected command 11, result 0

Ahrovan, 2017-06-17:
This problem solved. thx

closed 2017-06-17 by mhkabir

ashirsat, 2018-04-18:
@Ahrovan What did you do to solve this problem?