mavlink/mavros #725: Unable to Arm APM using ROS

galadriel123, 2017-06-03:
I am trying to arm the APM using ROS but whenever I run rosrun mavros mavsafety arm it says request failed. I used the mission planner to set SYSID_MYGCS to 1 and ARMING_CHECK to 0, but when I try to use rosrun mavros mavparam get SYSID_MYGCS(or ARMING_CHECK) it shows

Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/mavros/mavparam", line 103, in <module>
    main()
  File "/opt/ros/kinetic/lib/mavros/mavparam", line 99, in main
    args.func(args)
  File "/opt/ros/kinetic/lib/mavros/mavparam", line 53, in do_get
    print(param_get(args.param_id))
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/mavros/param.py", line 131, in param_get
    raise IOError("Request failed.")
IOError: Request failed.
 
The node log says that it is an Unknown Parameter.
Moreover, when I use rosparam list, I get

 /mavros/cmd/use_comp_id_system_control
/mavros/conn/heartbeat_rate
/mavros/conn/system_time_rate
/mavros/conn/timeout
/mavros/conn/timesync_rate
/mavros/distance_sensor/rangefinder_pub/field_of_view
/mavros/distance_sensor/rangefinder_pub/frame_id
/mavros/distance_sensor/rangefinder_pub/id
/mavros/distance_sensor/rangefinder_pub/send_tf
/mavros/distance_sensor/rangefinder_pub/sensor_position/x
/mavros/distance_sensor/rangefinder_pub/sensor_position/y
/mavros/distance_sensor/rangefinder_pub/sensor_position/z
/mavros/distance_sensor/rangefinder_sub/id
/mavros/distance_sensor/rangefinder_sub/orientation
/mavros/distance_sensor/rangefinder_sub/subscriber
/mavros/fcu_url
/mavros/gcs_url
/mavros/global_position/frame_id
/mavros/global_position/rot_covariance
/mavros/global_position/tf/child_frame_id
/mavros/global_position/tf/frame_id
/mavros/global_position/tf/send
/mavros/image/frame_id
/mavros/imu/angular_velocity_stdev
/mavros/imu/frame_id
/mavros/imu/linear_acceleration_stdev
/mavros/imu/magnetic_stdev
/mavros/imu/orientation_stdev
/mavros/local_position/frame_id
/mavros/local_position/tf/child_frame_id
/mavros/local_position/tf/frame_id
/mavros/local_position/tf/send
/mavros/mission/pull_after_gcs
/mavros/mocap/use_pose
/mavros/mocap/use_tf
/mavros/plugin_blacklist
/mavros/plugin_whitelist
/mavros/px4flow/frame_id
/mavros/px4flow/ranger_fov
/mavros/px4flow/ranger_max_range
/mavros/px4flow/ranger_min_range
/mavros/safety_area/p1/x
/mavros/safety_area/p1/y
/mavros/safety_area/p1/z
/mavros/safety_area/p2/x
/mavros/safety_area/p2/y
/mavros/safety_area/p2/z
/mavros/setpoint_accel/send_force
/mavros/setpoint_attitude/reverse_throttle
/mavros/setpoint_attitude/tf/child_frame_id
/mavros/setpoint_attitude/tf/frame_id
/mavros/setpoint_attitude/tf/listen
/mavros/setpoint_attitude/tf/rate_limit
/mavros/setpoint_position/tf/child_frame_id
/mavros/setpoint_position/tf/frame_id
/mavros/setpoint_position/tf/listen
/mavros/setpoint_position/tf/rate_limit
/mavros/startup_px4_usb_quirk
/mavros/sys/disable_diag
/mavros/sys/min_voltage
/mavros/target_component_id
/mavros/target_system_id
/mavros/tdr_radio/low_rssi
/mavros/time/time_ref_source
/mavros/time/timesync_avg_alpha
/mavros/time/timesync_mode
/mavros/vibration/frame_id
/mavros/vision_pose/tf/child_frame_id
/mavros/vision_pose/tf/frame_id
/mavros/vision_pose/tf/listen
/mavros/vision_pose/tf/rate_limit
/mavros/vision_speed/listen_twist
/rosdistro
/roslaunch/uris/host_raspberrypi__33145
/roslaunch/uris/host_raspberrypi__34973
/rosversion
/run_id

I don't see these parameters anywhere. I also changed target_system_id and target_component_id to 1. I am pretty new to programming using ROS.
ROS: Kinetic
Raspbian: Jessie
ArduPilot Version 3.2

Diagnostics Results
place here result of:
header:
  seq: 111
  stamp:
    secs: 1495825830
    nsecs: 256903030
  frame_id: ''
status:
  -
    level: 0
    name: mavros: FCU connection
    message: connected
    hardware_id: /dev/ttyAMA0:57600
    values:
      -
        key: Received packets:
        value: 4643
      -
        key: Dropped packets:
        value: 0
      -
        key: Buffer overruns:
        value: 0
      -
        key: Parse errors:
        value: 0
      -
        key: Rx sequence number:
        value: 152
      -
        key: Tx sequence number:
        value: 176
      -
        key: Rx total bytes:
        value: 127788
      -
        key: Tx total bytes:
        value: 3464
      -
        key: Rx speed:
        value: 1129.000000
      -
        key: Tx speed:
        value: 17.000000
  -
    level: 2
    name: mavros: GPS
    message: No satellites
    hardware_id: /dev/ttyAMA0:57600
    values:
      -
        key: Satellites visible
        value: 0
      -
        key: Fix type
        value: 0
      -
        key: EPH (m)
        value: 0.00
      -
        key: EPV (m)
        value: Unknown
  -
    level: 0
    name: mavros: Heartbeat
    message: Normal
    hardware_id: /dev/ttyAMA0:57600
    values:
      -
        key: Heartbeats since startup
        value: 151
      -
        key: Frequency (Hz)
        value: 0.965571
      -
        key: Vehicle type
        value: Quadrotor
      -
        key: Autopilot type
        value: ArduPilotMega / ArduCopter
      -
        key: Mode
        value: STABILIZE
      -
        key: System status
        value: Standby
  -
    level: 0
    name: mavros: System
    message: Normal
    hardware_id: /dev/ttyAMA0:57600
    values:
      -
        key: Sensor present
        value: 0x0020FC0F
      -
        key: Sensor enabled
        value: 0x00209C0F
      -
        key: Sensor helth
        value: 0x0020FC0F
      -
        key: 3D gyro
        value: Ok
      -
        key: 3D accelerometer
        value: Ok
      -
        key: 3D magnetometer
        value: Ok
      -
        key: absolute pressure
        value: Ok
      -
        key: 3D angular rate control
        value: Ok
      -
        key: attitude stabilization
        value: Ok
      -
        key: yaw position
        value: Ok
      -
        key: motor outputs / control
        value: Ok
      -
        key: AHRS subsystem health
        value: Ok
      -
        key: CPU Load (%)
        value: 58.0
      -
        key: Drop rate (%)
        value: 0.0
      -
        key: Errors comm
        value: 0
      -
        key: Errors count #1
        value: 0
      -
        key: Errors count #2
        value: 0
      -
        key: Errors count #3
        value: 0
      -
        key: Errors count #4
        value: 0
  -
    level: 1
    name: mavros: Battery
    message: Low voltage
    hardware_id: /dev/ttyAMA0:57600
    values:
      -
        key: Voltage
        value: 0.00
      -
        key: Current
        value: -0.0
      -
        key: Remaining
        value: -1.

 Check ID Results

OK. I got messages from 1:1.

---
Received 450 messages, from 1 addresses
sys:comp   list of messages
  1:1     0, 1, 2, 35, 36, 165, 33, 42, 29, 24, 163, 30, 150, 152, 27, 74, 62

vooon, 2017-06-03:
Please add console log, where you run roslaunch.

galadriel123, 2017-06-03:
@vooon 
It says that it is an unknown parameter :(

[ INFO] [1496481028.963215361]: CON: Got HEARTBEAT, connected. FCU: ArduPilotMega / ArduCopter
[ WARN] [1496481030.980415568]: VER: broadcast request timeout, retries left 4
[ WARN] [1496481031.982369460]: VER: broadcast request timeout, retries left 3
[ WARN] [1496481037.981306974]: VER: unicast request timeout, retries left 2
[ INFO] [1496481038.964846836]: HP: requesting home position
[ WARN] [1496481043.006720528]: PR: request list timeout, retries left 2
[ WARN] [1496481043.008704683]: VER: unicast request timeout, retries left 1
[ WARN] [1496481044.009918477]: PR: request list timeout, retries left 1
[ WARN] [1496481044.991306281]: WP: timeout, retries left 2
[ WARN] [1496481045.010812247]: PR: request list timeout, retries left 0
[ WARN] [1496481045.992354547]: WP: timeout, retries left 1
[ WARN] [1496481046.994241461]: WP: timeout, retries left 0
[ERROR] [1496481047.996924059]: WP: timed out.
[ WARN] [1496481048.029158552]: VER: unicast request timeout, retries left 0
[ INFO] [1496481048.964299153]: HP: requesting home position
[ WARN] [1496481053.047429656]: VER: your FCU don't support AUTOPILOT_VERSION, switched to default capabilities
[ WARN] [1496481058.957979324]: GP: No GPS fix
[ INFO] [1496481058.964629296]: HP: requesting home position
[ INFO] [1496481068.965017606]: HP: requesting home position
**[ERROR] [1496481073.537486173]: PR: Unknown parameter to get: SYSID_MYGCS**
[ INFO] [1496481078.965036902]: HP: requesting home position

vooon, 2017-06-03:
Please use code blocks! \```

Looks like you have an issue on node → fcu path. Check your OBC Tx → FCU Rx wiring, and GND-GND.

galadriel123, 2017-06-04:
@vooon 
I checked the wiring and it is fine. Is there any other possibility?

galadriel123, 2017-06-05:
I solved the problem. It was a firmware issue. I downgraded the firmware and it got solved.

closed 2017-06-05 by galadriel123