mavlink/mavros #730: Not getting /mavros/local_position/pose

Seanmatthews, 2017-06-15:
### Issue details

I am providing position to mavros on topic `/mavros/vision_pose/pose_cov` from a visual odometry node. If a position is being sent, should I see a callback from `/mavros/local_position/pose`? If I don't, does this mean that the quadcopter doesn't know its position?

When running the [PX4 mavros offboard example](https://dev.px4.io/en/ros/mavros_offboard.html) the quadcopter takes off, but moves forward without stopping. It does the same when I switch to using the PositionTarget with a [takeoff bitmask](http://discuss.px4.io/t/px4-hidden-flight-bitmasks/1371/6). See a video [here](https://streamable.com/u2tyr).

Any thoughts/advice?

### MAVROS version and platform

Mavros: 0.19
ROS: Indigo
Ubuntu (Linaro): 14.04

### Autopilot type and version

[  ] ArduPilot
[ X] PX4

Version: global_to_local branch

TSC21, 2017-06-15:
Are you able to switch to POSCTL on your vehicle? Do you get position lock on the vehicle after you send the position estimation?

> When running the PX4 mavros offboard example the quadcopter takes off, but moves forward without stopping. It does the same when I switch to using the PositionTarget with a takeoff bitmask. See a video here.

If the vehicle doesn't know it's position, it is pretty normal he doesn't respond normally in OFFBOARD mode. You have to first check if there is a position lock, meaning that the vehicle knows is current position.

TSC21, 2017-06-15:
@Seanmatthews I assume that you know that `global_to_local` branch is still WIP right?

Seanmatthews, 2017-06-15:
@TSC21 How do I check for position lock on the vehicle through mavros? My setup includes PX4 running on a Snapdragon Flight, so there's no light for position lock.

> @Seanmatthews I assume that you know that global_to_local branch is still WIP right?
Yes. Although the behavior is even worse when using PX4 master. [The global_to_local branch has apparently been tested SITL](https://github.com/PX4/Firmware/issues/7128).

TSC21, 2017-06-15:
`rostopic echo /mavros/local_position/pose` or check `LPOS` values on QGC logger.

Seanmatthews, 2017-06-15:
Thanks @TSC21. That's what I suspected. As noted above, I get no pose callback from `/mavros/local_position/pose`. What I do see is that the HomePosition plugin is requesting a home position: `[ INFO] [1497551223.080792531]: HP: requesting home position`. Could the lack of home position be causing a stoppage in the pipeline somewhere?

TSC21, 2017-06-16:
> Could the lack of home position be causing a stoppage in the pipeline somewhere?

As I said in the PX4 repo, no it doesn't have to do with it.

TSC21, 2017-06-21:
@Seanmatthews what is the status of this?

TSC21, 2017-06-24:
Closing for lack of updates.

closed 2017-06-24 by TSC21

LArayane, 2017-07-11:
I have no data when echoing `/mavros/local_position/pose` and that basically because there is no GPS but I need the pose data so is there a way to get it

pannitianshao, 2020-04-25:
I have a same question.when I rostopic echo /mavros/local_position/pose,nothing to callback.and  INFO] [1497551223.080792531]: HP: requesting home position. the GPS can work, and I can input GPS status by mavlink to pix to get the information of GPS.but can't get the information form ROS topic /mavros/global_position/global