mavlink/mavros #732: mavros_extras error

0Negm0, 2017-06-17:
I'm intalling ros on raspberry pi 3 model B and i get this error when building mavros_extras >>>

please help >>>>>>>>>>>>

pi@raspberrypi:~ $ cd /home/pi/ros_catkin_ws/build_isolated/mavros_extras && /opt/ros/indigo/env.sh make -j4 -l4
[ 80%] Linking CXX executable /home/pi/ros_catkin_ws/devel_isolated/mavros_extras/lib/mavros_extras/servo_state_publisher
[ 80%] Linking CXX executable /home/pi/ros_catkin_ws/devel_isolated/mavros_extras/lib/mavros_extras/copter_visualization
[ 80%] Linking CXX executable /home/pi/ros_catkin_ws/devel_isolated/mavros_extras/lib/mavros_extras/gcs_image_bridge
//usr/usrbin//binld/:ld :cannot  cannotopen  open output file /home/pi/ros_catkin_ws/devel_isolated/mavros_extras/lib/outputmavros_extras/copter_visualization: Is a directory
 file /home/pi/ros_catkin_ws/devel_isolated/mavros_extras/lib/mavros_extras/servo_state_publisher: Is a directory
collect2: error: ld returned 1 exit status
collect2: error: ld returned 1 exit status
CMakeFiles/servo_state_publisher.dir/build.make:155: recipe for target '/home/pi/ros_catkin_ws/devel_isolated/mavros_extras/lib/mavros_extras/servo_state_publisher' failed
make[2]: *** [/home/pi/ros_catkin_ws/devel_isolated/mavros_extras/lib/mavros_extras/servo_state_publisher] Error 1
CMakeFiles/Makefile2:195: recipe for target 'CMakeFiles/servo_state_publisher.dir/all' failed
make[1]: *** [CMakeFiles/servo_state_publisher.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
CMakeFiles/copter_visualization.dir/build.make:155: recipe for target '/home/pi/ros_catkin_ws/devel_isolated/mavros_extras/lib/mavros_extras/copter_visualization' failed
make[2]: *** [/home/pi/ros_catkin_ws/devel_isolated/mavros_extras/lib/mavros_extras/copter_visualization] Error 1
CMakeFiles/Makefile2:717: recipe for target 'CMakeFiles/copter_visualization.dir/all' failed
make[1]: *** [CMakeFiles/copter_visualization.dir/all] Error 2
/usr/bin/ld: cannot open output file /home/pi/ros_catkin_ws/devel_isolated/mavros_extras/lib/mavros_extras/gcs_image_bridge: Is a directory
collect2: error: ld returned 1 exit status
CMakeFiles/gcs_image_bridge.dir/build.make:228: recipe for target '/home/pi/ros_catkin_ws/devel_isolated/mavros_extras/lib/mavros_extras/gcs_image_bridge' failed
make[2]: *** [/home/pi/ros_catkin_ws/devel_isolated/mavros_extras/lib/mavros_extras/gcs_image_bridge] Error 1
CMakeFiles/Makefile2:552: recipe for target 'CMakeFiles/gcs_image_bridge.dir/all' failed
make[1]: *** [CMakeFiles/gcs_image_bridge.dir/all] Error 2
[ 80%] Built target mavros_extras
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2

vooon, 2017-06-17:
Do you follow [source installation instructions?][1]

[1]: https://github.com/mavlink/mavros/blob/master/mavros/README.md#source-installation

TSC21, 2017-06-21:
```bash
sudo apt-get install python-catkin-tools python-rosinstall-generator -y

# 1. Create the workspace: unneded if you already has workspace
mkdir -p ~/catkin_ws/src
cd ~/catkin_ws
catkin init
wstool init src

# 2. Install MAVLink
#    we use the Kinetic reference for all ROS distros as it's not distro-specific and up to date
rosinstall_generator --rosdistro kinetic mavlink | tee /tmp/mavros.rosinstall

# 3. Install MAVROS: get source (upstream - released)
rosinstall_generator --upstream mavros | tee -a /tmp/mavros.rosinstall
# alternative: latest source
# rosinstall_generator --upstream-development mavros | tee -a /tmp/mavros.rosinstall

# 4. Create workspace & deps
wstool merge -t src /tmp/mavros.rosinstall
wstool update -t src -j4
rosdep install --from-paths src --ignore-src -y

# 5. Build source
catkin build

# 6. Make sure that you use setup.bash or setup.zsh from workspace.
#    Else rosrun can't find nodes from this workspace.
source devel/setup.bash
```
Please follow this.

closed 2017-06-21 by TSC21