mavlink/mavros #733: Check If Waypoint is Reached

galadriel123, 2017-06-19:
Hey. I am working on a project where I need to ping another pi whenever a waypoint is reached using ros. Does anyone know how to do this?

vooon, 2017-06-19:
Please follow [template](https://github.com/mavlink/mavros/blob/master/.github/ISSUE_TEMPLATE).

Does param download succeed? Did you tried to setup streams (`rosrun mavros mavsys rate --all 10`)?

galadriel123, 2017-06-19:
@vooon 
Yeah. I am able to get mavros parameters and set them. I tried rostopic echo /diagnostics  and got the following. It says no satellites are visible but it clearly has a GPS fix in the mission planner. This is driving me crazy. Please help 
```
    level: 2
    name: mavros: GPS
    message: No satellites
    hardware_id: /dev/ttyAMA0:57600
    values:
      -
        key: Satellites visible
        value: 0
      -
        key: Fix type
        value: 0
      -
        key: EPH (m)
        value: Unknown
      -
        key: EPV (m)
        value: Unknown
  -
```

vooon, 2017-06-19:
@galadriel123 i asked **full** log output and diagnostics + versions, because it's really hard guess what the problem when i do not see environment.

Mission planner connected to mavros gcs internal bridge (`gcu_url`) via UDP?

TSC21, 2017-06-21:
@galadriel123 please post full log of `rostopic echo -n1 /diagnostics` and `rosrun mavros checkid` please.

TSC21, 2017-06-28:
Closing this as staled. @galadriel123 let us know if the problem persists and re-open if that's the case.

closed 2017-06-28 by TSC21